tsduck/tsduck #148: Question: MPE injection

ypo, 2018-10-24:
I would like to encapsulate multicast UDP packets to MPE and mux it to a TS file. 

I think what I want to do is very similar to 5.2.21 Multi-Protocol Encapsulation (MPE) of the User's Guide, except  that I don't have a DVB tuner for the input and the output is a file.

Any idea ?

ypo, 2018-10-25:
I only want to do MPE insertion.

My solution so far 

```
tsp \
	--bitrate 20000000 \
	-I null \
	-P regulate --bitrate 20000000 \
	-P pat --add-service 700/5000  --add-service 701/5002  \
	-P inject pmt-int.xml --pid 5000  --bitrate 15000  \
	-P inject int.xml --pid 5001  --bitrate 15000 \
	-P inject pmt-mpe.xml --pid 5002 --bitrate 15000 \
	-P sdt --service-id 700 --name "Demo INT" --provider "TSDuck" --type 0x0C \
	-P sdt --service-id 701 --name "Demo MPE" --provider "TSDuck" --type 0x0C \
	-P mpeinject 224.1.1.3:4000 --max-queue 10240 \
				--new-destination 224.16.16.2:10002 --pid 3000 \
 	-O file test.ts
```
pmt-int.xml, int.xml and  pmt-mpe.xml have been copied from the User's Guide.
int.xml IPv4_addr has been modified to 224.16.16.2
pmt-mpe.xml elementary_PID has been modified to 3000

When I run tsanalyze, I do not see the 2 services "Demon INT" and Demo MPE".

```

===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ............. Unknown  |  Services: .............. 0  |
|  Bytes: ..................... 3,609,575,184  |  PID's: Total: .......... 5  |
|  TS packets: ................... 19,199,868  |         Clear: .......... 5  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 0  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 4  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ...................... Unknown          Unknown  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................................. Unknown  |
|  First TDT UTC time stamp: ....................................... Unknown  |
|  Last TDT UTC time stamp: ........................................ Unknown  |
|  First TOT local time stamp: ..................................... Unknown  |
|  Last TOT local time stamp: ...................................... Unknown  |
|  TOT country code: ..............................................  Unknown  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless explicitly specified otherwise, all bitrates are based on   |
|  188 bytes per packet.                                                      |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 8,111,005, PID's: 1 (clear: 1, scrambled: 0)                   |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C          Unknown  |
|  0x1FFF  Stuffing ..................................... C          Unknown  |
|=============================================================================|
|  Unreferenced PID's                                                         |
|  TS packets: 11,088,863, PID's: 4 (clear: 4, scrambled: 0)                  |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Unreferenced PID's ........................... C          Unknown  |
[... 25 lines cut ...]
|  PID: 0x1389 (5001)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: ....... Unknown  Packets: ........ 14,404  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: .. 14,404  |
|=============================================================================|
|  PID: 0x138A (5002)                                           Unreferenced  |
|-----------------------------------------------------------------------------|
|  Unreferenced PID          Transport:                Discontinuities:       |
|  Bitrate: ....... Unknown  Packets: ........ 14,404  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: .. 14,404  |
[... cut ...]
|-----------------------------------------------------------------------------|
|  Global PID                Transport:                Discontinuities:       |
|  Bitrate: ....... Unknown  Packets: ..... 8,111,005  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ................. 0  Unit start: ....... 0  |
===============================================================================
```

lelegard, 2018-10-25:
~~~
	-I null \
	-P pat --add-service 700/5000  --add-service 701/5002  \
	-P sdt --service-id 700 --name "Demo INT" --provider "TSDuck" --type 0x0C \
	-P sdt --service-id 701 --name "Demo MPE" --provider "TSDuck" --type 0x0C \
~~~

The input is full of null packets and then you ask to _modify_ non-existent PAT and SDT. This works only on TS with a pre-existing structure. You need to inject the PAT and the SDT, as you do for the PMT's.

ypo, 2018-10-26:
Thanks, it works perfectly now

closed 2018-10-26 by lelegard