tsduck/tsduck #149: MERGE plugin: coredump with Input files

lars18th, 2018-10-26:
Hi,

I don't know what I'm doing wrong:

```
tsp \
 -I file ./20mbps-CBR.ts \
 -P svremove -e -s 102 \
 -P svremove -e -s 103 \
 -P svremove -e -s 104 \
 -P svremove -e -s 105 \
 -P merge "tsp -I file ./5mbps-CBR.ts" \
 -O file ./output.ts
```

The file `20mbps-CBR.ts`has 6 services (100-105) at VBR with a MUX bitrate of 20M. I "remove" 4 of them (converting the space to NULLs). And the `5mbps-CBR.ts` is a file with 2 services at VBR with a MUX bitrat of 5M.

The execution fails with this error:
```
tsp: tsplugin_merge.cpp:411: virtual void ts::MergePlugin::main(): Assertion `read_size % PKT_SIZE == 0' failed.
Aborted (core dumped)
```

Any idea?

lars18th, 2018-10-26:
Hi @lelegard ,

It's the plugin `merge` compatible with files? Or we need to use `mux` every time for files?

In the last case, the `mux` requires then o support at least: `-d|--drop` parameter to not mix PSI tables.

Please clarify this point.
Thank you!

rezna11, 2018-10-26:
Have you tried `-P regulate` when reading the files? I've had some issues with playback after doing `merge` so maybe it's related.

lelegard, 2018-10-29:
Hi,

The plugin `merge` creates a process which runs independently from the `tsp` command it is part of. Consequently, there is no determinism in the synchronization of the two processes. If the two processes run at the playout speed of their respective streams and the system is not at 100% CPU, statistically the two streams will be merged correctly. But still without strict determinism. Running the same command twice, you will get similar streams but the packets won't be necessarily in the same order.

If the two processes run at full speed (in fact at disk reading speed), the small statistical differences are no longer about a few packets but potentially about the complete file. In practice, it won't give any useful result.

This is by design. This is inherent to concurrently running processes.

Concerning the coredump, this is a bug (horror !) I will fix right away. It occurs when the pipe is closed in the middle of a packet.

rezna11, 2018-10-30:
FYI: I just ran into the same issue when trying to merge an output from ffmpeg (using latest build).

lelegard, 2018-10-31:
> FYI: I just ran into the same issue when trying to merge an output from ffmpeg (using latest build).

Which issue? Merging problem without regulate or the coredump? The former is by design. The latter should be fixed with latest builds.

rezna11, 2018-10-31:
the coredump crash - I tried to encode a TS with ffmpeg which would play a video loop and 3 audio tracks (3 radio stations) - it crashes, but only if the video component is being encoded

the following is debug log from tsduck (ffmpeg's output is silent)

```
[root@localhost dvblast]# tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.15-998
[root@localhost dvblast]# tsp --debug=4 --bitrate 3000000 -v -I null -P regulate -P merge -t "ffmpeg -loglevel quiet -re -i /home/user/Sea-6399.mp4 -filter_complex loop=-1:size=275:start=0 -i \"http://shoutcast.movemedia.eu/c-dance\" -i \"http://46.17.3.52:8400/dtr4\" -i \"http://radio-contact.ice.infomaniak.ch/radio-contact-high.mp3\" -map 0:0 -map 1:0 -map 2:0 -map 3:0 -c:v mpeg2video -b:v 1M -c:a libfdk_aac -profile:a aac_he_v2 -b:a 64k -f mpegts -" -O ip 239.0.0.115:1150
debug level set to 4
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 3,000,000 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 4
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 2
  Input plugin 1:
    Name: null
    Type: input
  Packet processor plugin 1:
    Name: regulate
    Type: packet processor
  Packet processor plugin 2:
    Name: merge
    Type: packet processor
    Arg[0]: "-t"
    Arg[1]: "ffmpeg -loglevel quiet -re -i /home/user/Sea-6399.mp4 -filter_complex loop=-1:size=275:start=0 -i "http://shoutcast.movemedia.eu/c-dance" -i "http://46.17.3.52:8400/dtr4" -i "http://radio-contact.ice.infomaniak.ch/radio-contact-high.mp3" -map 0:0 -map 1:0 -map 2:0 -map 3:0 -c:v mpeg2video -b:v 1M -c:a libfdk_aac -profile:a aac_he_v2 -b:a 64k -f mpegts -"
  Output plugin 1:
    Name: ip
    Type: output
    Arg[0]: "239.0.0.115:1150"
* Debug[4]: debug level set to 4
trying to load /usr/bin/tsplugin_null.so
symbol tspNewOutput not found in /usr/bin/tsplugin_null.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_null.so
trying to load /usr/bin/tsplugin_ip.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_ip.so
trying to load /usr/bin/tsplugin_regulate.so
symbol tspNewInput not found in /usr/bin/tsplugin_regulate.so
symbol tspNewOutput not found in /usr/bin/tsplugin_regulate.so
trying to load /usr/bin/tsplugin_merge.so
symbol tspNewInput not found in /usr/bin/tsplugin_merge.so
symbol tspNewOutput not found in /usr/bin/tsplugin_merge.so
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug[4]: null: debug level set to 4
* Debug[4]: regulate: debug level set to 4
* Debug[4]: merge: debug level set to 4
* Debug[4]: ip: debug level set to 4
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: merge: creating process "ffmpeg -loglevel quiet -re -i /home/user/Sea-6399.mp4 -filter_complex loop=-1:size=275:start=0 -i "http://shoutcast.movemedia.eu/c-dance" -i "http://46.17.3.52:8400/dtr4" -i "http://radio-contact.ice.infomaniak.ch/radio-contact-high.mp3" -map 0:0 -map 1:0 -map 2:0 -map 3:0 -c:v mpeg2video -b:v 1M -c:a libfdk_aac -profile:a aac_he_v2 -b:a 64k -f mpegts -"
* Debug: merge: receiver thread started
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* Debug: null: initial buffer load: 44,620 packets, 8,388,560 bytes
* null: input bitrate is 3,000,000 b/s
* Debug: null: input thread started
* Debug: regulate: packet processing thread started
* regulate: regulated at bitrate 3,000,000 b/s
* Debug: regulate: new regulation, burst: 10,000,000 nano-seconds, 19 packets
* Debug: ip: output thread started
* Debug: merge: packet processing thread started
tsp: tsplugin_merge.cpp:411: virtual void ts::MergePlugin::main(): Assertion `read_size % PKT_SIZE == 0' failed.
Aborted (core dumped)

```

lelegard, 2018-11-01:
Hi @rezna11,

The core dump issue was not correctly fixed. I think it is now.

rezna11, 2018-11-05:
I re-tested on CentOS 7.5 - building from today's source and it's working now. Thanks!

closed 2018-11-05 by lelegard