tsduck/tsduck #150: syntaxe plugin MPE

odoukara, 2018-10-26:
Bonjour,
Je ne suis pas sûr si je pouvais m'exprimer directement en français mais voila, je m'essaye !

je voulais tester le module MPE de TsDuck avec un encapsulateur mpe commercial que j'ai dans mon réseau. Mon architecture est la suivante :
server de fichier ---> encapsulateur mpe ---> modulateur dvb-s ---> BUC --> antenne omni
La reception se fait à l'aide d'un tuner DVB-S2

Mon objectif est de pouvoir recuprer mon data avec tsduck sur le receiver. Je n'arrive pas à ecrire la bonne commande pour les deux options suivantes:

1. Recuperer mon mpeg-ts directement à partir du tuner du receiver et enregistrer le fichier. Parametre satellite : frequence 11,320,000,000 --symbol rate 30,000,000 --fec 5/6 --polarity vertical
2. Recuperer mon data encapsulé avant le modulateur, à partir d'un IP multicast. udp://@224.10.10.5:5005. La source multicast contient plusieurs programmes.
Merci pour les suggestions

lelegard, 2018-10-29:
Bonjour,

Thanks for the effort to write in French. Answering in English so that everyone can share or participate.

What you want to do is similar to the example in section 5.2.21 in the TSDuck user's guide. The base command for extraction is the following:
~~~
tsp -I dvb --frequency 11,320,000,000 --symbol rate 30,000,000 --fec 5/6 --polarity vertical \
    -P mpe --udp-forward --redirect 224.10.10.5:5005 \
    -O drop
~~~
Then, if there are several MPE streams in the TS or if the MPE signalization is not complete, you may need to add to plugin `mpe` selection or filtering options such as `--pid` (MPE PID) or `--destination` if the MPE PID contains several UDP streams.

odoukara, 2018-10-30:
Hi, thanks a lot. Just tried it and this part of my puzzle is fixed.