tsduck/tsduck #158: Suggestion: short help

lars18th, 2018-11-07:
Hi @lelegard ,

A simple enhancement for a more user-friendly funcionality.

- At time, the call to `tsp --help` prints the help. That's OK.
- But a call to `tsp` (without any parameter) does nothing, no help, and needs to be closed (^C)

I feel it's preferable to print a short help when calling with no parameters.
You agree?

Regards.

lelegard, 2018-11-07:
Hi,

This is a standard situation. The command `tsp` is very similar to the standard Unix command `cat`: it does a lot of things depending on options but the default input is stdin and the default output is stdout. The only difference is that `cat` processes bytes while `tsp` processes TS packets.

Typing `cat` without parameter in a shell also does nothing until you send an EOF (^D) or interrupt (^C). This is also the case for `tstables`, `tstabdump`, `tsdump` and zillions of classical Unix / Linux / macOS / Solaris / etc. commands.

lars18th, 2018-11-07:
OK. Thank you for the response.
I close the issue! :wink:

closed 2018-11-07 by lars18th