tsduck/tsduck #160: EIT Issues

WinSta0, 2018-11-09:
Hello everyone,

I have been learning about DVB-T and its operations from the transport stream to the transmission. So far, I have been able to get; NIT, PAT, PMT, SDT, TDT to all work and display the video and audio to the DVB compliant set top box and TV. However, unfortunately I haven’t quite gotten the EIT to display on the set top box or the TV.

I have been able to generate an EIT table containing 2 sections: Event 1 and Event 2 (with their text and parental rating descriptor) using the OpenCaster version 3.2.2. The EIT table is then multiplexed with the other tables through OpenCaster to create a DVB-T compatible stream. This works with both receivers (including the EIT).

With the TS stream working on both receivers I used TSDuck to dump: NIT, PAT, PMT, SDT, TDT and finally the EIT. Using the command “tstabdump EIT.bin” show me two sections. However, when I convert the bin file to xml then back to bin, I see only one section (instead of two). I have been able to play around the xml file and some of TSDuck’s commands. I been able to get very close (to the original bin file) with two sections displaying but both are Section 0 instead of Section 0 at the top and Section 1 below. I use the command ”tstabdump EIT-ReComp.bin” to verify.

I would like to know if I’m missing something and what I can do through advice in the TSDuck forums in GitHub to fix the issue. The commands I used to retrieve the EIT:

“tsp -I file tsdump.ts -P tables --pid 18 --binary-output EIT.bin -O drop”

“tsp -I file tsdump.ts -P tables --pack-all-sections --all-once --pid 18 --binary-output EIT.bin -O drop”

The Commands I used to decompile the bin file to xml:
“tstabcomp -d EIT.bin”

The Commands I used to compile the xml file back to bin:
“tstabcomp -c EIT.xml -o EIT_ReComp.bin”

I have attached 2 text files: 1 “Working EIT.txt” and 2 “Non-Working EIT”.

I hope this is not too long. Wanted to cover some ground.

Alexander
[Working EIT.txt](https://github.com/tsduck/tsduck/files/2564630/Working.EIT.txt)
[Non-Working EIT.txt](https://github.com/tsduck/tsduck/files/2564631/Non-Working.EIT.txt)

lelegard, 2018-11-11:
Hi,

Sorry, but your explanations are not very clear. It is hard to understand exactly what is happening and how it is different from what you expected. Could you please post a sequential list of TSDuck commands and indicate what is incorrect?

lelegard, 2018-11-12:
> the compiled EIT bin file contains 2 sections which are both Section 0 not Section 0 and below Section 1 like the original transport stream file.

This is the effect of option `--pack-all-sections` in plugin `tables`. See the user's guide.

EIT's are strange tables which are not easy to tame. You need to clearly understand the concepts of tables with _long_ sections and tables with one _short_ section. EIT's are in between. Syntactically, there are long sections. But, in practice, EIT are rarely complete as _tables_ because they have sparse or missing sections.  You can capture all EIT _sections_ without altering them using option `--all-once` and collect them in a `.bin` file. However, in XML format, you can only manipulate complete _tables_. With short sections, a section is a table, so there is no problem. With long sections, you need all sections of a table to manipulate this table. In operational systems where EIT sections are sparse and incomplete, you need option `--pack-all-sections` to artificially rebuild a table from each EIT section in order to manipulate them as tables in XML. But if your operational system always send complete EIT tables (with all announced sections being present), then you do not need this option to manipulate them as XML.

Again, EIT's are special beasts, syntactically long tables but often sent as if they were short tables. You need to understand this and how your system actually works to manipulate them.

WinSta0, 2018-11-13:
Thanks again for the response. I didn't know that EITs were that complex. But now it makes sense. My original EITs came from OpenCaster which were easy to generate from a python script and then multiplexed onto a transport stream. The original EITs consist of two sections being 0 and 1. They are only short_event_descriptor with parental_rating_descriptor that's it. This worked on both the set top box and the TV.

I tried running “tsp -I file tsdump.ts -P tables --all-once --pid 18 --binary-output EIT.bin -O drop” and gave me a 284 bytes bin file. Then I ran the command ”tstabdump EIT.bin” to see the EITs and found there were to sections: Section 0 and Section 1. Upon decompiling the EIT.bin to EIT.xml the two sections are merged into one. When the EIT.xml is compiled back to EIT2.bin and verified through the command ”tstabdump EIT2.bin” I see only one section, Section 0 with two events in one section (the EIT2.bin file was 266 bytes not 284 like the original EIT.bin file).

I apologise if I'm repeating the questions. I mean no disrespect at all.
Q1) Is there a remote possibility that I can specify in the XML file "Section: 0 (last: 1)" and below (next one) Section: 1 (last: 1)? I have looked at the EIT example on the user documentation and found that there is no way to specify separate sections (e.g. Section 0, Section 1, Section 2, etc ) in the XML example it's self.

Q2) If this is how the EIT XML file works is there way to compile it so it has two sections (Section 0 and Section 1 not both 0)?

I hope I can get closer to how TSDuck works. I will certainly continue to read through the documentation.

Thanks,

Alexander

lelegard, 2018-11-13:
I have one question in return: You get the two events in the same section, so what is the point ?

A section is not a complete information. The complete information is a table. And the table is made of sections. So, having two events in two sections 0/1 and 1/1 of the same table and having the same two events in a single section 0/0 is exactly the same according to DVB. The fact that the first table is made of two sections and the second table is made of one section is just a matter of serialization, of data representation. At the end, in both cases, you get one table containing two events. 

This is why the important question is why do you want to create a table with two sections? Is this just to replicate the original sections? Or does it matter functionally?

More in details, with TSDuck, an XML structure represents a complete table. Most of the time, there are many different ways to serialize this table in binary. There is no deterministic way to guarantee one specific binary representation and, specifically, how many sections will be generated. It is only guaranteed that the generated table is correct and exactly contains what the XML structure describes. Usually, TSDuck fills sections up the point where the next structure (events in the case of EIT's) does not fit in the current section. At that point, the current section is closed and a new one (in the same table) is open. But it would be also correct to generate many small sections in the same table.

This is somewhat similar to a TCP communication. You send a message of N bytes and, at the end, the receiver gets those N bytes. But how many individual IP packets will convey the message is unknown and depends on many factors. You can have only one N-bytes IP packet if N is not too large or (stupidly enough), N 1-byte packets. In all cases, the receiver gets N bytes. This is the same for the TV receiver. At the end, it will get 2 events from one table and the fact that the table was transported as 1 or 2 sections is irrelevant.

Concerning EIT's specifically, there are specialized tools which create individual sections without global consistency (ie. sparse tables with missing sections). We can see that on many DVB networks. This is why I call EIT's "special beasts" because, formally speaking, they break the MPEG rules about tables. TSDuck is a generic tool which exactly follows the MPEG rules about table structures. EIT's receive no specific treatment and no waiver to break the rules (which, admittedly, may be considered as a limitation here). This is why;

1. option `--pack-all-sections` exists to turn invalid sparse EIT tables into many smaller valid EIT tables when - _and only when_ - the EIT's are sparse on a specific network. Which does not seem to be the case in your example.
2. XML EIT's are always translated as valid binary tables with event grouping inside large sections.

There is currently no other option for EIT's in TSDuck.

WinSta0, 2018-11-14:
Hello,

Thanks again for the response. Quote: “I have one question in return: You get the two events in the same section, so what is the point ?” The point is it works perfectly fine on the Samsung UA32H5000AW TV but not on the TEAC HDB850 set top box and not on the Sony Bravia KDL-46W550 TV.

The TEAC set top box displays one of the parental rating descriptors but the rest of the EPG is blank.

The Sony Bravia well nothing at all. For some reason the Samsung works without problem. VLC also displays EIT perfectly fine.

Quote: “This is why the important question is why do you want to create a table with two sections?” The reason being when I was using OpenCaster some time ago I had the same issue. The only way for it to work (according to the forums) was to create two separate EITs one being Section 0 the other being Section 1. You could also add more if your wanted to but a minimum of two sections had to exist. This solved the problem and worked with all receivers. The only question that I could not get an answer why both sections (0 and 1) had to exist (all I knew it worked that way).

Quote: “Is this just to replicate the original sections?” Yes. If I can replicate the original sections then this will certainly work on the stubborn TEAC set top box and the Sony Bravia TV.

As a side note I have created an XML EIT file with 2 sections. I then compiled it and ran “tstabdump EIT.bin” and found there were to sections. Both were 0. Using a hex editing software, I edited EIT.bin file and changed 3 bytes in the BIN file and saved the changes. Running the “tstabdump EIT.bin” this time it gave me two sections however now being Section 0 at the top and Section one bellow. E.g. before:
Section: 0 (last: 0), version: 1, size: 78 bytes
Section: 0 (last: 0), version: 1, size: 78 bytes

After hex editing:

Section: 0 (last: 1), version: 1, size: 78 bytes
Section: 1 (last: 1), version: 1, size: 78 bytes

Once verified I multiplexed the NIT, PAT, PMT, SDT, TDT and the hex edited EIT bin file (“tsp -P inject --force-crc” for EIT only) to the video and audio stream. This time all the receivers (even the stubborn ones) received the EIT the way it was meant to be.

I would like to know for future reference if I need to hex edit the BIN files every time I create / modify EIT or can TSDuck be used directly without hex editing. It would be nice if hex editing is not needed. For now, the EIT is pretty simple so this is a small inconvenience.

Thanks,

Alexander

lelegard, 2018-11-14:
The reason is probably that you send two 1-section tables with the same version and same table id. According to MPEG rules this means that they are the same table and a receiver is allowed to ignore the second one.

Instead of doing nasty hex editing, I suggest that you modify the XML file and set a different version of the table in the second `<EIT>`. This way, instead of artificially building one 2-section table, you send two 1-section tables with distinct versions.

The fact that some of your receivers ignore the second event in a 1-section EIT is a blatant violation of the definition of a DVB EIT as specified in ETSI EN 300 468.

As another option, I made a recent modification in the XML definition of NIT, BAT and EIT. In the case of an EIT, you can add the optional attribute `preferred_section` in an `<event>`  structure to force the placement of an event in a given section of the EIT. You have to pull the repo and rebuild TSDuck to use it.

lelegard, 2018-11-15:
> specifying in the first part of the EIT tag “<EIT type="0"” and not “pf” worked. 

What do you mean by "work"?

> This time the EITs displayed on all receivers.

Displayed where?

Did you try changing `type="pf"` to `type="0"` with a reason or just as a test? Because it makes all the difference. With `type="pf"`, you create an "EIT present/following". With `type="0"` (up to `type="15"`), you create an "EIT schedule". These are distinct forms of EIT's which are designed for different purposes and usually displayed in distinct applications of the receivers.

Before using EIT's, you must completely understand what they are and what they should contain before building them. This is a DVB question which is not specific to TSDuck.

WinSta0, 2018-11-16:
Hello,

Thanks again for the response. Quote” specifying in the first part of the EIT tag “<EIT type="0"” and not “pf” worked. - What do you mean by "work"?” When I changed the EIT type from “p/f” to “0 “ the short_event_descriptor with the event_name and text appeared completely on all receivers (the whole thing). Since this is not a “p/f” type EIT the Present following part of the receivers (TVs and Set top box) option was blank. In others words there was no current EIT running for this service. However, since the EIT type was set to ‘0” which meant it was (in a nutshell) an EIT schedule, this was displayed by pressing the EPG button on those receivers. That’s how it worked.

Quote: “This time the EITs displayed on all receivers. - Displayed where?”  like the previous paragraph above the EIT type “0” is displayed on the EPG section of the receivers. Because this is an EIT schedule. Since there is no present/following part in the EIT XML the receivers will only see the schedule.

Quote: “Did you try changing type="pf" to type="0" with a reason or just as a test?” It was simply a test to see if this was working at all. This obviously worked.

Now I will need to find out how to get the EIT type p/f to work. Are there any examples you might have? (two p/f EITs will be fine). Please let me know if this alright.

On the other hand, I discovered some limitations (at least for now) that there are missing options in the EIT XML fields that can be specified (in TSDuck). In this PDF link “ https://www.etsi.org/deliver/etsi_en/300400_300499/300468/01.03.01_60/en_300468v010301p.pdf ” in page 21 and 22 (EIT) you will see “Semantics for the event information section:”. There are some options that are not included in TSDuck. The ones I had to hex edit were: section_number and last_section_number. If these two were included in TSDuck (in the EIT XML) then I can pretty much guarantee that I would have the EIT p/f working without a problem and therefore, no hex editing required.

I also looked into an EPG generator (cherryEPG) and also did some sniffing on DVB-T free to air channels. These two had one thing in common that all of the section numbers were increments of one with their corresponding last_section_number. In other words, let say I had two EITs, the first one being Section 0 and the other Section 1. Since these two can be linked the first Section would be 0 with a last_section_number being 1. Section 1 would then have the last_section_number 1. For this reason, the p/f type EIT works without problem in OpenCaster.

Is there any chance some updates can be added to TSDuck?

Thanks,

Alexander

lelegard, 2018-11-16:
> In other words, let say I had two EITs, the first one being Section 0 and the other Section 1. Since these two can be linked the first Section would be 0 with a last_section_number being 1. Section [...]

This is exactly how TSDuck will build any table with long sections, including EIT. Simply because this is the only valid way to build MPEG tables. When TSDuck serializes a table and realizes that the table needs two sections, then you will always have:

- Section 0: section_number = 0, last_section_number = 1
- Section 1: section_number = 1, last_section_number = 1

WinSta0, 2018-11-19:
Hello,

Thanks, lelegard and colinu69. Yes, I have been suspecting it. Anyway, now the EIT p/f tables are all working across all receivers. So, thanks again.

Just as a side note the way I got this EIT p/f tables to work were to create an XML file with one big section. That is add one short_event_descriptor and text per event_id and also add 3 extended_event_descriptor (per event_id). With the descriptors being max out to DVB standards the XML was complied to BIN which caused the compiled BIN file to have two Sections. One being Section: 0 (last: 1) and the other being Section: 1 (last: 1). From here the EIT p/f tables work without issue.

Since the issue has been resolved, it can now be closed.

Thanks,

Alexander

lelegard, 2018-11-20:
Thanks @colinu69 for pointing to the SI guidelines. The serialization of EIT's has been modified to follow the 2-sections rule for EIT p/f and the segmentation rules for EIT schedule. Please pull and rebuild if you'd like to test.

closed 2018-12-06 by lelegard

premultiply, 2019-08-30:
Is there a way to make the XML export indicate the section number to identify the EIT present event?
The section number is indicated in Text export 
```
* EIT p/f Actual, TID 78 (0x4E), PID 18 (0x0012)
  Section: 0 (last: 1), version: 9, size: 492 bytes
```
but it can no be found in XML export:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <EIT type="pf" version="9" current="true" actual="true" service_id="0x0036" transport_stream_id="0x0036" original_network_id="0xFFFF" last_table_id="0x4E">
    <!-- PID 0x0012 (18) -->
```

I also tried to remove the section 1 from the EIT but I was not successful.