tsduck/tsduck #162: monitoring DVB-C system with TSDuck/TBS6209

jimmylim88, 2018-11-13:
Hello,

we are looking at using TSDuck to monitor our MMDS transmitters output carriers using the TBS6209 OctaTV Tuner card.

The application is fairly simplistic - we simply need to tune in to a carrier frequency using the TBS6209 card and if it's successful, then it's deemed as "UP" and move on to the next frequency.  We plan on using a single card to cycle through the 21 frequencies that we have.

We'd like to know if there'd be any caveats in our application before purchasing the TBS6209 card - any assistance is greatly appreciated.

Thanks!

lelegard, 2018-11-13:
Hi @jiml88 

I haven't used any TBS6209 (or any other DVB-C tuner) myself. Currently, no user has posted feedback on it in issue #24. Other TBS tuners received good feedback on Windows (no feedback on Linux unfortunately, neither good nor bad). 

Normally, if the driver is functional on a given OS, it should work with TSDuck. If your OS choice is open, I would recommend Linux (if a driver is available for that tuner). For that kind of application, Linux has always proven to be more reliable on the long term. On Windows, the DirectShow framework introduces reception artifacts sometimes.

If you encounter problems in the usage of this device with TSDuck, let us know here.

jiml88, 2018-11-14:
OK, thanks for the prompt reply.  We are going to move forward with this solution and I'll post feedback on it once we're done.