tsduck/tsduck #163: Request: range for "filter" filter

lars18th, 2018-11-14:
Hi @lelegard ,

At time, we have the option to use "pid ranges" with the filters: `duplicate`, `remap`, `encap` and perhaps others that I don't remember now. But, in the filter `filter` this isn't available.

Please, see this example:
```
tsp \
 --add-input-stuffing 1/30
 -I file input.ts \
 -P duplicate 0-21=7700 \
 -P encap -p 7700-7721 -o 7722 \
    ...
 -P decap -p 7722 \
 -P filter -n -s -p 0x0 -p 0x1 -p 0x2 -p 0x3 -p 0x4 -p 0x5 -p 0x6 -p 0x7 -p 0x8 -p 0x9 -p 10 -p 11 -p 12 -p 13 -p 14 -p 15 -p 16 -p 17 -p 18 -p 19 -p 20 -p 21  \
 -P remap -n -u 7700-7721=0 \
 -O drop
```

I feel I don't need to explain the requirement, right? :wink: 

You can improve this filter, please?
Regards.

lelegard, 2018-11-16:
Hi @lars18th,

I implemented this in a more general way. Now, in all commands and all plugins, all options with an integer value which can be specified multiple times can now use ranges. This includes your request but also all potential similar requests. The fix was done in the generic command line arguments class.

From the user's guide, section 3.1.2 (latest update):

Example: The following sets of options are equivalent:
~~~
--pid 0 --pid 0x20 --pid 0x21 --pid 0x22 --pid 0x23 --pid 0x24 --pid 0x25 --pid 0x40
--pid 0 --pid 0x20-0x25 --pid 0x40
~~~

closed 2018-11-17 by lelegard

lars18th, 2018-11-17:
You're so cool! :1st_place_medal: