tsduck/tsduck #164: CSA-Test not working...

neo7530, 2018-11-14:
Hi,
I have tested the CSA scrambler and descrambler, but it isn't working.
`tsp -v -I file 16Q_25_44.ts -P scrambler "Charlotte Video 1" -e localhost:2222 --super-cas-id 0x05000001 --access-criteria 0123456789 --pid-ecm 0x1111 -P descrambler "Charlotte Video 1" -O play`
>>
* file: input bitrate is 25,439,625 b/s
* play: using media player command: "/usr/bin/vlc" - --play-and-exit
* zap: found service "Charlotte Video 1", service id is 0x1001
* zap: found service id 0x1001, PMT PID is 0x1000
* scrambler: found service "Charlotte Video 1", service id is 0x1001 (4097)
* scrambler: found service id 0x1001 (4097), PMT PID is 0x1000 (4096)
* scrambler: starting scrambling PID 0x0100
* scrambler: starting scrambling PID 0x0101
VLC media player 3.0.3 Vetinari (revision 3.0.3-1-0-gc2bb759264)
* descrambler: found service "Charlotte Video 1", service id is 0x1001 (4097)
* descrambler: found service id 0x1001 (4097), PMT PID is 0x1000 (4096)
* descrambler: using ECM PID 4369 (0x1111)
* descrambler: using scrambling mode: DVB-CSA2

But the Video stays scrambled. I had a test with static CW, then it works, but with tsecmg it dosn't.

What is going wrong?

lelegard, 2018-11-14:
Hi,

My guess is that you do not have any stuffing in the input stream, meaning no room to insert any packet, meaning no ECM inserted, meaning no CW for the descrambler.

As a test, use option `tsp --add-input-stuffing 1/10`. If this works, my guess is right. But do not simply stay with this. This is a basic principle of `tsp` about data insertion that you must understand. Otherwise, you will have problems all the time. See the TSDuck user's guide, `tsp` reference documentation, section named _"Modifying, inserting and deleting packets"_.

neo7530, 2018-11-14:
ahh, now it is working... will read on in the user's guide... Thank you!

neo7530, 2018-11-15:
it is an issue of ffmpeg, when i mux multiple video-files to a multiple-program TS. All stuffing data will add to video-stream #0 so there are no "visible stuffing packets". How can i filter these packets and replace it with stuffing-packets on the fly? tryed
`-P filter --max-payload-size 0 --negate -s`

is this the right way ?

Thank you. - Marco.

lelegard, 2018-11-15:
This is the right way in this context.

But, beyond this, why does ffmpeg generate packets with an empty payload? Is this the right source of empty packet for ECM's?

lelegard, 2018-11-15:
OK, so the reason is that ffmpeg has its own interpretation of "stuffing" :wink:

neo7530, 2018-11-15:
Yep :) 

So i will close this issue now.

Thanks anyway, Marco.

closed 2018-11-15 by neo7530