tsduck/tsduck #165: --raw-dump output line length question

rezna11, 2018-11-15:
Hi, I wanted to ask if there's a way to save each packet of 188 bytes in 1 line of text. When I use` --raw-dump` option with `tstables` or `tsdump`, I get text in blocks with 16 bytes per line. Can TSDuck do 188 bytes per line without modifying the source code? In this particular case I know that all tables fit in 188 bytes but theoretically if they don't, then the line could be longer than 188 bytes, as long as the whole table. 
The reason I ask for this is to be able to visually compare multiple occurances of a PID and easily spot differences in a simple text file.

lelegard, 2018-11-16:
Did you try `tstables --log --log-size 200` (or any greater value) ?

This is for tables. Do you need this for `tsdump` too ?

lelegard, 2018-11-17:
Added options `--log`, `--log-size`, `--no-headers` and `--pid` to `tsdump`.

rezna11, 2018-11-17:
Thank you, I didn't realize there's the `--log` option for `tstables` - it sure does what I wanted. I could not try `tsdump` yet because it won't compile in Windows... I will try Linux.

[image]

rezna11, 2018-11-17:
I tried in Linux and I'm not sure if it works as it should... with command

`tsp -v -I ip 239.0.0.54:3003 | tsdump -r --pid 5677 --log --log-size 200
`
I get this output, which looks like full of null packets (PID 5677 has always data, there is no stuffing) so the PID filter isn't working and also there is never a new line character, plus after 16 bytes a space is missing:

[image]

lelegard, 2018-11-17:
Remove the `-r` option (`--raw`). In that case, the file is not even interpreted as a TS file, it can dump any _raw_ binary file. 

I will make sure that normally incompatible options such as `--raw` and `--log` are rejected.

Concerning Windows, I will check. I tested the last commit on macOS only. I hate recompiling on Windows. Using identical hardware, building TSDuck with Visual Studio is 10 to 15 times longer than on macOS with LLVM or Linux with GCC.

rezna11, 2018-11-17:
Alright, that makes sense and in that case, filtering PID works as well as line formatting!

lelegard, 2018-11-17:
No problem rebuilding on Windows. The message you got is related to some old code that did not change for ages anyway. Maybe there is some inconsistencies in your VS code base. Previous interrupted compilation, maybe. You could try to cleanup the 4 compilation directories (Release-x64, etc.) and retry?

rezna11, 2018-11-17:
thanks, Clean and Rebuild helped

lelegard, 2018-11-17:
Added option consistency checking in `tsdump` to disallow `--raw --log`.

closed 2018-11-21 by lelegard