tsduck/tsduck #166: BUG plugin ENCAP initial start inconsistencies

lars18th, 2018-11-15:
Hi @lelegard ,

After a lot of testing (several days), I found a very strange bug in the `encap` plugin.

The trouble is with the starting packets. In particular with the **first** packet. Without using the `--pack` mode, without PCRs, and with sufficient NULL packets, the expected beheviour is:

- One packet has the PUSI flag set, the byte 5 is `0` as it points to an starting packet, and the payload has 184 bytes of the original packet.
- And the following packet doesn't have the PUSI flag, and the Adaptation padding stuff occupies 180 bytes, so it only leaves 4 bytes at end of the original packet.

This is right. And it works with _all_ packets... **except** with the _first_ and _second_ packet. Instead of it:

- The first packet, instead of `0` in byte [5] it has a `4`. And the payload (the start of the packet) doesn't starts in the position {4} (byte [6] of the TS packet) but in the byte [12] (0x0C in the TS packet). And the bytes between [5]-[11] have random data.
- And the second packet, instead of 4 bytes at end of payload, it has 8 bytes.

I checked the code for an initialization error, but I can't find anything.
Any ideas?
Please, help me!

lars18th, 2018-11-15:
Hi @lelegard ,

This is a **very obscure bug**!!! :open_mouth: 
In fact, using _only_ the encap filter, the problem doesn't appears.
Everything seems to point to a memory leak inside the library.

Fortunately I have found a test that reproduces the error.
It uses the "test-001.ts" file from the tesduck-test:
https://github.com/tsduck/tsduck-test/blob/master/input/test-001.ts

Here a command that **doesn't** fail:
```
tsp  --add-input-stuffing 1/20 \
 -I file test-001.ts \
 -P duplicate 0-18=8000 \
 -P encap -p 8000-8018 -o 8032 \
 -P svrename -p "EDIT" 8801 \
  -O file out-no-error.ts
```

And this command produces these two first packets of pid 8032 (the one of the capsulation):
```
Packet: 2603
File Offset: 489364
Time: 00h00m00:078
?

Header:

sync_byte: 0x47 (71)
transport_error_indicator: 0
payload_unit_start_indicator: 1
transport_priority: 0
PID: 0x1F60 (8032)
transport_scrambling_control: 0 (not scrambled)
adaptation_field_control: 1 (no adaptation_field, payload only)
continuity_counter: 0x6 (6)

Data:
         0001 0203 0405 0607 0809 0A0B 0C0D 0E0F 0123456789ABCDEF
0x000000 475F 6016 005F 401F 0000 B039 0438 D900 G_`.._@....9.8..
0x000010 0000 00E0 1022 61E0 6422 62E0 C822 63E1 ....."a.d"b.."c.
0x000020 2C22 64E1 9022 65E1 F422 66E2 5822 67E2 ,"d.."e.."f.X"g.
0x000030 BC22 68E3 2022 69E3 8422 6AE3 E822 C3F0 ."h. "i.."j.."..
0x000040 032C AEB2 5AFF FFFF FFFF FFFF FFFF FFFF .,..Z...........
0x000050 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x000060 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x000070 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x000080 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x000090 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x0000a0 FFFF FFFF FFFF FFFF FFFF FFFF FFFF FFFF ................
0x0000b0 FFFF FFFF FFFF FFFF FFFF FFFF           ............
```
```
Packet: 2624
File Offset: 493312
Time: 00h00m00:078
?

Header:

sync_byte: 0x47 (71)
transport_error_indicator: 0
payload_unit_start_indicator: 0
transport_priority: 0
PID: 0x1F60 (8032)
transport_scrambling_control: 0 (not scrambled)
adaptation_field_control: 3 (adaptation_field followed by payload)
continuity_counter: 0x7 (7)

[... 91 lines cut ...]
         0001 0203 0405 0607 0809 0A0B 0C0D 0E0F 0123456789ABCDEF
0x000000 475F 6018 08FF FFFF FFFF FFFF FF1F 5218 G_`...........R.
0x000010 7269 6361 696E 2072 E961 6C69 73E9 2065 ricain r.alis. e
0x000020 6E20 3230 3039 2028 372F 3133 292E 2022 n 2009 (7/13). "
0x000030 5265 6C69 7175 6573 2064 276F 766E 6973 Reliques d'ovnis
0x000040 222E 2041 206C 6120 7265 6368 6572 6368 ". A la recherch
0x000050 6520 6465 206D 6963 726F 7363 6F70 6971 e de microscopiq
0x000060 7565 7320 7061 7274 6963 756C 6573 206F ues particules o
0x000070 7520 6465 2067 7261 6E64 6573 2070 69E8 u de grandes pi.
0x000080 6365 7320 6661 6974 6573 2064 6520 6DE9 ces faites de m.
0x000090 7461 6C20 696E 636F 6E6E 7520 7375 7220 tal inconnu sur 
0x0000a0 5465 7272 652C 2071 7569 2063 6F6E 7374 Terre, qui const
0x0000b0 6974 7565 6E74 206C 6573 2070           ituent les p
```
And here you can see the error. The _first_ encapsulated packet has this data (bytes[0]-[15]):
```
475F 6017 0420 616D E95F 401F 0000
```
Please, note that the first four bytes are correct. But the byte[5] is `0x04` instead of `0x00`. Futhermore, the start of the original packet (it's a PAT table, so PID=0) it's at byte[12]. And the bytes[5]-[11] are spurious. Futhermore, the second packet has 8 bytes of the original packet (instead of 4). So at any case, the start in the first packet should be at byte[8] to be true. And yes, the second packet is valid (the pointer is correct); but the _first_ one is erroneous!

So, the conclusion is that some error is "overwriting" the data. As I don't see any error in the `encap` plugin. All Debug I done presents correct values in the code. But the **output** packet in the file is incorrect. That makes me think it's another part of the code that's overwriting something.

Please, can you check the code?
I hope you can identify the cause.
Regards.

lars18th, 2018-11-15:
Hi,

Regarding the value "4" for the initial pointer, it is in consistence with the data in the second packet. However, the start of the "real" data in the first packet is not in byte[8] (a value of "4" is ok in this case); but in byte[12].

So, two inconsistences appear:
1. Why the first pointer is "4" instead of "0"?
2. And why the data of the first inner packet starts at position 12 instead of 8? Even if we accept that it's valid a "4" value for the initial pointer.

I can't found a valid response for this behaviour. :interrobang:

lars18th, 2018-11-16:
Hi @lelegard ,

I'm tired with this bug. :unamused: 

See this code:
```
            // The output packet shall be exactly full.
            assert(pktIndex == PKT_SIZE);
            std::cerr << " pkt.b[4]== " << std::hex << static_cast<int16_t>(pkt.b[4]) << std::dec << "\n";
            pkt.b[4] = uint8_t(0x88);
            std::cerr << " pkt.b[4]== " << std::hex << static_cast<int16_t>(pkt.b[4]) << std::dec << "\n";
        }
```
That's correspond with
https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsPacketEncapsulation.cpp#L328

So, if I compile and execute this code with the "fail" test, then the byte[4] of all outer packets has the value `0x88`. However, if I comment out the line ` pkt.b[4] = uint8_t(0x88)`, then even if the print console shows values like `0x00` & `0xb3`. **EVEN IN THIS CASE** the *first* packet has byte[4]==`0x04` and the *second* byte[4]==`0x08`. After that all following packets show the correct values.

Any ideas?

lars18th, 2018-11-16:
Hi @lelegard ,

Please merge this PR #167 to fix the problem.

Regards. :wink:

lelegard, 2018-11-16:
Hi @lars18th,

I am sorry that you spent so much time on this problem. I hope that I am not wrong but, after a careful analysis of the problem, it seems that there is no bug and the fix is the PR is not pertinent. Let me explain this in details.

First, the symptom you observe occurs only when you use the plugin `svrename` _after_ the plugin `encap`. Whenever you remove the plugin `svrename`, the first output packet in PID 8032 always starts with an inner packet (PUSI = 1, pointer field at offset 4 = 0), with or without option `--pack`, regardless of `--add-input-stuffing`.

So, somehow, the problem is linked to the presence of plugin `svrename`. Of course, we may imagine nasty random memory corruption as we sometimes see in complex systems. I do not want to say that this is impossible with TSDuck but the stringent design and development rules I use should prevent this and, in practice, I never got that kind of bugs with TSDuck.

Sadly, the problem is much simpler. This is the normal behavior of `svrename`. All plugins which globally change the signalization nullify the first input packets until they know exactly what to do. For instance, when you modify something in a service, you may have to change the PMT. But, as long as you did not get the PAT (and SDT if the service is known by name), you do not know on which PID the PMT is. So, to avoid letting an old version of the PMT pass by accident, the plugin nullifies all initial packets until the PMT PID is known. This is just an example. But the same principle applies to several plugins: in the initial service discovery phase, if there is a risk to propagate invalidated information, all packets are nullified.

This is what happens here. I tried the command adding `-P file` before and after `svrename`. Then, I extracted the PID 8032 in these two files and compared the result with `tscmp`. Try this and you will see that exactly the first 39 packets of PID 8032 are deleted by `svrename`. All subsequent packets are present and identical.

So, what you _think_ is the first packet of PID 8032 is just the first _final output_ packet of this PID. But this is in fact the 40th packet which was generated by `encap`. And the value 4 that you see simply indicates that there are the 4 last bytes of a previous inner packet.

But this is normal, this works and the `decap` will work properly thanks to the PUSI + pointer filed synchronization mechanism. Now, you can see why it was important to have both the PUSI and pointer field.

This explanation was a bit long but, given the time you spent on this false problem, you deserved it.

lars18th, 2018-11-17:
Hi @lelegard ,

Thank you for your explanation. Now, I know **where** is the problem.

However, please don't reject the PR #167 and merge it.
Why you _like_ to do it?
1. It overcomes the problem, so no more 0x04 in the first pointer with not apparant cuase.
2. It not changes or breaks the algorithm, so after initial loops the value it's set to "1" and **never** returns to "0" (please, note that NEVER is _never forever_; as last line of function fillPacket() will imposes it).
3. And, and this is the most relevant, when implementing some future changes in the `encap` or `decap` plugin this "fix" will assure a more consistent output.

So, with all the respect! I expend a lot of time because this, and I don't see any _real_ cause to not merge the change. So, please (I repeat)...

> Please merge this PR #167 to fix the problem.

lelegard, 2018-11-17:
Hi @lars18th,

My guess is that the commands that you tried with and without the patch were slightly different (in files, options or initial conditions). Because the effect of the patch is ... __nothing at all__. It does not change anything in the processing.

When I first read the modification, I was surprised because it catches an impossible condition. I read the rest of my code one more time in case I missed something. But nothing. So, I just made a precise test:

- Based on the latest code in the repo.
- Build a script using the same command you provide (the one with `--add-input-stuffing 1/60`); just adding a `-P file` after `-P encap` in order to get two files: before and after `svrename`.
- Run the script on the latest code.
- Rebuild with your patch (just adding `if (_lateIndex < 1) _lateIndex = 1;`).
- Re-run the script.
- Compare the two pairs of files, with and without the patch.
- Each pair of file is exactly identical => the patch did not change anything (not surprising to me but needed to be proved by experiment).

Again, the 4 bytes you are seeing are normal and __necessary__. Don't try to delete them ! They are part of a previous inner packet. The fact that you cut the outer PID in the middle (because `svrename` nullifies the initial sequence of the stream) does not change anything.

Remember that, in the general case, a transport stream has no beginning and no end. This is a broadcast format and broadcast is a continuous process. You never _start_ a stream, you _jump into the middle_ of a continuous stream. So, you jump into the middle of a video PID, a section stream or, here, an outer encapsulation stream. The first packet you see on each PID (video, section, whatever) is typically partly or entirely useless. Each packetization format (PES, section, T2-MI, TSDuck `encap`) is designed to detect this and properly resynchronize as soon as possible. In the case of your example, when you jump into the outer PID, you start with a packet the first 4 bytes of which are useless _in this context_. Fine. But if you jumped simply one packet sooner in that PID, you would have got the start of the previous inner packet and the 4 bytes you name useless would have become useful.

This is broadcast, this is digital TV, no more, no less.

So, I am sorry, but I will not merge the patch because it is useless. Nothing personal or emotional here, just rational analysis.

lars18th, 2018-11-17:
Hi @lelegard ,

You explanation is good. And I accept and agree with it!
In addition, I don't want to be disrespectful or controversial. My objective is only to collaborate. :smile: 

> This is broadcast, this is digital TV, no more, no less.

Yes. Correct! I agree! :+1: 

> Nothing personal or emotional here, just rational analysis.

Me too! :smile: 
My problem, it's because I'm doing some enhancement of the `encap` and `decap` filters (I put the PR soon). And in my case the current behaviour is a problem. Therefore, I need to ensure  that _any_ new packet starts at the boundaries. so it's not about a bunch of stubbornness. It's a requirement for some processing. And as you say...

> Because the effect of the patch is ... nothing at all.

So, as the patch **doesn't break anything**. Please, review the _second version_. #168

I hope you like to consider it. :wink:

lelegard, 2018-11-17:
Just looked at #168. But I see no improvement either. The effect is the same as #167: no change at all in the output. I see no other improvement in the code quality. I think it is time to move forward to something else.

lars18th, 2018-11-17:
> I think it is time to move forward to something else.

OK. I accept it (but with some objection, sorry for it!).

Therefore, I'll include it in my enhanced version of `encap` and `decap`. I'm sure you will see the requirement -and the difference- in this case.

Be on hold!
See you later. :wink:

closed 2018-11-17 by lars18th