tsduck/tsduck #173: Building with OpenEmbedded

athoik, 2018-11-21:
Hi @lelegard,

Following the changes in #172 I will post details about the openembedded build here.

We are using openembedded to cross compile several images for machines that make use of linux dvb at @OpenPLi, @SatDreamGr, @oe-alliance, etc. One of the utilities we do use for checking TS, frontends is `dvbsnoop` but is not actively maintain any more. So having a tool to better analyze stuff is great.

Several architectures are build like mipsel, arm, aarch64 and few more.

Output from `uname` for the most popular architectures follows:

- mipsel box:
```
root@osnino:~# uname -a
Linux osnino 4.8.17 #2 SMP Sun Jan 14 18:59:46 EET 2018 mips GNU/Linux
root@osnino:~# uname -m
mips
```
- arm box:
```
root@vusolo4k:~# uname -a
Linux vusolo4k 3.14.28-1.8 #1 SMP Mon Apr 9 17:15:11 EEST 2018 armv7l GNU/Linux
root@vusolo4k:~# uname -m
armv7l
```
- aarch64 box:
```
root@osmio4k:~# uname -a
Linux osmio4k 4.19.0 #2 SMP Thu Oct 25 19:16:24 EEST 2018 aarch64 GNU/Linux
root@osmio4k:~# uname -m
aarch64
```

The mipsel box requires an extra patch in tsPlatform just like we did #172, I will create a pull request shortly.

Both arm7l and mips are failing due to [`-Werror=cast-align`](https://github.com/tsduck/tsduck/blob/master/Makefile.common#L793). So I had to disable it (it was faster) in order to build for those architectures. 

If there is a way to properly solve it, without hardcoding all possible architectures causing problem with cast-align would be great.

```
  [CXX] tsAACDescriptor.cpp
In file included from tsByteBlock.h:36:0,
                 from tsAbstractSignalization.h:36,
                 from tsAbstractDescriptor.h:36,
                 from tsAACDescriptor.h:36,
                 from tsAACDescriptor.cpp:34:
tsPlatform.h: In function 'void ts::PutUInt40BE(void*, uint64_t)':
tsPlatform.h:1935:67: error: cast from 'uint8_t* {aka unsigned char*}' to 'uint32_t* {aka unsigned int*}' increases required alignment of target type [-Werror=cast-align]
         *(reinterpret_cast<uint32_t*>(static_cast<uint8_t*>(p) + 1)) = CondByteSwap32BE(static_cast<uint32_t>(i));
                                                                   ^
tsPlatform.h: In function 'void ts::PutUInt48BE(void*, uint64_t)':
tsPlatform.h:1970:67: error: cast from 'uint8_t* {aka unsigned char*}' to 'uint32_t* {aka unsigned int*}' increases required alignment of target type [-Werror=cast-align]
         *(reinterpret_cast<uint32_t*>(static_cast<uint8_t*>(p) + 2)) = CondByteSwap32BE(static_cast<uint32_t>(i));
                                                                   ^
tsPlatform.h: In function 'void ts::PutUInt48LE(void*, uint64_t)':
tsPlatform.h:1993:67: error: cast from 'uint8_t* {aka unsigned char*}' to 'uint16_t* {aka short unsigned int*}' increases required alignment of target type [-Werror=cast-align]
         *(reinterpret_cast<uint16_t*>(static_cast<uint8_t*>(p) + 4)) = CondByteSwap16LE(static_cast<uint16_t>(i >> 32));
                                                                   ^
cc1plus: all warnings being treated as errors
../../Makefile.common:900: recipe for target 'release-mips32el/tsAACDescriptor.o' failed
make[2]: *** [release-mips32el/tsAACDescriptor.o] Error 1
make -C build install
install -d -m 755 /opt/build/tmp/work/mips32el-oe-linux/tsduck/3.16+gitrAUTOINC+ca68df619e-r0/image/etc/udev/rules.d
install -m 644 80-tsduck.rules /opt/build/tmp/work/mips32el-oe-linux/tsduck/3.16+gitrAUTOINC+ca68df619e-r0/image/etc/udev/rules.d/
install -d -m 755 /opt/build/tmp/work/mips32el-oe-linux/tsduck/3.16+gitrAUTOINC+ca68df619e-r0/image/etc/security/console.perms.d
install -m 644 80-tsduck.perms /opt/build/tmp/work/mips32el-oe-linux/tsduck/3.16+gitrAUTOINC+ca68df619e-r0/image/etc/security/console.perms.d/
```` 

Also something I considered strange was the build continuing and installing files under /etc. With above make command, an empty package created although build failed. 

More over, it would be nice when running make without target to make the build only (openembedded first calls make without any parameters on do_compile and later on do_install is calling with parameter install).

Additionally the --help flag seems to use flags not compatible with busybox `less`, so I guess we should build the full `less` instead of using busybox one. 

```
# tsp --help
/usr/bin/less: invalid option -- 'Q'
BusyBox v1.24.1 (2018-09-06 16:36:47 EEST) multi-call binary.

Usage: less [-EIMmNSh~] [FILE]...

View FILE (or stdin) one screenful at a time

        -E      Quit once the end of a file is reached
        -I      Ignore case in all searches
        -M,-m   Display status line with line numbers
                and percentage through the file
        -N      Prefix line number to each line
        -S      Truncate long lines
        -~      Suppress ~s displayed past EOF
```

Once again thanks for your great utility, it's really the dvb swiss army knife 🥇. I already love the tsanalyze command.

athoik, 2018-11-21:
Below you can also find the automatically created from openembedded build scripts for arm,aarch and mipsel.

[run.do_install.arm.txt](https://github.com/tsduck/tsduck/files/2605462/run.do_install.arm.txt)
[run.do_install.mipsel.txt](https://github.com/tsduck/tsduck/files/2605463/run.do_install.mipsel.txt)
[run.do_install.aarch64.txt](https://github.com/tsduck/tsduck/files/2605467/run.do_install.aarch64.txt)

lelegard, 2018-11-21:
Hi @athoik 

Thanks for your efforts. Testing in all embedded environments is quite challenging and need a collective effort.

The option `-Wcast-align` already created errors with ARM (32 bits) and was explicitly disabled on it. It could be useful on Intel only. However, since it also creates issues with other embedded architectures and I don't really see the added value (other than performances on RISC architectures with strict memory access alignment constraints), I simply removed it.

Concerning the issue with `--help` (and a few other options), TSDuck is using the most appropriate pager for the operating system. If `less` is found, it is used. However, it is assumed to be a standard GNU `less`, not busybox. Any idea on how to detect that this is a busybox version ?

I will investigate the failing `make`  issue. A `make install` is run once in `src` (failing here) and once in `build` (where it installs some system configuration files). I will check why the make isn't interrupted when the first command fails.

lelegard, 2018-11-21:
> More over, it would be nice when running make without target to make the build only

I don't get the issue here. This is what `make` alone is doing, building all binaries and unitary tests only. What do you think it does?

athoik, 2018-11-21:
> I don't get the issue here. This is what make alone is doing, building all binaries and unitary tests only. What do you think it does?

You are right, it works fine. 

The "out of tree build" is failing, then switch to "in tree build", and .gitignore was hiding the build objects, so make had everything build, nothing to redo.

Sorry for the false alarm.

athoik, 2018-11-22:
> Any idea on how to detect that this is a busybox version ?

Maybe we can check if less is link, resolve the link and check if contains  busybox?

```
# ls -al /usr/bin/less
lrwxrwxrwx    1 root     root            19 Jan  1  1970 /usr/bin/less -> /bin/busybox.nosuid
```

lelegard, 2018-11-22:
> Maybe we can check if less is link, resolve the link and check if contains busybox?

Implemented in the latest code. If a pager is detected to link to busybox, all options are dropped and the pager is used with its default settings. Please test.

athoik, 2018-11-22:
@lelegard it works great! Thank you!

Eg.

```
# tslsdvb --help
List DVB tuner devices

Usage: tslsdvb [options]
....
....
```

lelegard, 2018-11-23:
Let me redefine what TSDuck is. TSDuck is designed to process complete transport streams for the various purposes of lab, test, demo, debug, integration, etc. TSDuck is also designed to be portable and provide the same level of service on all platforms. The only inevitable limitations are defined by the level of hardware support on some platforms (namely, no Dektec, HiDes and tuners on macOS).

Concerning the write access : if I remember correctly (this code was write ~ 12 years ago), it is necessary to send tune requests to the frontend.

Concerning the 0x2000 PID : All tuner drivers I used in the last 13 years on Linux had no problem with this. An STB is not a general purpose computer. I understand that, inside an STB hardware, a regular middleware only needs to read the PSI/SI and a given number of specified services at the same time. If a given tuner/driver which is used exclusively inside STB hardware cannot read the complete TS, my conclusion is that this is not a valid platform for TSDuck.

If you look at the code, you will see that making the class `ts::Tuner` portable across Linux and Windows was a challenge. This class is made to tune to a given frequency with some modulation parameters and read the complete transport stream. I will not add complexity to the most complex class in TSDuck to support a few dedicated STB platforms which are not part of the TSDuck target.

In other words, TSDuck is not supposed to replace official STB middleware.

lars18th, 2018-11-23:
Hi @lelegard ,

Concerning the 0x2000 PID, just my opinion (if you want to hear it): Not only STB platforms have limitations to read the full TS. In Linux/Windows a not negligible list of tuners has this limitation. In addition, some networkd tuners (SAT>IP for example) may also have this limit. I don't think it's a bad solution to do something like that:

```
  if ( set_pid_0x2000() < 0 ) {
     print_warning("Full TS not supported");
     set_pid_filter_range(0x0,0x20);
  }
```
OR... get the filter range from the command line, instead of always request the full TS.

However, it's only my personal opinion.

lelegard, 2018-11-23:
There is a wide variety of cheap USB or PCI tuners for Windows and Linux. We cannot test all of them. The TSDuck user's guide contains a table of tested tuners, either by me or other users. As of today, I never had the problem of not being able to read the complete transport stream from a tuner. The worst problem I had was a PCTV DVB-S2 Stick 461e (DVB-S/S2, USB) which returns everything except the stuffing PID. After I realized this, I never used it again since evaluating the stuffing is  important in broadcast for monitoring or to keep the appropriate bitrate for the target modulation.

Of course, I do not pretend that tuners not being able to get the complete TS do not exist. I just want to say that such tuners are only good enough to watch TV but are useless in a lab environment or field testing. Given the cheap price of most tuners, if you got one of those, just buy another one.

athoik, 2018-11-23:
> No need to change something, just sharing my experience

All work fine, I can still analyze recordings, ts files, ts http stream etc.

So at the moment, I can build tsduck using openembedded without any major issue.

The only trivial issue was build not failing and producing an empty package, but I can live with that.

Thank you!

rezna11, 2018-11-26:
> In other words, TSDuck is not supposed to replace official STB middleware.

But it would be very cool :) There are these Linux STBs with two dual FBC tuners. Then you can connect one Quad or Quattro LNB and get 16 demodulators, covering any 16 transponders (4 from each polarization/band) with just one STB. 

The problem is that there's no software on these set-top-boxes that would make any use of such interesting feature. I can't imagine anything better than TSDuck to do this. Tune in 16 transponders, filter my favotite channels, send out as IP multicast (these STBs usually have gigabit network). 

Do I understand right that the problem now is that TSDuck can't tune any transponder because PID Filter 8192 is not supported by these boxes?

lars18th, 2018-11-26:
Hi @rezna11 ,

I agree with the idea that TSDuck is cool!
And I think it can work in any embedded platform without any problem.

However, It's true that TSDuck can't do all. And the problem here is **the lack to full TS support in a lot of hardware tuners**. Not all hardware can handle it. So, the problem is not the driver support, but the hardware.

In addition, the mode as the TSDuck handles the TS requires the full TS. So, for this reason @lelegard doesn't target tuners without 0x2000 pid support.

But... I need to comment this: If you use _any_ other tool to do the tunning and capture of the TS. And after you pass the TS to TSDuck, then it can work. Why? Because you can apply some pid filtering in the capture tool. Perhaps TSDuck can't do all functions, but any that doesn't require the full TS will work.

After say that, I suggest to review the list of tools that capture a TS, and see if it is possible to add an option to specify a list of pids to capture. I'm sure that with this small change all people will be happy.

Just my five cents of my personal opinion! :smile:

lelegard, 2018-11-28:
AFAIK, most - if not all - USB/PCI tuners which work on a computer platform can get the complete TS, which is the usual input of TSDuck.

I am assuming here that tuners with this kind of limitation are more frequently - if not exclusively - found on open set-top boxes. 

So, let me clarify my position here and please read carefully. TSDuck is designed to support various lab activities (debug, integration, demo, test, etc). I see no _legitimate_ use of an open set-top box. I repeat, no __legitimate__ use. I am sure you understand what I mean.

closed 2018-12-01 by lelegard

lars18th, 2018-12-13:
> So, let me clarify my position here and please read carefully. TSDuck is designed to support various lab activities (debug, integration, demo, test, etc). I see no _legitimate_ use of an open set-top box. I repeat, no **legitimate** use. I am sure you understand what I mean.

Hi @lelegard ,

I understand perfectly what you're pointing out!

However, there are two things I would like to comment on:

1. _Limited_ tuners are not only inside set-top boxes, but inside SAT>IP servers too.
2. In DIY research it's common to use low-cost hardware devices.

After saying this, I want to add that in my research environment almost all _TUNER_ devices are network devices (STBs with SAT>IP server function, Hardware SAT>IP servers, HDHomeRuns, etc.). So I don't see why is not legitimate to use such devices as the input tuner for TSDuck. For example, using the well known vtuner satip client in a Linux box you can emulate an internal DVB Tuner (compatible with TSDuck) using a remote SAT>IP server. So what's the problem? That many of them doesn't support the full TS. In this case, a _simple_ (and I want to enphasize the word **simple**) of just adding a pid_range option to avoid the lack of 0x2000 PID I feel is not a bad idea.

This is just my opinion. I hope you can understand my position. And may it not bother you.