tsduck/tsduck #175: HLS input problem

anatolijv, 2018-11-23:
When running tsp with hls input, tsp gets from 2 to 4 chunks of strem, and exits with.

`debug level set to 1
* Debug: debug level set to 1
trying to load /usr/bin/tsplugin_hls.so
symbol tspNewOutput not found in /usr/bin/tsplugin_hls.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_hls.so
trying to load /usr/bin/tsplugin_ip.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_ip.so
trying to load /usr/bin/tsplugin_continuity.so
symbol tspNewInput not found in /usr/bin/tsplugin_continuity.so
symbol tspNewOutput not found in /usr/bin/tsplugin_continuity.so
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: hls: debug level set to 1
* Debug: continuity: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading http://api.tv.ipnet.ua/api/v1/manifest/1293299250.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.10.3 (Ubuntu)
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:09 GMT
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: HTTP header: Content-Length: 289
* Debug: hls: announced content size: 289 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Allow: GET, HEAD, OPTIONS
* Debug: hls: HTTP header: X-Frame-Options: SAMEORIGIN
* Debug: hls: HTTP header: Vary: Cookie
* Debug: hls: MIME type: application/x-mpegurl
* hls: downloaded 1293299250.m3u8, master playlist, static, 1 media playlists
* hls: selected playlist: http://82.193.96.98/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE1NDMwNTEwMjksInRpbWUiOjE1NDI5NjQ2MjksImNoYW5uZWwiOjEyOTMyOTkyNTAsIm1hYyI6ImJyb3dzZXJfNDYyNzQwODYifQ.zGkH2n2ihoc4xN46qWEPnv0Z_xfIKmfJ9xL_JXchQsM/index.m3u8, 1,000,000 b/s
* Debug: hls: downloading http://82.193.96.98/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE1NDMwNTEwMjksInRpbWUiOjE1NDI5NjQ2MjksImNoYW5uZWwiOjEyOTMyOTkyNTAsIm1hYyI6ImJyb3dzZXJfNDYyNzQwODYifQ.zGkH2n2ihoc4xN46qWEPnv0Z_xfIKmfJ9xL_JXchQsM/index.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:09 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 331
* Debug: hls: announced content size: 331 bytes
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* hls: downloaded index.m3u8, media playlist, updatable (live), 4 segments, 6 seconds/segment
* Debug: hls: internal push-input thread started
* hls: downloading from http://82.193.96.98:9000/hls/1293299250/00993.ts
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:09 GMT
* Debug: hls: HTTP header: Content-Type: video/mp2t
* Debug: hls: HTTP header: Content-Length: 3645320
* Debug: hls: announced content size: 3645320 bytes
* Debug: hls: HTTP header: Last-Modified: Fri, 23 Nov 2018 09:16:45 GMT
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: ETag: "5bf7c57d-379f88"
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: initial buffer load: 4 packets, 752 bytes
* hls: unknown input bitrate
* Debug: hls: input thread started
* Debug: continuity: packet processing thread started
* Debug: ip: output thread started
* hls: downloading from http://82.193.96.98:9000/hls/1293299250/00994.ts
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:10 GMT
* Debug: hls: HTTP header: Content-Type: video/mp2t
* Debug: hls: HTTP header: Content-Length: 3617308
* Debug: hls: announced content size: 3617308 bytes
* Debug: hls: HTTP header: Last-Modified: Fri, 23 Nov 2018 09:16:53 GMT
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: ETag: "5bf7c585-37321c"
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: HTTP header: Accept-Ranges: bytes
* hls: downloading from http://82.193.96.98:9000/hls/1293299250/00995.ts
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:10 GMT
* Debug: hls: HTTP header: Content-Type: video/mp2t
* Debug: hls: HTTP header: Content-Length: 3627272
* Debug: hls: announced content size: 3627272 bytes
* Debug: hls: HTTP header: Last-Modified: Fri, 23 Nov 2018 09:17:01 GMT
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: ETag: "5bf7c58d-375908"
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: downloading http://82.193.96.98/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE1NDMwNTEwMjksInRpbWUiOjE1NDI5NjQ2MjksImNoYW5uZWwiOjEyOTMyOTkyNTAsIm1hYyI6ImJyb3dzZXJfNDYyNzQwODYifQ.zGkH2n2ihoc4xN46qWEPnv0Z_xfIKmfJ9xL_JXchQsM/index.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:11 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 331
* Debug: hls: announced content size: 331 bytes
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: no new segment in playlist
* hls: downloading from http://82.193.96.98:9000/hls/1293299250/00996.ts
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:11 GMT
* Debug: hls: HTTP header: Content-Type: video/mp2t
* Debug: hls: HTTP header: Content-Length: 3636672
* Debug: hls: announced content size: 3636672 bytes
* Debug: hls: HTTP header: Last-Modified: Fri, 23 Nov 2018 09:17:09 GMT
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: ETag: "5bf7c595-377dc0"
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: downloading http://82.193.96.98/eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJleHAiOjE1NDMwNTEwMjksInRpbWUiOjE1NDI5NjQ2MjksImNoYW5uZWwiOjEyOTMyOTkyNTAsIm1hYyI6ImJyb3dzZXJfNDYyNzQwODYifQ.zGkH2n2ihoc4xN46qWEPnv0Z_xfIKmfJ9xL_JXchQsM/index.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.12.1
* Debug: hls: HTTP header: Date: Fri, 23 Nov 2018 09:17:11 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 331
* Debug: hls: announced content size: 331 bytes
* Debug: hls: HTTP header: Connection: close
* Debug: hls: HTTP header: Access-Control-Allow-Origin: http://online-tv.ipnet.ua
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: no new segment in playlist
* hls: HLS playlist completed
* Debug: hls: internal push-input thread completed
* Debug: continuity: packet processing thread terminated after 77,269 packets, 77,269 passed, 0 dropped, 0 nullified
* Debug: hls: input thread terminated after 77,269 packets
* Debug: ip: output thread terminated after 77,269 packets (77,269 output)
* Debug: Report logging thread terminated
`

lelegard, 2018-11-23:
Hi @anatolijv 

Thank you for reporting this. It was a timing issue which depends on the way the server produces updates of the HLS live media playlist. It is fixed now. Could you pull, rebuild and test ?

More precisely, in live streams, the media playlist is regularly updated with new segments. If the existing segments are downloaded faster than the playlist update interval, the `hls` plugin reached an end of stream. This highly depends on the server response time. In my tests on the URL you specified, the number of retrieved segments was never the same, between 2 and 10 typically.

This was a bug in the `hls` plugin, it was not tolerant enough with the server. Now, when no new segment is available at end of all downloads, instead of returning an end of stream, the plugin retries the playlist updates up to the estimated wallclock time of the end of the downloaded segment playout time.

Ironically, the IPnet HLS server does not have a very secure way to use geo-protection. When I tried your command, on the master playlist, my request was rejected with "403 Forbidden". Since I am located abroad, this is acceptable. However, if I access the media playlist as displayed by the debug trace, there is no restriction. At least this allowed me to do my tests.

Please close the issue when you consider it solved.

mhzbg, 2018-11-24:
Hello Guys, 
can someone post the full command to receive HLS and output to a transport stream, possibly creating a DVB compliant one that can be fed to receivers ? A bonus will be to insert EIT from an XML  Thanks

lelegard, 2018-11-24:
Hi @mhzbg 

How could anyone post a "full command" when you do not even specify how the stream "can be fed to receivers" ? DVB-S, DVB-T, Dektec or HiDes modulator, UDP/IP ?

More generally, TSDuck is a flexible toolbox for TV engineers, not a pre-packaged application for production. So, there is no definitive "full command" for a given purpose. The best we can do here is to provide sample template commands and then you need to read the documentation and adapt the command to your specific need.

Such a template could be
~~~
tsp -a 1/10 \
    -I hls playlist-url \
    -P inject -p 18 -b 10000 -s eit.xml \
    -P filter -n -p 0x1FFF \
    -O IP address:port
~~~
I let you read the documentation to understand the sample options so that you can adapt them to your need.

If you want a reusable and flexible command, you should write a script where all your specific cases and your application logic are coded, using TSDuck for the plumbing.

mhzbg, 2018-11-24:
Hello @lelegard,
first of all thanks for your prompt and sharp responses and the great utility and documentation you have put together, Its is indeed one of the best maintained and documented tools I have seen. 
You are right indeed, that this is a very broad question and there is a great amount of documentation. 
I can assure you I always have the latest PDF open and browse constantly search trough it. 

The input and the output plugins are always the easy part compared to the filter/plugin chain (not the right term perhaps but you get the idea). Manipulation of the streams is always where one needs a little help vs the input and the output, specially when both are IP. 
Generally I always spend the most time in the examples section which is the most useful one provided there is an example for what one is looking for. Perhaps we can start and example wiki or expand the examples section, it gets newbies like me going quickly and also shows the power of the tools. 

Thanks again, 
cheers

lelegard, 2018-11-25:
The Wiki exists (see the link at the top of the page). It was created upon request from a user who wrote a few pages. It is open for writing by experienced users.

closed 2019-01-02 by lelegard