tsduck/tsduck #176: running tsduck app in a container

fredino, 2018-11-29:
hey every body,

I'm just new in docker and tsduck enviroment,
I compiled the source code of tsduck in my container and I got all the binaries.
I want now to know how can I run my app inside the docker container or with another words , how can I check if my tsduck app is running inside my container or not ?

thank you for your help !

fredino, 2018-11-29:
thank you for your first

fredino, 2018-11-29:
thank you for your answer , from inside TSDuck utilities ?

closed 2018-11-29 by fredino

lelegard, 2018-11-29:
> from inside TSDuck utilities 

So, you would like to modify TSDuck and check whether you are running inside a container or not ? What is the purpose ? What kind of specific decision should TSDuck make if it is running inside a container ?

fredino, 2018-11-29:
actually after I compield the source code of tsduck I got the transport stream analysis report table . I want in the next step to use grafana dashboard so I can show that table in grafana, I mean to connect the table that I got to grafana dashboard by importing the ip address of my stream which I got .

lelegard, 2018-11-29:
If you want to import `tsanalyze` output inside a monitoring tool such as grafana, I recommend option `--normalized` which produces a predefined normalized output for use by applications or scripts.

If you want to write a script which uses `tsanalyze` and then checks the container, this is a not a TSDuck question, rather a Docker question.

If you want to build a specific application on top of transport analysis which can also check and manipulate the local container, you may use the [TSDuck C++ library](https://tsduck.io/doxy/) and specifically class [`ts::TSAnalyzerReport`](https://tsduck.io/doxy/classts_1_1_t_s_analyzer_report.html) to analyze a TS and produce a report.

fredino, 2018-11-29:
thank you very much for your infos :) , last question : do you have any idea how  can I transfer the the stream analysis report which I got to grafana dashboard, I mean there is any command which trigger another command on docker so I can do that ?

lelegard, 2018-11-29:
>  I mean there is any command which trigger another command on docker so I can do that ?

This is a docker question, we cannot help you here/

fredino, 2018-11-29:
thank you for your help and infos :)

themoog, 2021-04-06:
hi has anyone got an example Dockerfile for building / compiling from scratch?