tsduck/tsduck #178: MPE decapsulation

odoukara, 2018-12-05:
Hi Lelegard
I am getting a multicast MPE transport stream trhough the Ethernet port of a commercial encapsulator @ 224.2.2.2:5001. I am trying to decapsulate the stream with tsduck and forward the udp packet into the network. My command is like this : tsp -I ip 224.2.2.2:5001 -P mpe --udp-forward -O drop
When I sniff the network with wireshark, my streams remain as MPEG-TS. I was expecting to see udp packets after the decap. Is there anything wrong with my command ?
By the way I can decapsulate the same stream correctly with Novra S200 receiver.

lelegard, 2018-12-05:
You cannot see MPEG-TS packets with Wireshark. You see IP packets. Each IP packet has a UDP type and the UDP payload contains TS packets. This is exactly what MPE extraction & forwarding is supposed to do.

What did you expect? What does your receiver send over the network and how is this different from what you are seeing with TSDuck?

If you think that something is wrong, please post an extract of the TS with MPE.

odoukara, 2018-12-05:
Hi Lelagard, I just want to make sure I have the right command to decap and forward. The problem might come from my setup. Regarding wireshark, 
you have
 No. | Source | Destination | Protocol | Lengh | Info

Sniffing my interface before decapsulation, the protocol column will show MPEG-TS. Normal because I am really sending MPEG-TS stream with MPE at this stage.
Sniffing the same interface after decapsulation with Novra S200, the protocol column show UDP (No more MPEG-TS) and I can save my udp packets (I use udpcast client /server for handling my files).

Now, when I tsp -I ip 224.10.10.1:5001 -P mpe --udp-forward -O drop instead of my Novra S200, the protocol column remains MPEG-TS (I thought after decap I should get only udp and no more MPEG-TS) and I can't save my udp packets.

I am wondering if the decapsulation is really happening. Will dig deeper during the day and post my findings.

lelegard, 2018-12-06:
> I am wondering if the decapsulation is really happening. 

This is precisely what I explained. Your input is TS with MPE inside. If you see something on the IP network with Wireshark, then decapsulation has already happened, necessarily.

odoukara, 2018-12-07:
Hi @lelegard , here is a print screen of the streams I need to decap and forward. I will send a .pcapng file later.

[image]

lelegard, 2018-12-07:
Thanks for the screenshot.

There are two distinct UDP streams which are mixed in the network. Each UDP stream contains TS packets (inside UDP packets).

- UDP steam 1 (with expanded packet in the middle pane)
  - source 172.16.1.100:4009, destination: 224.10.10.1:5001
  - this stream comes from the Novra (from the OUI in the source MAC address)
- UDP stream 2 (UDP ports unknown since no packet is expanded)
  - source 172.16.1.2, destination: 224.2.2.2

The strange thing is the Wireshark output. In the top pane, the highlighted line is packet 33, source 172.16.1.2 (so, stream 2). But, in the middle pane, the output is "Frame 33" with source 172.16.1.100 (so, stream 1). There is an inconsistency here which is strange because Wireshark is usually an extremely robust tool.

I think that this is either a bug in Wireshark or a manual error. If you carefully in the middle pane, you will see that the frame has been rebuilt from 9 message fragments : #32(184), #32(184), ..., #33(184), #33(184). But this is incorrect because, when you look in the top pane, the frame #32 belongs to stream 1 while frame #33 belongs to stream 2 (not the same addresses).

- First possibility: Wireshark is incorrectly grouping Ethernet frames from distinct UDP streams to rebuild large messages and this is a Wireshark bug.
- Second possibility: the user explicitly grouped the two frames and this is a manual error.

In all cases, the rebuilt message in the middle pane is inconsistent because it is the aggregation of fragments from distinct streams.

odoukara, 2018-12-07:
@lelegard ,
lets close the issue for now. I will take more time to analyse what is happening. I think my set up is causing the problem.
Thanks

closed 2018-12-07 by lelegard