tsduck/tsduck #184: Ringbuffer File

ReubenM, 2018-12-21:
I thought I would leave a note here about a very useful linux kernel ringbuffer module. It was very useful in integrating tsduck tools and other tools in scenarios where using UDP multicast for a one-to-many stream distribution is not an option.

There is linux module that is quite old actually, called emlog. It's intended purpose is for embedded systems to write log output to. The ringbuffer is a fixed size character device that just loops over itself indefinitely. The size is 1024 bytes base size times whatever the assigned Minor device number is. Even with the Minor device number at 256, that's still too small to be very useful for video, but if you change the base size in the code to a MB, it can scale to sizes that work great for video. One writer process and any number of processes reading it like a normal file, but no worries about using up storage space.

https://github.com/nicupavel/emlog

lelegard, 2018-12-21:
Hi @ReubenM 

The ringbuffer is a very common design pattern. It has been used for decades, way before Linux existed and not only in embedded systems. In TSDuck, the [internal packet buffer](https://tsduck.io/doxy/devplugin.html#pdevbuffer) inside `tsp` is precisely such a ringbuffer with several concurrent threads (one per plugin) reading and writing it.

How would you like to use this pattern with TSDuck? For instance, I would understand the need for an output TS packet file the size of which would not exceed some predefined limit. However, while ringbuffers are very useful in memory, they are usually not such a good idea for files since the ringbuffer alone is not sufficient, you need to maintain the current wrap-up point in metadata and make sure that all applications handle it properly. With files, the _file rotation_ is a better pattern: close files when they reach a given limit, reopen a new one, purge old files (see the `logrotate` command on Linux as an example).

ReubenM, 2018-12-21:
I was aware of the ringbuffer design, I had just never seen it exposed as a character device supporting normal file IO.

My use case was that I found this useful to be able to provided a live encoded stream running continuously and outputting to the character device, and then have multiple instances of proprietary software that only support file IO to play from the character device, and disconnect / reconnect as needed without interrupting the live source feed. This only works if the software reading just tries to start playing without trying to load the file into memory or seek in the file past what is needed for realtime playback. When both the writing and reading are at the same speed, the wrap point isn't crossed. 

I wonder how difficult it would be to modify it so that when file IO is started, the fd starts reading a stub header with packets describing the stream, and then jumps to a read pointer in the ringbuffer that has its location tracking the write pointer in a way so that it's always pointing to a sync byte. (or whatever the most ideal entry point into a stream is)