tsduck/tsduck #186: Extended Descriptor: audio_preselection_descriptor 0x19

athoik, 2018-12-30:
Hi, 

Currently tsduck doesn't handle audio_preselection_descriptor documented on [DVB BlueBook
A038](https://www.dvb.org/resources/public/standards/a038_dvb_spec_december_2017.pdf).

Here is a full transponder TS from [19.2E 10993H](https://mega.nz/#!XAE2XSaS!_9jA28NacfQSiMwZn2J-JY5h4Pfv3E7pGbNH_EmbLBY), recorded with test_dvr using all pids discovered via dvbsnoop pidscan. The descriptor is available on SES UHD Demo Channel, PMT PID 97.

```
    Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 513 (0x0201)
    - Descriptor 0: Stream Identifier (0x52, 82), 1 bytes
      Component tag: 8 (0x08)
    - Descriptor 1: Extension Descriptor (0x7F, 127), 3 bytes
      Extended descriptor: AC-4 (0x15, 21)
      Dialog enhancement enabled: 0, channel mode: 0x01 (Stereo)
    - Descriptor 2: Extension Descriptor (0x7F, 127), 7 bytes
      Extended descriptor: unknown (0x19, 25)
      0000:  08 01 08 64 65 75                          ...deu
```

Thanks!

lelegard, 2019-01-02:
Done in commit 2313796cad325aaedaa09f01a61238a4d94b6020

Thank you for pointing out this draft. I assume, however, that there will be modifications before it is merged into EN 300 468. See the "sequential" allocation of new descriptor tag extension values (new values since EN 300 468 v1.15.1, the latest one to my knowledge) :

Descriptor | tag ext.
--- | ---
S2X_satellite_delivery_system_descriptor | 0x17
protection_message_descriptor [48] | 0x18
audio_preselection_descriptor | 0x19
TTML_subtitling_descriptor [56] | 0x20

`0x20` following `0x19` 😢

athoik, 2019-01-02:
Well I guess it will become available here: [DVB Standards](https://www.dvb.org/standards).

Later will check new version and report.

Thank you!

athoik, 2019-01-02:
Output from tstables now is perfect!  👍

```
    Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 513 (0x0201)
    - Descriptor 0: Stream Identifier (0x52, 82), 1 bytes
      Component tag: 8 (0x08)
    - Descriptor 1: Extension Descriptor (0x7F, 127), 3 bytes
      Extended descriptor: AC-4 (0x15, 21)
      Dialog enhancement enabled: 0, channel mode: 0x01 (Stereo)
    - Descriptor 2: Extension Descriptor (0x7F, 127), 7 bytes
      Extended descriptor: Audio Preselection (0x19, 25)
      - Preselection id: 0
        Audio rendering indication: 1 (Preferred reproduction channel layout is stereo)
        Audio description: no
        Spoken subtitles: no
        Dialogue enhancement: no
        Interactivity enabled: no
        Language code: "deu"
```

Thank you!

closed 2019-01-02 by athoik