tsduck/tsduck #188: Suggestion: HLS (output) plugin

lars18th, 2018-12-31:
Hi,

The current HLS-input plugin is a great tool. However, it will be interesting to have another HLS-output plugin. I know about the complexity of it. But perhars a simple implementation will be possible. For example:

- Only for SPTS streams.
- Writes only files to the filesystem (so the HTTP server part isn't required).
- With relaxed restrictions: for example, the length of each chunk can be variable, based only in GOPs.

And any other limit that simplifies the implementation.
What you think?

lars18th, 2019-01-07:
> Hi,
> 
> A tentative implementation has just been added but not yet fully tested. Please review it and test. Feedback will be welcome.

Thank you!
I'll test it when I have more time.

joolzg, 2019-01-13:
Just tested and each scenario i  came up with the solution worked great.

Thanks, the more additions/fixes you make the less i need to use other multiple tools. Just continuous PCR needed now

closed 2019-01-13 by lelegard