tsduck/tsduck #189: Request: IP (input/output) with FEC support

lars18th, 2018-12-31:
Hi,

Another request for an enhancement. Please, consider it only if you found it feasible:

- The current implementation of the IP unicast/multicast modules are very powerful. However, they lack support for FEC (aka Pro-MPEG error correction). This FEC encapsulation is based on simple RTP sub-streams with PARITY data.
- Futhermore you can see one SEND implementation in FFmpeg: https://github.com/FFmpeg/FFmpeg/blob/master/libavformat/prompeg.c
- And one RECEIVE implementation in Prompeg-decoder: https://github.com/cy91244/prompeg-decoder

My goal is this: For a very CPU consuming task (multiple realtime transconding), I like to use multiple machines for processing the same MPTS. The process will be controlled by local instances of TSDuck. And I like to overcome any glitch in the network. So one _simple_ solution to make the streaming more robust is to add simple FEC codes in all multicast streams. So, as the TSDuck supports RTP for input/output then it will be good if the FEC data is supported.

I hope you think that's a good idea.
Regards.

mhzbg, 2019-05-31:
upvoting this. 
Also if I may add, this is now becoming an industry standard by SMPTE 2022 and supported by encoders such as Elemental and encapsulators such as  Zixi.

bbgdzxng1, 2019-08-21:
In case it is of any use... Here's another implementation of FEC, this time for Gstreamer.  https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/blob/92138dc3d6770c2ea269b8bf826dfd98cbcc95eb/gst/rtp/gstrtpulpfecdec.c

lars18th, 2019-12-09:
Hi,

Some user called @jcekstrom-xumo has implemented the Output FEC support:
https://github.com/jcekstrom-xumo/tsduck/commit/8fbe85a4d22ed396cf982c8bdc68345e01d03422

I hope this user will provide some PR, or @lelegard will want to contact him to incorporate his implementation.

lelegard, 2019-12-10:
Thanks @lars18th, this is quite interesting.

@jcekstrom-xumo, is your implementation of ProMPEG FEC correction fully functional? If yes, it would be beneficial for everyone to integrate it in the master code base. Please submit a pull request when you consider your implementation as ready.

MathieuDuponchelle, 2020-10-07:
https://gitlab.freedesktop.org/gstreamer/gst-plugins-good/-/merge_requests/753 in case that's useful

lelegard, 2020-10-07:
Thanks all for your pointers, especially on gstreamer. But remember that gstreamer is released under the GPL license and is hereby incompatible with the BSD-2 license of TSDuck. So, code from gstreamer cannot be included here. GPL code has a "Berlin Wall" mindset where everything that is on one side of the wall shall not mix with the other side. Far from the original intent, this significantly reduces the freedom to use the code. As an example, [libsrt](https://github.com/Haivision/srt), now included in TSDuck, after a brief start with GPL, quickly switched to MPL after realizing that GPL would significantly impair its deployment. So, if anyone is willing to write a BSD-compatible implementation of Pro-MPEG for integration in TSDuck, you are welcome. But links to GPL code is unfortunately of little use.

lars18th, 2020-10-07:
Hi,

I hope @jcekstrom-xumo will want to provide a PR, as hi has implemented the send functionality inside TSDuck.
And as @MathieuDuponchelle is the programmer of the GStreamer plugin for FEC support, then I hope hi can provide the receive functionality inside TSDuck based on his experience.

So if both provide the PR and sign for the collaboration, then the BSD-2 license will be done, as they are the original developers.
I'm right @lelegard ?

MathieuDuponchelle, 2020-10-07:
@lelegard GStreamer is LGPL v2, and decidedly not GPL, especially not v3 :)

MathieuDuponchelle, 2020-10-07:
@lars18th I did not intend to provide a PR here, just googled for `SMPTE 2022-1 GStreamer` to check if I could close some bugs, figured I would drop a link here in case that was useful. I'll be happy if my work can help tsduck in some way, I've used it in the past, it was very helpful :)

lelegard, 2020-10-07:
> @lelegard GStreamer is LGPL v2, and decidedly not GPL, especially not v3 :)

OK, but AFAIK gstreamer is not a simple shared library / DLL (especially on Windows) against which TSDuck can dynamically link. And gstreamer is a huge beast, not a simple Pro-MPEG library.

So, in practice, if we want Pro-MPEG support in TSDuck, we need to copy pieces of source code from gstreamer to TSDuck, which is incompatible with the LGPL license (or this would make TSDuck LGPL, which I don't want to).

To integrate Pro-MPEG in TSDuck, we need either:
- A simple LGPL shared library, already packaged and available in binary form on all macOS, Linux and Windows.
- Or source code which can be integrated into TSDuck source code base without making TSDuck LGPL.

Do we have this?

MathieuDuponchelle, 2020-10-07:
@lelegard what I can say is I'm completely OK with tsduck taking my code, porting it to their own data structures / data flow (the element uses `glib` data structures, eg `GSequence`, and deals with IO in accordance with GStreamer's API, eg. gets data from `GstPad::chain()` and outputs it with `gst_pad_push()`), and I'd grant permission to relicense. I'm the sole copyright holder and didn't use any third party code as the basis for that code, just worked from the spec.

lelegard, 2020-10-07:
@MathieuDuponchelle, thanks, that is great. Could you provide a snapshot of your code out of gstreamer, an archive of strictly necessary and sufficient code to implement Pro-MPEG, only your own code and with a permission to relicense?

Of course, if you are volunteer to port it as a TSDuck plugin yourself, that would be even better...

MathieuDuponchelle, 2020-10-07:
@lelegard if you check the MR the relevant code comes in standalone files already. Extracting it into a separate / standalone library will require doing the things I mentioned though (port away from glib, change how data IO works), and I'm afraid I don't have the time to do that. It's also probably best done by someone who knows what interface would work best for use within tsduck.

lelegard, 2020-10-08:
@MathieuDuponchelle, I got the code and looked at it. The code is clear and quite readable but tightly integrated with gstreamer in terms of architecture and tons of calls to `gst_`  functions. So, minor modifications would not be sufficient to integrate it with TSDuck or any other framework. The most practical way would be to rewrite a TSDuck plugin with the spec on one hand and your code on the other hand, using it as a reference implementation for protocol aspects. At least, there won't be licensing issue anymore :) 

I do not have time to do that and, without a proper test infrastructure, I will not be able to correctly test it anyway. So, if anyone _really_ needs Pro-MPEG / SMPTE 2022-1 in TSDuck, he will have to implement and test it using @MathieuDuponchelle's code as reference.

lars18th, 2020-10-13:
Hi @lelegard ,

> I do not have time to do that and, without a proper test infrastructure, I will not be able to correctly test it anyway. So, if anyone _really_ needs Pro-MPEG / SMPTE 2022-1 in TSDuck, he will have to implement and test it using @MathieuDuponchelle's code as reference.

If you need one _infrastructure_ to test it, you can do it using software only:
- You can generate/send RTP Mpeg-TS packets with Pro-MPEG FEC using the FFMpeg tool: https://ffmpeg.org/ffmpeg-protocols.html#prompeg
- You can receive/test from a RTP Mpeg-TS stream with FEC using this tool: https://github.com/kirintwn/prompeg-decoder

I've used the first option to generate a CBR stream with FEC in software and injecting it in a hardware headend. Futhermore, sending it over an _unreliable_ channel with packet lost and with good results for the recovery. So the code is robust. But for the the recive tool I don't test it personally.

In the other hand, if you ask about if we _need_ to have Pro-MPEG inside TSDuck... then personally I _would_ very much like it to be. However, I'm not actively using it at the moment. So you need to consider to implement it or not.

Regards.