tsduck/tsduck #192: Updated list of USB receivers

RubenDK, 2019-01-08:


RubenDK, 2019-01-08:
Is there an updated list of dvb receivers? The current list only contains obsolete DVB-S/S2 receivers.
I have an Hauppauge DVB-S2 461e which is detected, but I cannot download a TS stream

lelegard, 2019-01-08:
Hi @RubenDK 

The list in section 6.1.4 of the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf) is updated when users report feedbacks, mostly [here](https://github.com/tsduck/tsduck/issues/24). Some devices are recent, some are old.

This is not a restricted list of supported devices. This is a list of _actually tested_ devices. Any DVB tuner with a driver which is fully compliant with DirectShow (Windows) or V4L (Linux) should work. However, some devices are not fully compliant. On Windows, some devices work correctly only using the provided proprietary TV watching application. On Linux, some drivers or combinations of firmware/driver/kernel are broken or malfunctioning.

> I have an Hauppauge DVB-S2 461e which is detected, but I cannot download a TS stream

The 461e is indeed listed in the user's guide. Restrictions are documented: null packets are not reported on Windows and the driver is unreliable on Linux. I personally purchased that one 2 years ago and was extremenly disappointed about it. I file it into the "bad device" category. Then, I purchased a DVBSky S960 which works like a charm on Linux and Windows. A much better choice.

RubenDK, 2019-01-08:
Hi @Thierry

Thanks for your fast reply :) 
Yes, agree, the DVB-S2 461e is not working, it can even make make Win10 CMD freeze when issuing a --tune command.
I tried to find your DVBSky S960 but it looks like it is a very old unit, I need a new DVB-S2 USB receiver, cheap also, we are going to use it in many places, so if anyone know of such a unit I would be very happy to know.
Fantastic to see that the user guide is update so often, current version is January 2019 :) Thats great.

lelegard, 2019-01-08:
It seems that the DVBSky S960 was released around 2014. It depends on what you mean by "new". Is even "new" a good selection criteria? The 461e was probably newer than the S960 but it proved to be much worse. So, do you need a "working" device or a "new" device?

You may not like the form factor of the S960 (grey box) and prefer the "giant USB stick" form factor of the 461e. But do you need a "working" device or a "pretty" device?

Cheap? The S960 is priced 60 euros on Amazon. Good luck to find a cheaper S2 device. I mean both cheap and correctly working.

RubenDK, 2019-01-08:
I only found them at around 100 USD, but took a second look at Amazon and found the one for 60,- € 
Looking forward to try it, as I find TSDuck a fantastic tool, working so good with our Dektec units. BTW Thierry, do you know how to capture the entire TS stream from a DTE-3137? Looks like we only can save a single TS channel.

lelegard, 2019-01-08:
> do you know how to capture the entire TS stream from a DTE-3137? Looks like we only can save a single TS channel.

Sorry, but I never used that kind of device.

According to its [product description](https://www.dektec.com/products/IP/DTE-3137/):

> DekTec's standalone power-over-ethernet enabled DVB-S2 Satellite Receiver, allowing reception of the full Transport Stream and transmit it to IP and ASI networks.

So, getting the full TS should be possible. This is probably a matter of software. Currently, there is no specific support for network Dektec devices (DTE-31xx) in TSDuck. I do not think it can work since there is no support for attaching IP devices.

Supporting a new class of device (Dektec or whatever) needs at least one physical device for tests. So, such code contribution can come only from developers owning such a device. Do you think you could add support for DTE devices?

RubenDK, 2019-01-08:
Hmmm, Dektec is really a part of TSDUck, very integrated.
Have just ordered a S960 so eager to try to test it :)
Support on Dektec? Maybe, we need to discuss license and the use of TSDuck commecial

lelegard, 2019-01-08:
TSDuck is free open source. The source code is available to anyone. TSDuck has nothing to do with the Dektec company. It is not a commercial product.

TSDuck users with programming skills routinely [submit code contribution](https://tsduck.io/doxy/contributing.html) (ie. modification of the source code) to support new features or new devices. If anyone wants TSDuck to support DTE devices, he will have to modify TSDuck to do that and, preferably, submit the code modification back to the project. And, BTW, he will not receive any money for that, as I never received any cent to have written the 200,000+ lines of code of the project.

closed 2019-05-26 by lelegard