tsduck/tsduck #193: HLS Input

joolzg, 2019-01-08:
A nice feature to help debug would be a --save-segments which woul save the segments as they are received to the current directory or even better a specified directory.

The reason is due to a problem pulling a HLS stream with TSP and then processing this with FFMPEG, PTS and DTS errors, so no video frames are processed.

lelegard, 2019-01-08:
Why not using `-P file something.ts` between `-I` and `-O` ? It won't save the segments individually but you get the content file.

joolzg, 2019-01-09:
My idea was to be able to run the tests with HLS offline, hence needing the segments.

lelegard, 2019-01-13:
Added option `--save-file` which specifies a directory where all downloaded files, playlists and media segments, are saved.

joolzg, 2019-01-13:
Thanks for that, will aid me in finding out why my filtering code is not working.

closed 2019-01-13 by joolzg