tsduck/tsduck #194: Question and possible typo fix

nautilus7, 2019-01-09:
Hi, in [tsMPEG.h](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tsMPEG.h#L600) you list stream_type value 0x02 as _MPEG-2 video_. The [ITU-T H.222.0 specification](https://www.itu.int/rec/T-REC-H.222.0-201703-S/en), on table 2-34 page 55, lists stream_type 0x02 as _Rec. ITU-T H.262 | ISO/IEC 13818-2 Video or ISO/IEC 11172-2 constrained parameter video stream_.

So, my understanding is that in stream_type 0x02 we can have either _MPEG-2 video_ or _MPEG-1 constrained parameter video_ stream. Is this correct?

Thanks.

lelegard, 2019-01-09:
Formally, you are right. In practice, had anyone ever seen any MPEG-1 video in a _transport stream_ ?

This video type is typically seen in _program streams_ (ie. DVD), not in transport streams (ie. DTV, BluRay, IPTV, HLS).

nautilus7, 2019-01-09:
OK, I understand. I have no experience from actual content. I guess it's better using a name that describes what happens in real life, like you do now. Thanks.

closed 2019-01-09 by lelegard