tsduck/tsduck #195: Multiple outputs

joolzg, 2019-01-13:
Now we have HLS as an output is there a way (fork maybe) to output a stream to HLS and IP (multicast)?

lelegard, 2019-01-13:
Yes, the plugin `fork` was designed for that purpose. Something like:
~~~
tsp ... -P fork "tsp -O ip ..." -O hls ...
~~~

joolzg, 2019-01-13:
Not working

Ubuntu 16.04.3
tsp 3.16-1098

tsp ... -P fork "tsp -O file out.ts" -O hls -d 12 -l 5 --playlist fred.m3u8 fred.ts
Creates the TS and M3U8 file but not the out.ts

tsp ... -P fork "tsp  -O hls -d 12 -l 5 --playlist fred.m3u8 fred.ts" -O file out.ts
Creates out.ts but NO HLS

Anything i can do to help debug this

lelegard, 2019-01-13:
It works for me (Ubuntu 18.10, TSDuck 3.16-1098):
~~~
$ tsp -I file astra192E-ts1080-2018-05-11.ts -P fork "tsp -O file out.ts" -O hls -d 12 -l 5 --playlist fred.m3u8 fred.ts
$ ls -l
total 767644
-rw-r--r-- 1 thierry thierry  86532640 jan 13 19:27 fred000001.ts
-rw-r--r-- 1 thierry thierry  86300084 jan 13 19:27 fred000002.ts
-rw-r--r-- 1 thierry thierry  86322644 jan 13 19:27 fred000003.ts
-rw-r--r-- 1 thierry thierry  86241240 jan 13 19:27 fred000004.ts
-rw-r--r-- 1 thierry thierry   4444132 jan 13 19:27 fred000005.ts
-rw-r--r-- 1 thierry thierry       370 jan 13 19:27 fred.m3u8
-rw-r--r-- 1 thierry thierry 436205308 jan 13 19:27 out.ts
$ 
~~~

Just as in #180, something just does not work at all on your system while it works as expected on mine.

You may try `-P fork "tsp --debug -O file out.ts"` and look for hints.

You may use `-P fork "cat >out.ts"` instead of `-P fork "tsp -O file out.ts"`. It does the same without starting `tsp`. If it works, then you have an issue starting `tsp` in a forked process.

joolzg, 2019-01-13:
Ok tried with 
tsp -I file mheg_test.ts -P fork "tsp -O hls -d 12 -l 5 --playlist fred.m3u8 fred.ts" -O file out.ts
This works and i get both HLS and out.ts

Change the input to HLS i get out.ts only

joolzg, 2019-01-13:
OK added the debug and i get this

* Debug: debug level set to 1
trying to load /usr/bin/tsplugin_file.so
trying to load /usr/bin/tsplugin_hls.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_hls.so
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000

lelegard, 2019-01-13:
Sorry, but it works for me. Using a reference HLS test stream from Apple:

~~~
$ tsp -I hls https://devstreaming-cdn.apple.com/videos/streaming/examples/img_bipbop_adv_example_ts/master.m3u8 -P fork "tsp --debug -O hls -d 12 -l 5 --playlist fred.m3u8 fred.ts" -O file out.ts
debug level set to 1
* Debug: debug level set to 1
trying to load /home/thierry/tsduck/src/tsplugins/release-x86_64/tsplugin_file.so
trying to load /home/thierry/tsduck/src/tsplugins/release-x86_64/tsplugin_hls.so
symbol tspNewProcessor not found in /home/thierry/tsduck/src/tsplugins/release-x86_64/tsplugin_hls.so
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: file: debug level set to 1
* Debug: hls: debug level set to 1
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: input bitrate is 2,535,180 b/s
* hls: creating media segment fred000000.ts
* Debug: hls: creating file fred000000.ts, fd=5, error_code=2
* Debug: file: input thread started
* Debug: hls: output thread started
* hls: using service id 0x0001 (1) as reference, PMT PID 0x0100 (256)
* hls: using video PID 0x0101 (257) as reference
* hls: creating media segment fred000001.ts
* Debug: hls: creating file fred000001.ts, fd=5, error_code=0
* hls: creating media segment fred000002.ts
* Debug: hls: creating file fred000002.ts, fd=5, error_code=0
* hls: creating media segment fred000003.ts
* Debug: hls: creating file fred000003.ts, fd=5, error_code=0
* hls: creating media segment fred000004.ts
* Debug: hls: creating file fred000004.ts, fd=5, error_code=0
* hls: creating media segment fred000005.ts
* Debug: hls: creating file fred000005.ts, fd=5, error_code=0
* hls: deleting obsolete segment file fred000000.ts
* hls: creating media segment fred000006.ts
...
~~~

And all files are created, HLS segments, the playlist, the file `out.ts`.

But there are many more messages than what you copied. Didn't you see the debug traces about the creation of segment files ? Can you try the very same command (the Apple test stream is always online) ?

joolzg, 2019-01-13:
Im sorry but in further testing the HLS input I'm using takes 80 seconds, 10 segments received, before a the new HLS is created, from then on it works

Thanks again.

closed 2019-01-13 by joolzg