tsduck/tsduck #196: ISO 6937 based DVB text not converted correctly

petrn, 2019-01-13:
The conversion of the DVB text encoding to Unicode using the "default character coding table" (based on ISO 6937) as specified in EN 300 486 Figure A1 is handling the non-spacing diacritical marks incorrectly.

The order used in ISO 6937 and DVB is that the non-spacing symbol/diacritical mark comes FIRST, followed by the base character (see for instance https://en.wikipedia.org/wiki/ISO/IEC_6937). 

In Unicode, the "combining marks" FOLLOW the base character (see for instance http://www.user.uni-hannover.de/nhtcapri/combining-marks.html).

The existing code in tsduck (for instance in handling EIT tables) does incorrectly convert the non-spacing symbols to Unicode combining marks without correctly adjusting the order and keeps them in the order of the original DVB encoding. Also, for any further processing, it would be highly desirable to actually convert the text to use the precomposed letters rather than the combinations.

Please note that for instance the Czech D-Book mandates the use of the default coding table (Figure A1) for all texts and that Czech has a high frequency of accented characters (higher than French for example), so this is actually a major issue to use the tool for any SI/PSI in this region.

So this correct text (randomly chosen from EIT text in Czech):

Dědictví po bohaté osamělé tetě je pádným důvodem k příbuzenské návštěvě...

is handled by tsduck incorrectly as the following (this may not display properly as most of the combined characters do not exist):

Ďedictv́i po bohat́e osam̌eĺe teťe je ṕadńym d̊uvodem k p̌ŕibuzensḱe ńav̌sťev̌e...

lelegard, 2019-01-13:
Hi @petrn,

Thank you for reporting this. You clearly found a bug. The order of letter and diacritical mark was not addressed at all.

This problem does not exist in English (for obvious reasons). It is usually avoided in French, Spanish, Hebrew, Greek, Arabic and many other languages for which an appropriate DVB table is available (ISO 8859-15 or similar). The problem does exist only in Latin languages with many diacritical marks. Czech seems to be one of them (sorry for not understanding a word of Czech).

This will be fixed. Could you provide a couple a tables with complex Czech text with many diacritical marks? Please extract tables in binary format (`tstables --binary`) and attach the binary files here. These files will be added in the non-regression test suite.

From your description, I see two problems: a bug and an enhancement.

The bug is what you describe: incorrect order of diacritical marks when converting from ISO 6937 to Unicode. The fix should also, as you suggest, convert to precomposed letters when possible.

The enhancement comes from what you mentioned: _"the Czech D-Book mandates the use of the default coding table"_. When TSDuck converts a Unicode string (from XML for instance) into a DVB table, it tries a few character sets and, if none can convert all characters, it uses UTF-8 (DVB code table 0x15). This is always safe. However, this won't match the requirements of the Czech D-Book. So, there should also be an option to force the generation of ISO 6937 by decomposing accented letters into (reversed) diacritical combinations. This will take more time.

One question comes to my mind. The Wikipedia article you mention on ISO 6937 clearly indicates that the diacritical mark precedes the letter. But what about ISO 8859, as used in many other DVB tables? The article at https://en.wikipedia.org/wiki/ISO/IEC_8859 does not mention the order. Any idea?

petrn, 2019-01-13:
hi @lelegard,

there is a similar table to the one you mention for French - the 8859-2 - but the DVB use here prefers the 6937 method. Now it is enforced by the D-Book.

I will provide binary EIT tables - this is easy for me, I am the author of the EIT generation software used by the Czech public broadcaster Czech TV to generate their EPG data for all their distribution channels.

Regarding the accents in the 8859 family of standards, there is a fundamental difference there - while the 6937 uses non-spacing accents, the accents in 8859 are SPACING accents and thus do not compose with the letter before or after them, they are just taking a whole character position on their own. In 6937, you can achieve the spacing accent by having the non-spacing accent, followed by space.

Unicode has spacing accents as well, you use them correctly in your tables for the 8859 family.

So there does not seem to be any issue related to the 8859 tables and accents, as these are all spacing accents.

lelegard, 2019-01-14:
>  I am the author of the EIT generation software used by the Czech public broadcaster Czech TV to generate their EPG data

Excellent, so you are the right person to discuss with.

> I will provide binary EIT tables

Thanks. You may also send them, as well as any valuable technical data, to contact@tsduck.io.

> there is a similar table to the one you mention for French - the 8859-2 - but the DVB use here prefers the 6937 method. Now it is enforced by the D-Book.

I assume that you are refering to the Czech D-book here. Is there a publicly available English version of the Czech D-Book?

In fact, all French operators use ISO 8859-15 (aka Latin-9) because 8859-2 (aka Latin-1) provides partial coverage only for the French language while 8859-15 provides full coverage. 

Note that at least one French satellite operator is doing worse. Its signalization implicitly uses 8859-15 without specifiying 0x0B as first byte in DVB strings. This is clearly a violation of the standard because such strings shall be interpreted as ISO 6937. This is the reason why I added the option `--default-charset` in several tools and plugins. It states that, in the absence of a leading character set indicator, DVB strings shall use the specified character set by default, instead of ISO 6937. For instance, the analysis of streams from such operators makes sense only with option `--default-charset ISO-8859-15`. Without this option, texts with accented letters are displayed as garbage.

petrn, 2019-01-14:
@lelegard 

Yes, I am aware of the issues with the missing character set specifiers - I believe the Polish satellite operators have a similar problem (or at least had such a problem some time ago), they have been using their local default charset but failed to signal it. I believe that is why the 6937 encoding is used here, as it is impossible not to signal it as there is no signalling.

Regarding the Czech D-Book, the old version from 2012 was available in English, but the current 2016 version is only available in Czech (at the time of publication, it was promised that the English version will be published "later", so we still have to see when that "later" is).

I am on a business trip this week, so I may only send you the samples after I come back on Saturday. Sorry about the delay, but I might find some example and provide that. Most likely I will take the PID 18 of the public broadcaster mux and send that to you as a transport stream, you can use your tool to process it. I will also send you the correct decoding from an analyser I have written some years ago, which handles all the character sets according to the standard (I hope).

lelegard, 2019-01-14:
> Regarding the accents in the 8859 family of standards, there is a fundamental difference there - while the 6937 uses non-spacing accents, the accents in 8859 are SPACING accents and thus do not [...]

I am not sure that ISO 8859 diacritical marks are spacing.

First, in ETSI EN 300 468, Annex A specifies the following color coding: _light pink - non-spacing symbols (diacritical marks)_. Several ISO 8859 tables in the same annex contain such cells, same as used in the default ISO 6937 table.

Second, an old draft of ISO 8859-11 I have mentions _"Some of these characters are combining characters. They are identified in table 1 as such."_ And, in fact, the named table 1 identifies some characters are "combining".

But, in all cases, there is no mention of the order into which the diacritical mark and the associated letter shall be used.

In the absence of more precise information, I will fix the order in ISO 6937 only.

lelegard, 2019-01-15:
Hi @petrn 

A proposed fix has been committed last night.

- The order of diacritical marks has been fixed when decoding and encoding ISO 6837 (default DVB charset).
- Decoding ISO 8858 (any other DVB charset) remains unchanged until a consensus is found on its order of diacritical marks.
- The pairs of letter + combining diacritical mark are combined into one character when possible when decoding DVB strings.
- Encoding remains otherwise unchanged. Czech texts and other texts with lots of diacritics will remain encoded in UTF-8 in DVB tables.

Please test the fix. 

I am still interested in real-life tables (EIT's for instance) to insert them in the non-regression test suite.

lelegard, 2019-01-20:
@petrn sent a set of Czech EIT's and validated the results after the fix. The provided test sections were added in the [TSDuck test suite](https://github.com/tsduck/tsduck-test) as [test-038](https://github.com/tsduck/tsduck-test/blob/master/tests/test-038.sh). Closing the issue.

closed 2019-01-20 by lelegard

petrn, 2019-02-01:
Hi Thierry,

sorry not to follow up on this before, but I was traveling out of the country recently.

I would like to refer here to the following to substantiate my claim that all the accent characters in ISO/IEC 8859-1 are SPACING. The Unicode uses the C1 “controls and Latin-1 Supplement” in the same code positions as ISO/IEC 8859-1 (you can say the 8859-1 is a proper subset of Unicode). The Unicode gives a clear definition of each of the code position, also referring to other related code positions.

Here : https://www.unicode.org/Public/6.2.0/charts/CodeCharts.pdf <https://www.unicode.org/Public/6.2.0/charts/CodeCharts.pdf>

you will find the Unicode code tables and then in that document, on page 8 you will find the spacing accents such as Macron, Acute Accent and Cedilla. In all cases, the standard clearly says “this is a spacing character” and refers to the combining version of the accept in a code position in the range 0300 to 0322.

For the 8859-2 (the one for my region), you have the same: all accents are from the C1 range or the 02Dx range and they are again all spacing (the 02Dx range is even called “Spacing clones of diacritics”).

When you check the tables for the 8859 variants based on Latin script (that is, -1, -2, -3, -4, -9, -10, -13, -14, -15, -16), you will find that all accent characters are either in the C1 range or in the 02Dx range and none of them are in the “0300 to 0322 Combining Diacritical Marks”.

I did not check the non-latin script tables, but I did check all the above listed ones.

I hope this helps.

Best regards,

lelegard, 2019-02-03:
Hi Petr,

I checked a few other codes and my conclusion is that you are right and ... _DVB is wrong_ !

Let's take an example. In ETSI EN 300 468 figure A.4: _Character code table 03 - Latin/Greek alphabet with Unicode equivalents_, documented as compatible with ISO/IEC 8859-7, we see three light pink cells which, according to the color code, means __"non-spacing symbols (diacritical marks)"__. Two of these codes are Unicode 0384 and 0385. Looking into your chart code, they are documented in page 36 as __"spacing accent marks"__. Amazingly, I never realised that DVB could be wrong that way.

Thanks again, I appreciate with your precision.

So, TSDuck is now correct because I implemented inversion on ISO 6937 (default table) and not on other tables.