tsduck/tsduck #197: HLS Input when index has lots of segments and is a live list

joolzg, 2019-01-13:
http://yletvhdliveworld-lh.akamaihd.net/i/yletv1hdworld_1@187592/index_184_av-p.m3u8?sd=10&rebase=on

If you pull this, the HLS input starts pulling all segments, around 1530, so for a live stream you start 1530 * 10 seconds behind.

How do we make the HLS input start with the last segment in the list and only add new segments from that point onwards.

Maybe a "live" option like in the HLS output.

lars18th, 2019-01-14:
Great idea!

I suggest to add the parameter `--start-point <value>` with:

- "-1" : Start with the _last_ listed segment.
- "0" : Start with the _first_ listed segment (just is done now) [default value].
- "1": Start with the second (+1) segment.
- "2": Start with the third (+2) segment.
etc.

I think that's how you look at every possible case.
You agree?

lelegard, 2019-01-15:
OK, to make everyone happy, I implemented both 😄 

For @lars18th, the option `--start-segment` has exactly the same semantics as requested and for @joolzg  the option `--live` is an alias for `--start-segment -1`, which means start at the last segment.

joolzg, 2019-01-15:
Thank you again.

closed 2019-01-20 by lelegard