tsduck/tsduck #199: UDP burst after merging

MikeGogia, 2019-01-14:
After merging  udp flow at the output of the analyzer we get 7,4,1, ..etc packet mapping . How to compensate (stabilize) to 7.
[image]

lelegard, 2019-01-14:
Hi @MikeGogia,

Currently, the packing of TS packets in UDP packets is not enforced. When the `ip` output plugin receives a set of TS packets, it groups them _up to_ the specified burst.

For instance, if the output plugin receives 16 TS packets, it immediately sends 3 UDP packets containing 7, 7 and 2 TS packets respectively. It does not buffer the remaining 2 TS packets, delaying their departure until 5 more TS packets are available.

I should probably add some `--enforce-burst` option to enforce a fixed number of TS packets per UDP packet, at the expense of some TS packets being delayed.

MikeGogia, 2019-01-14:
Thanks, this option would be very helpful.

lelegard, 2019-01-15:
The `--enforce-burst` option has been added.

MikeGogia, 2019-01-18:
Thank you, this option has appeared. Now we are testing

rezna11, 2019-01-18:
I've tried this option too, and it worked :) I tested on Linux because I could not get TSDuck to compile on Windows this time.