tsduck/tsduck #201: Test request for plugin pcradjust (continuous PCR resync)

lelegard, 2019-01-20:
Hi TSDuck users,

The problem of the recomputation (or smoothing) of PCR's has been raised several times. In commit 055f962dcb8b0c41d417a5b117cf6a29c782ef50, the plugin `pcradjust`has been added and does (more or less) what is needed for PCR smoothing.

It works correctly with constant bitrate only. On CBR streams, the value of a PCR is simply computed from the packet distance and bitrate. To do PCR smoothing on VBR streams, you need a real time application on live streams which reads each packets exactly at the right time. Otherwise, you do not know the appropriate PCR time for the packet and you cannot compute the right PCR. Because of the asynchronous, buffered and heavily multi-threaded nature of `tsp`, the precise reception time of a packet is unknown. So, unless I miss something obvious, there is no way to reliably smoothen PCR's in VBR streams with `tsp`.

The other topic is the related adjustment of PTS and DTS. Significantly modifying the PCR's without modifying PTS and DTS most likely results in an unplayable video and audio since PTS and DTS become completely decorrelated with the system reference clock. Some choices have been made and are explained in the header comments of [tsplugin_pcradjust.cpp](https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_pcradjust.cpp).

To all users who are interested in this feature, please test and provide feedback here.

Thanks
-Thierry

lars18th, 2019-01-21:
Hi @lelegard ,

First, thank you for this "must have" plugin! :+1: 

After reading the documentation and the code (and before testing it), I have some comments about it:

1. Why isn't a parameter like `--force-dts` & `--force-pts`? In the code you described a simple strategy to recompute the PTS/DTS values: if the distance of two consecutive PCR are more than 500ms. But, if we like to recompute PTS/DTS in all cases? One user case: prior to mux a SPTS with a MPTS, we can smooth it recomputing all timing marks: PCR and PTS/DTS. The reason can be because the source is something not very DTV conformant, for example one HLS source from a live camera or a video from a VOD server. So combining this `pcradjust` with other plugins (like `pcrbitrate` or others) then we can improve the quality of the muxing. And _after_ the muxing, we can use another time this `pcradjust` plugin to just put correct PCR marks in the overall MPTS. I hope you can see the two uses of this plugin: _before_ a muxing, and _after_ the muxing task.
2. I feel that one option for changing the distance (500ms) can be useful.
3. Regarding the impossibility of recompute PCR marks with VBR Transport Streams, I have one "suggestion". The idea is a tool that "plays" (based on PTS/DTS) a VBR stream and generates a CBR with a maximum bitrate value. Then this CBR can be streamed, processed by the `pcradjust` plugin and muxed inside a MPTS. I don't see any other option to do it.

Regards.

lelegard, 2019-01-21:
Thanks @lars18th for your feedback.

The difficulty of recomputing time-stamps is determining the right value to set, finding the reliable time reference on which a value shall be set. And sometimes, it ends up with a chicken & egg issue.

The PCR timestamp in a packet is defined as the expected system time where the packet is received in the playout device, typically a set-top box. In other words, the PCR _is the reference clock_. The PTS/DTS are used by the codec to process frames, relative to the system clock. PCR and PTS/DTS are two independent classes of time-stamps. In the general case, _you cannot use one to compute the other_.

The reference system clock is supposed to be univeral. It is a monotonically increasing clock. In the simple case of packet reordering, the placement of a packet changes in the stream. So, it also changes the placement in time. And the PCR must be modified accordingly. But how? How do we get the correct new value?

Recomputing PCR's in a CBR stream is relatively easy, just count packets.

> Regarding the impossibility of recompute PCR marks with VBR Transport Streams, I have one "suggestion". The idea is a tool that "plays" (based on PTS/DTS) a VBR stream and generates a CBR with a [...]

PTS/DTS are defined as relative to the Program Clock Reference (PCR). Decoding and presentation time stamps are not even monotonic, they can go back and forth depending on the type of frame. In the general case, you cannot deduce the PCR from the PTS/DTS.

> Why isn't a parameter like --force-dts & --force-pts? In the code you described a simple strategy to recompute the PTS/DTS values: if the distance of two consecutive PCR are more than 500ms. But, if we like to recompute PTS/DTS in all cases?

There is a misunderstanding here. The 500ms threshold is not between two PCR's. It is the difference between the PCR and the PTS or DTS in the same TS packet (or the theoretical PCR of this packet if there is no PCR in the packet containing a PTS/DTS).

This addresses the problem of computing a sensible value for a PTS/DTS. There is no predefined algorithm to recompute the PTS/DTS from a PCR. In the same packet, the PCR, PTS and DTS (when they are all present) contain slightly different values. The PCR is a reference clock. When you move a packet back and forth in the stream, the PCR must be adjusted accordingly. But the frame will remain to be decoded (DTS) and presented (PTS) at unmodified times. So, _first case_, when TS packets are slighlty moved in a stream (muxing jitter), the PCR must be adjusted but the PTS and DTS _shall not be modified_.

But there is also a _second case_, when the PCR value suddenly leaps to a value with a quite different order of magnitude. This can happen when you loop over a stream file: after the last PCR value, the next PCR suddenly reverts back far in the past. This can also happen when a stream or service is suddenly replaced with another one, using its own quite different time reference. In that case, the PTS and DTS _shall be modified_. Otherwise, the decoding and presentation times are so far away in the past or future that it never happen (ie. no video at all).

The 500ms threshold is simply an empirical way to differenciate the two cases (must not modify PTS/DTS vs. must modify them).

Note that the second case shall be split into two subcases:

- 2.1 : The time reference is completely changed but the packet ordering and packet count remain the same (simple file loop for instance). In that case, a safe way to recompute the PTS/DTS is to apply the same difference between the new PCR and PTS/DTS as measured between the old PCR and PTS/DTS. This is what the `pcradjust` plugin is doing when the "second case" (PCR leap) is detected.

- 2.2 : Both the time reference and the muxing is changed (loop + merge for instance). The bad news is that 1) there is no way to detect this situation and 2) even if we could detect it, there is no magic formula to compute the rigth value for PTS/DTS.

So, since there is nothing feasible in the second subcase, the `pcradjust` plugin always applies the "PCR - PTS/DTS diff" method when a PCR leap is detected.

> I feel that one option for changing the distance (500ms) can be useful.

Why not. It does not hurt. However, the frontier between the two main cases above is between "not much" (first case) and "a lot" (second case). Fine tuning is probably useless.

As a conclusion, while adjusting PCR in a CBR stream is relatively easy, modifying them in VBR streams and modifying PTS/DTS in all cases is quite difficult. I mean that it is simply difficult to specify what should be done precisely.

lars18th, 2019-01-24:
Hi @lelegard ,

Thank you very much for your excellent explanation. The experience you have is noticeable. :smiley: 
I have very little to say accordingly...

- For the case 2.2 (loop & merge) I feel the best solution is to use "two" `pcradjust` filters. One only for the input loop file (to catch the PCR's return to the past) and another one after the merge for smothing the placement of PCRs.

I'm right?

- And regarding the VBR case, the most simple solution is the one used by some professional tools: encapsulate the VBR stream in a CBR TS with padding and restamp the PCR marks. And then you can do the MUXING without troubles.

So what would be the best way to do that with TSDuck?

lelegard, 2019-01-24:
> For the case 2.2 (loop & merge) I feel the best solution is to use "two" pcradjust filters. One only for the input loop file (to catch the PCR's return to the past) and another one after the merge for smothing the placement of PCRs.

Using two sequential `pcradjust` is normally useless. The second one will have nothing to adjust since all PCR's should be correctly aligned in the stream at the output of the first one.

The problem with loop & merge is that some time information has been destroyed and cannot be recreated. I see no solution _at all_. Any solution I have missed is welcome.

> And regarding the VBR case, the most simple solution is the one used by some professional tools: encapsulate the VBR stream in a CBR TS with padding and restamp the PCR marks. And then you can do the MUXING without troubles.

Encapsulating a VBR stream into a CBR stream cannot be a _solution_ to adjust PCR. On the contrary, having correct (or correctly adjusted) PCR's is a _prerequisite_ to encapsulate a VBR into a CBR. Otherwise, you do not know how many null packets you have to insert between VBR packets. After building the CBR, you have a to slightly adjust PCR again, but very slightly only (to realign time stamps at packet boundary).

I wonder how so-called professional tools recreate PTS and DTS in streams after time leap (loop or whatever) and packet reordering / jitter, at the same time. By "how", I mean "how exactly", what kind of precise computation, formula or algorithm they use to end up with a meaningful PTS/DTS value.

lars18th, 2019-01-24:
> Using two sequential pcradjust is normally useless. The second one will have nothing to adjust since all PCR's should be correctly aligned in the stream at the output of the first one.

Right if we use the `merge` plugin. However, after the muxing perhaps other processing can modify something in the stream; and another `pcradjust` must be good to make the output TS full conformant.

> The problem with loop & merge is that some time information has been destroyed and cannot be recreated. I see no solution at all. Any solution I have missed is welcome.

Well, the brute force solution -in my opinion- is to transmux the input. For example, using the FFmpeg tool you can _decode_ and _reencode_ the content... even when the input has some missing time information. The output of this tool will then have all the full timing information. So afterwards we can use the TSDuck to do the remaining work.

> Encapsulating a VBR stream into a CBR stream cannot be a solution to adjust PCR.

Well, if packets are comming at the correct speed, then the VBR can be muxed in a CBR, right?

I'm assuming that with _perfect_ CBR Transport Streams (Simple or Multiple) the TSDuck can do a correct muxing... even when some ES are VBR. It's this assumption correct, or not? If not, then please comment in which case it can fail.

I'm asking because I'd like to know more about it.
Regads.

lelegard, 2019-01-24:
> Well, if packets are comming at the correct speed, then the VBR can be muxed in a CBR, right?

This is right only for a dedicated and extremely precise real-time application which picks every incoming packet at the very precise time they individually arrive. `tsp` is a general-purpose and flexible framework which equally works on live streams and TS files. To face the variety of situations, it is buffered and highly multi-threaded. Packets are processed in chunks and the latency is too high for this kind of operation.

> I'm assuming that with perfect CBR Transport Streams (Simple or Multiple) the TSDuck can do a correct muxing... even when some ES are VBR. It's this assumption correct, or not? If not, then please comment in which case it can fail.

In the case of a perfect CBR, the PCR and PTS/DTS are distinct cases:
- PCR: the `pcradjust` plugin always computes the exact value for all PCR.
- PTS & DTS: If there is no packet reordering or jitter (ie. no `merge` plugin), the PTS/DTS are correctly computed, even in the case of time leap (loopback for instance). In the case of packet reordering, the correct PTS/DTS information is lost and I do not see how to recompute them. The PTS & DTS will be adjusted to some approximately correct value, but not extremely precise.

I have thought about this problem for quite some time. Addressing all cases is not trivial. Some cases are easy, some others have (I think) no accurate solution in the context of `tsp`. I may have missed good ideas and suggestions are welcome but they now should come in the form of an algorithm description.

lelegard, 2019-01-25:
> My idea is about this: if we can use some "external" tool to garantee with a precise outputing of the packets of a VBR TS, then the TSDuck can mux it perfectly inside a CBR TS?

Not now. The idea is to enhance the `merge` plugin with some option that would use the input PCR of the merged stream as a reference for placement in the stream (with final small PCR adjust anyway).

This and another option for smoothen merge (when input PCR are not really reliable) may be in version 3.17 (hopefully).

> So, as I understand it, when you do the mixing the PCRs are not the real problem, but the PTS/DTSs. Is that so?

Exactly. More precisely when doing time adjustment in `pcradjust`. 

Think in terms of a 4 dimentional matrix of 2 conditions each and study the 16 different environments.

1. Adjust PCR vs. PTS/DTS.
2. Input CBR vs. CBR
3. Muxing jitter or not (and how to detect it)
4. Time leap or not (and how to detect it)

A solution must be found in each of the 16 contexts.

lelegard, 2019-01-25:
I would prefer to have any problem with `pcradjust` reported as soon as possible. I hope to have time to release version 3.16 over the week-end, unless some critical issue is found in the meantime.
Thanks
-Thierry