tsduck/tsduck #204: Transcoding MPTS strem of type UDP to UDP

sachin0606, 2019-01-24:
Hi, 

Thanks for the tool, i dont think its rite platform to ask this, but even i feel this tool may help in this process so asking this question here.

currently i am getting MPTS stream from IRD, and using FFMPEG to transcode the h264 codec to convert MP2 mpegts stream. but after processing some time  [may few hours] getting udp streams size getting down and its picture starts freezing and even packate drop issue i am facing also all the PSI data will be lost as maping manually in FFMPEG.

any way i can pipe the stream out of tsduck to FFMPEG and map the metadata of the stream to output transcoded video?

current command which i am using  with FFMPEG is  -> 

ffmpeg -i udp://224.2.2.11:1011 -map 0:p:1 -map 0:p:2  -vcodec  mpeg2video -acodec copy -b:v 4M -b:a 128k -program title=XYZ:st=0:st=1 -program title=XYZ1:st=2:st=3 -f mpegts "udp://230.0.0.5:1012?pkt_size=1316&localaddr=192.168.2.58&localport=22

any suggestion???

lars18th, 2019-01-24:
> any suggestion???

Yes! If you need to do some processing with the FFmpeg tool, but you like to preserve the PSI data, then you need to use the `encap` plugin with the PES mode.

See the examples in the documentation about it. :wink:

sachin0606, 2019-01-24:
Encap i understood, But how to use tsduck and FFMPEG together?

lars18th, 2019-01-25:
> But how to use tsduck and FFMPEG together?

With PIPES !

For example: `ffmpeg ... -f mpegts | tsp ...`
Or: `tsp ... | ffmpeg -re ...`

Or when you like to mux from FFmpeg to a TS throught TSDuck then use the `merge` plugin.
Please, read the documentation https://tsduck.io/download/docs/tsduck.pdf
and take note of the examples, like _"5.2.23 Merging transport streams"_

The TSDuck is a _very_ good documented project!

closed 2019-05-26 by lelegard