tsduck/tsduck #205: gcc 9: error: "implicitly-declared '[...]::operator=([...])' is deprecated [-Werror=deprecated-copy]"

barsnick, 2019-01-28:
I'm building tsduck as a Fedora RPM from a SPEC file. (It's a self-crafted SPEC file, and I'm building myself instead of using your provided RPMS, so that it shows up in a Fedora-compatible repository, a so-called "copr".)

On Fedora rawhide (the runner-up to F30), gcc 9 is being used. It seems gcc doesn't like some constructs. My build errors out on tsBAT.cpp (it may be warnings, as it says "```cc1plus: all warnings being treated as errors```", but the quoted command line option is "```-Werror=deprecated-copy```"):

```
  [CXX] tsBAT.cpp
tsBAT.cpp: In member function 'ts::BAT& ts::BAT::operator=(const ts::BAT&)':
tsBAT.cpp:75:45: error: implicitly-declared 'ts::AbstractTransportListTable& ts::AbstractTransportListTable::operator=(const ts::AbstractTransportListTable&)' is deprecated [-Werror=deprecated-copy]
   75 |         AbstractTransportListTable::operator=(other);
      |                                             ^
In file included from tsBAT.h:36,
                 from tsBAT.cpp:34:
tsAbstractTransportListTable.h:95:9: note: because 'ts::AbstractTransportListTable' has user-provided 'ts::AbstractTransportListTable::AbstractTransportListTable(const ts::AbstractTransportListTable&)'
   95 |         AbstractTransportListTable(const AbstractTransportListTable& other);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
tsBAT.cpp:75:52: error: implicitly-declared 'ts::AbstractTransportListTable& ts::AbstractTransportListTable::operator=(const ts::AbstractTransportListTable&)' is deprecated [-Werror=deprecated-copy]
   75 |         AbstractTransportListTable::operator=(other);
      |                                                    ^
In file included from tsBAT.h:36,
                 from tsBAT.cpp:34:
tsAbstractTransportListTable.h:95:9: note: because 'ts::AbstractTransportListTable' has user-provided 'ts::AbstractTransportListTable::AbstractTransportListTable(const ts::AbstractTransportListTable&)'
   95 |         AbstractTransportListTable(const AbstractTransportListTable& other);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [../../Makefile.common:898: release-x86_64/tsBAT.o] Error 1
```

Since the "make" call doesn't use "-k", there may be more warnings/errors in the pipeline.

Older gccs seems to work fine.

The complete log is currently located here:
https://copr-be.cloud.fedoraproject.org/results/barsnick/non-fed/fedora-rawhide-x86_64/00851258-tsduck/builder-live.log

barsnick, 2019-01-28:
Update:
By adding "```-Wno-error=deprecated-copy```" to COMMON_WARNINGS in Makefile.common, I can get tsduck to compile successfully.

Of course, I recommend to fix the code. Don't ask me how though. ;-)

Background: in c++11 mode, as tsduck is using, this is given:

"The implicit definition of a copy constructor as defaulted is deprecated if the class has a user-declared copy assignment operator or a user-declared destructor. The implicit definition of a copy assignment operator as defaulted is deprecated if the class has a user-declared copy constructor or a user-declared destructor."

lelegard, 2019-01-28:
Hi @barsnick,

Thank you for the feedback. TSDuck was never compiled with GCC 9 so far. I am usually extremely strict with warnings since they potentially spot problems. The code will be definitely cleaned up as soon as I can use GCC 9. The warning is clear and I know what to do.

In the meantime, you have already found a workaround with option `-Wno-error=deprecated-copy`. I even suggest `-Wno-deprecated-copy` in order to avoid the warning messages.

lelegard, 2019-01-28:
Now fixed in commit c69e977aedb7f8c8977162364c336b191b9603a5. Most fixes are added "`= default`" for copy constructors or assignment operators. In all cases, it was known that one of the two was needed while the other's default was just fine. In other words, the fixes do not add any generated code. They just tell the compiler "_yes, we understand that the default implementation is appropriate here_".

barsnick, 2019-01-29:
Nice, applying commit c69e977 as a patch to 3.16-1110 works "perfectly".

Builds are here:
https://copr.fedorainfracloud.org/coprs/barsnick/non-fed/build/851603/
and here's a log - "look mom, no warnings":
https://copr-be.cloud.fedoraproject.org/results/barsnick/non-fed/fedora-rawhide-x86_64/00851603-tsduck/builder-live.log

Thanks for the quick fix!

closed 2019-01-29 by barsnick

lelegard, 2019-01-30:
Here is the way to install GCC 9 on a standard Fedora 29 (without a full upgrade to Fedora Rawhide, upcoming Fedora 30).

Simply upgrade the following packages. Depending on packages which were already installed on your Fedora 29, you may need additional packages. Just follow error messages from `rpm -Uvh`.
~~~
cpp-9.0.0-0.3.fc30.x86_64.rpm
gcc-9.0.0-0.3.fc30.x86_64.rpm
gcc-c++-9.0.0-0.3.fc30.x86_64.rpm
gcc-gdb-plugin-9.0.0-0.3.fc30.x86_64.rpm
libgcc-9.0.0-0.3.fc30.x86_64.rpm
libgomp-9.0.0-0.3.fc30.x86_64.rpm
libstdc++-9.0.0-0.3.fc30.x86_64.rpm
libstdc++-devel-9.0.0-0.3.fc30.x86_64.rpm
libstdc++-static-9.0.0-0.3.fc30.x86_64.rpm
~~~

The necessary packages were downloaded from here: https://dl.fedoraproject.org/pub/fedora/linux/development/rawhide/Everything/x86_64/os/Packages/

TSDuck is successfully built. The two test suites successfully pass as well.
~~~
$ cat /etc/fedora-release 
Fedora release 29 (Twenty Nine)
$ uname -a
Linux vmfedora 4.20.3-200.fc29.x86_64 #1 SMP Thu Jan 17 15:19:35 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
$ tsversion --version=compiler
GCC 9.0.0, C++ std 2011.03
$
~~~