tsduck/tsduck #206: problem using the --atis-idsa option to descramble an AES stream

marcabramson, 2019-01-29:
I have a stream which is scrambled in ATIS/IDSA (my hardware is able to descramble correctly) and , for test purpose, I would like to descramble it using tsduck

my command line is 
tsp --verbose -P descrambler --atis-idsa --cw-file cw.txt  1102-I file scrambled.ts O file clear.ts

with cw_file.txt the file containing the 16 bytes CWs, scrambled.ts my stream, and 1102 the service id of the stream I would like to descramble

the problem is that, when I run ts duck, I have the following output
* file: input bitrate is 50,000,232 b/s
* descrambler: found service id 0x044E (1102), PMT PID is 0x0100 (256)
* descrambler: using scrambling mode: DVB-CSA2

for me, since I have put the --atis-idsa option in the command line, the descrambling mode shall NOT go tot DVB-CSA2 even if it's true that the service does NOT contain the expected scramling_descriptor in the PMT of the service.

lelegard, 2019-01-29:
Hi Marc,

I am currently traveling and not able to run test. Looking at the code, it seems that the message is misleading. It represents the scrambling mode from the PMT (here CSA). But the actual descrambling should use the explicit algorithm.

Could you please test that the descrambling is correct, using ATIS? I cannot run any test by now.

If this is the case (should be), I will fix the incorrect message.

marcabramson, 2019-01-30:
Hi Thierry
I can confirm that ATIS descrambling is working , and that only message is incorrect.

lelegard, 2019-01-30:
Thanks for the feedback. The message is fixed in commit 49fc963ed02816ce6282da7bfd7b1436ff5b568d.

I must admit that this message went unnoticed because all tests were done on conformant streams. The `scrambler` plugin also automatically adds a _scrambling_descriptor_ when needed.

closed 2019-01-30 by lelegard

marcabramson, 2019-01-30:
Re: [tsduck/tsduck] problem using the --atis-idsa option to descramble an AES stream (#206)

Bonjour Thierry,
my experience with many different CAS vendor shows that for many of them, this scrambling descriptor is NOT present when the stream is scrambled in ATIS... or not present in their test stream (I have to admit that I have never seen any AES scrambled stream in live broadcast up to now).

Best regards
Marc

Le mercredi 30 janvier 2019 à 13:42:20, vous écriviez :

Thanks for the feedback. The message is fixed in commit 49fc963.
I must admit that this message went unnoticed because all tests were done on conformant streams. The scrambler plugin also automatically adds a scrambling_descriptor when needed.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv