tsduck/tsduck #208: problem with AES ATIS descrambling (again)

marcabramson, 2019-01-30:
Hello Thierry, 
sorry to find issues and to hassle you as I do .. and I hope that my issues are not bugs, but bad use of the TSduck software from my side

this time, I have a stream that is scrambled in AES ATIS. It's perfectly descrambled using my hardware, but I would like to descramble it using tsduck
I know that the 4 first CW used are 
B3B83F7A7DC95262718B6E188CAD3629
F16F7EB000E9804366B066D15625E271
C69AE040F51AA5796CBFBD5944981718
DB39BFB9D27B8E4DD0E1C9829DB62B2A

and that the crypto period is 15s.
when I do the following command line
tsp --verbose -P descrambler --atis-idsa --cw-file fix_cw.txt  1102 -I file scrambled.ts -O file clear.ts

if I put in the fix_cw.txt file the first CW (so B3B83F7A7DC95262718B6E188CAD3629) then, I can see (using VLC) that file_clear.ts is correctly  descrambled (for both audio and video)  from 0 to 15 s, which is exactlly when I'm expecting
if now, I put the second CW F16F7EB000E9804366B066D15625E271, then, clear.ts is correctly  descrambled from 15s to 30s
if I put Rd CW  C69AE040F51AA5796CBFBD5944981718, then , it's correctly descrambled from 30s to 45s
and to finish, if I put4th CW , DB39BFB9D27B8E4DD0E1C9829DB62B2A, then, it's correctly  descrambled from 45s to 60s

Until now, this is 100% logical. As long as I put only ONE line in the CW file, it means that this CW is used to descramble ALL crypto period... and so, it's normal that correct descrambling is working ONLY for the crypto period in which this CW is corresponding.

But now, the strange point
if I put the following 4 lines in fix_cw.txt
B3B83F7A7DC95262718B6E188CAD3629
F16F7EB000E9804366B066D15625E271
C69AE040F51AA5796CBFBD5944981718
DB39BFB9D27B8E4DD0E1C9829DB62B2A

then, the result I'm expecting is that the file shall be descrambled from 0 to 60s... do you agree ?
This is not at all the result I have.. In this case, descrambling is NOT working at all. I have only a few macroblocks in the picture, and also bad sound from 0 to 60s.
it looks then that part of the TP are correctly descrambled, and some part are not...

am I doing an error ? if no, can you explain the result...

thanks in advance for your help

lelegard, 2019-01-31:
Hi Marc,

I have one possible explanation for what you are seeing. If the hypothesis is true, this is a normal behavior. Please check if the explanation below makes sense in your context.

From your professional background, you obviously know how a CAS and a descrambler work but maybe not everyone is that familiar, so let me explain since it can be a bit tricky. 

To descramble a stream, you need to configure an even CW and an odd CW. Depending on the _transport_scrambling_control_ value in the TS packet header, the descrambler use either the even or odd one.

The CA synchronization is usually achieved on the reception of a new ECM (new table id, either 0x80 or 0x81). The ECM is provided to the CAS library (and traditionally to a smart-card) and the returned even and odd CW are set to the descrambler. Note that, nowadays, most CAS return an encrypted value of the CW's and some key ladder or hardware mechanism inside the chipset is reponsible for providing the final CW's to the hardware descrambler.

Here, the point is to determine _the moment_ where the CW's are provided to the descrambler. On a traditional CAS stream, the even and odd CW are both set upon reception of an ECM and not modified until an ECM with new table id is received. Since at least one ECM is caught in each crypto-period, everything works fine.

But in your case, you do not use CW from ECM's. You use a file of CW's. So, the important question is: when do we select the next CW from the list? The answer is RTFM as usual:

> --cw-file name
> Specifies a text file containing the list of control words to apply. Each line of the file must contain exactly 16 hexadecimal digits (32 digits with `--atis-idsa`). The next control word is used each time a new _transport_scrambling_control_ value is found in the header of a TS packet. At the end of the list of control words, restart with the first one.

This does not really match the situation of a real stream. Especially, at a crypto-period boundary, the audio and video PID's may not exactly switch exactly at the same time. So, let's call V and A the video and audio PID's and CW1 and CW2 the control words for two successive crypto-periods (one of them is even, the other one is odd, whichever the order is). It is quite possible that the stream contains packets in the following order:

~~~
... (a) A-CW1 (b) V-CW1 (c) A-CW2 (d) V-CW1 (e) V-CW2 (f) A-CW2 (g) ...
~~~

In the example, the video PID switches to CW2 one packet later than the audio PID. With a standard CAS, it works because the CW1 and CW2 (even/odd or odd/even) were set ahead and no longer modified until the next crypto-period. But if we apply the `--cw-file` rules, we use the second CW at point (c), third CW at point (d), fourth CW at point (e). The result is obviously not the expected one.

Of course, this scenario is hypothetical and may not match your case. But it seems a plausible one which explains the behavior you see.

If we think globally, the scrambling and descrambling systems shall be consistent with each other. We have two possible scenarios:

- Scramble and descramble using a CAS, based on ECM synchronization. With TSDuck, this is possible either using the CAS emulator which consists in `tsecmg` as ECMG emulator and plugins `scrambler` and `descrambler` without fixed CW. The plugin `descrambler` understands the format of the fake ECM's from `tsecmg`. This is typically used to test a MUX or a transmission system.
- Scramble and descramble using TSDuck plugins `scrambler` and `descrambler` with option `--cw-file`. In that case, the crypto-period transition is neat. This mainly works on stream files, not on live streams. Note that the fix from #207 is required.

But if you scramble with a (real or emulated) CAS and ECM synchronization but descramble with plugin `descrambler` and option `--cw-file`, then you mix the two scenarios and it may not work.

marcabramson, 2019-01-31:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
if I understand correctly (and I think I do) if I try to descramble a component instead  of the whole service, using -p option, it shall work.. do you agree ?
than, iuf I want to descramble the whole service, i only have to run the descrambler many time, one time for each component...

I will try that and tell you the result

best regards
Marc

Le jeudi 31 janvier 2019 à 10:39:48, vous écriviez :

Hi Marc,
I have one possible explanation for what you are seeing. If the hypothesis is true, this is a normal behavior. Please check if the explanation below makes sense in your context.
From your professional background, you obviously know how a CAS and a descrambler work but maybe not everyone is that familiar, so let me explain since it can be a bit tricky.
To descramble a stream, you need to configure an even CW and an odd CW. Depending on the transport_scrambling_control value in the TS packet header, the descrambler use either the even or odd one.
The CA synchronization is usually achieved on the reception of a new ECM (new table id, either 0x80 or 0x81). The ECM is provided to the CAS library (and traditionally to a smart-card) and the returned even and odd CW are set to the descrambler. Note that, nowadays, most CAS return an encrypted value of the CW's and some key ladder or hardware mechanism inside the chipset is reponsible for providing the final CW's to the hardware descrambler.
Here, the point is to determine the moment where the CW's are provided to the descrambler. On a traditional CAS stream, the even and odd CW are both set upon reception of an ECM and not modified until an ECM with new table id is received. Since at least one ECM is caught in each crypto-period, everything works fine.
But in your case, you do not use CW from ECM's. You use a file of CW's. So, the important question is: when do we select the next CW from the list? The answer is RTFM as usual:

--cw-file name
Specifies a text file containing the list of control words to apply. Each line of the file must contain exactly 16 hexadecimal digits (32 digits with --atis-idsa). The next control word is used each time a new transport_scrambling_control value is found in the header of a TS packet. At the end of the list of control words, restart with the first one.

This does not really match the situation of a real stream. Especially, at a crypto-period boundary, the audio and video PID's may not exactly switch exactly at the same time. So, let's call V and A the video and audio PID's and CW1 and CW2 the control words for two successive crypto-periods (one of them is even, the other one is odd, whichever the order is). It is quite possible that the stream contains packets in the following order:
... (a) A-CW1 (b) V-CW1 (c) A-CW2 (d) V-CW1 (e) V-CW2 (f) A-CW2 (g) ...
In the example, the video PID switches to CW2 one packet later than the audio PID. With a standard CAS, it works because the CW1 and CW2 (even/odd or odd/even) were set ahead and no longer modified until the next crypto-period. But if we apply the --cw-file rules, we use the second CW at point (c), third CW at point (d), fourth CW at point (e). The result is obviously not the expected one.
Of course, this scenario is hypothetical and may not match your case. But it seems a plausible one which explains the behavior you see.
If we think globally, the scrambling and descrambling systems shall be consistent with each other. We have two possible scenarios:

Scramble and descramble using a CAS, based on ECM synchronization. With TSDuck, this is possible either using the CAS emulator which consists in tsecmg as ECMG emulator and plugins scrambler and descrambler without fixed CW. The plugin descrambler understands the format of the fake ECM's from tsecmg. This is typically used to test a MUX or a transmission system.
Scramble and descramble using TSDuck plugins scrambler and descrambler with option --cw-file. In that case, the crypto-period transition is neat. This mainly works on stream files, not on live streams. Note that the fix from #207 is required.

But if you scramble with a (real or emulated) CAS and ECM synchronization but descramble with plugin descrambler and option --cw-file, then you mix the two scenarios and it may not work.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-01-31:
Hi again
I tried to descramble only the video
so, I start with a fix_cw.txt containing only the first key B3B83F7A7DC95262718B6E188CAD3629 and the command line being
tsp --verbose -P descrambler --atis-idsa --cw-file fix_cw.txt -p 2324 -I file scrambled.ts -O file clear.ts

and I check, and yes, the first 15 of the video (and only video) were correctly descrambled

I also put only 2nd CW (F16F7EB000E9804366B066D15625E271) in the fix_cw.txt, , re run the descambler, and then the video (and only the video ) was correctly descrambled from 15s to 30s

but, then, I put the 2 CW in the fix_cw.txt which is then
B3B83F7A7DC95262718B6E188CAD3629
F16F7EB000E9804366B066D15625E271
re- run tsp -descramble a 3rd time, expecting that the first 30 s of video shall now be in clear

and the result is: no mode video in clear at all.. (and no macroblock at all). full dark screen.

so, even if I think your explanation was brilliant , and was explaining why we have some partial  descrambling, it does not explain why, selecting only ONE pid, it's not working... or did I mis-understand (again) something...

marcabramson, 2019-01-31:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
just seen that I can probably use srmove plugin to remove the emm and ecm... I will clean the ts and send it to you ? how shll I send it ?

BR
Marc

Le jeudi 31 janvier 2019 à 11:28:07, vous écriviez :

Please post the sample stream. I cannot give a better explanation without it.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-01-31:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
stream is now CAS cleaned and ready to be sent to you. just tell me how (shall I use your direct email adress)
Br
Marc

Le jeudi 31 janvier 2019 à 11:28:07, vous écriviez :

Please post the sample stream. I cannot give a better explanation without it.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

lelegard, 2019-01-31:
Depending on the file size, you may attach it to a post here, send it by email or use a file transfer site such as dl.free.fr or mega.nz.

marcabramson, 2019-02-01:
Hello again
based on my experience, I might have an explanation (which is also depending on your implementation)
I have often seen in real life tha scrambler have problem, and, for unknow reason, they "often" drop some tp and forget to scramble them.. for example, you might have one non scrambled tp every 100 tp (so, 99 % scrambled). from an "viewer" point of view,  this can not been seen, because one clear tp does not allow to get any picture..
if your implementation is: changing the CW line each time the TSC bit is changing (even when going from clear to scrambled), then, this means that this clear packet will prevent any correct scrambling after this packet...
my opinion is that going to the next CW shall be done ONLY when switching from odd to even (or even to odd) or when there are more than  N consecutive clear TPs between scramlbled TPs (and this number can be a parameter of the descrambler plugin)

marcabramson, 2019-02-01:
Concerning the problem of TSC bit not "phased" between different ES sharing the same CW in a service, you are 100% right. But this can also be solved by changing your descrambling strategy. Instead of going to the next CW from the cf-file each time the TSC are changing, you can do the following
1) start by reading all the CW from the CW file and keep them in a CW table
2) have a CW_index different for each ES, and, for each es, memorize the last TSC (instead of having one GLOBAL CW for all es)
3) increment the CW_index of an ES when the TSC for this ES change, and use the CW from the CW_table corresponding to this es_CW_index  to descramble.
I unfortunately don't have time to dig in your source code to implement this myself (and correctly following your 221 coding convention rules) (I don't want to create bad side effect or regression by doing an incorrect modification of the code)

lelegard, 2019-02-01:
Hi,

I cannot descramble anything from the file you uploaded. I guess that the CW are incorrect.

Beyond the global descrambling, it is relatively easy to check the validity of a CW with TSDuck. In all PES PID's, a TS packet with PUSI set contains a PES header. And a PES header always start with 00 00 01. So, simply extract one packet with PUSI from the video PID and descramble it. If the payload starts with 00 00 01, the CW is good. Otherwise, it is not.

Example dumping the first TS packet with PUSI in your stream:
~~~
$ tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 2, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 90 30 31 72 B2 10 B3 92 2B 60 48 49 2D 38 88 58 04 54 3C 59 69
  91 79 E3 15 AE C1 37 48 12 A1 14 27 25 50 DD EE 05 4D 4F 8C 3B ED 16 82
  0D 98 AC 88 86 39 A0 63 7B FE 66 D9 5A B8 9F 82 CE D5 DF 48 A0 25 53 C4
  B2 2D AF 82 C8 EF 78 DB 7E 62 AC C7 16 CB 11 E3 D0 F3 BA B9 4E 05 4B 85
  F8 51 9D 47 88 2E EC 02 0B D1 12 E2 5A 75 BF 90 7A 60 E1 2B 76 19 CF 5E
  11 96 72 E0 7B 93 0C D2 E1 6F 83 EB 43 D1 F1 13 03 E4 F6 1F EF 80 43 6B
  17 4C FF F0 CF E6 AA B3 28 B7 17 69 9D C4 CC AE 50 06 D4 70 44 FE B0 11
  08 0D 0D 1D 52 DF B0 8B 76 E4 DF CB C6 C7 34 35 68 C1 7C D3
~~~

We see that the packet is scrambled with the transport scrambling control value 2. Its payload starts at the 5th byte.

In the following command, we descramble this packet with each CW from file `fix_cw.txt` sequentially. In no case, the payload starts with 00 00 01. So, no CW matches.
~~~
$ for cw in $(cat fix_cw.txt); do tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 -P descrambler --pid 0x914 --atis-idsa --cw $cw | tsdump; done

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 10 78 54 3D 12 7D F5 74 8A 6C 23 F9 56 96 0E CB 73 BC 6D 93 7B
  AB BB E9 58 9F CB 9C DA E0 2D 87 B2 35 C1 42 53 99 FC 90 F4 83 34 14 30
  19 18 0D D1 D6 B6 3C F0 FA 58 17 5A 1C CF D0 87 73 C7 2B D0 BA FE AD C2
  37 EC 7D 51 A8 8A B1 BA 39 9B 1A 2E D9 29 AA 66 A3 39 5D 82 2F 99 41 BD
  05 5B 3A C4 26 E5 ED 13 D5 A3 6A 3C DB 2C 95 C8 EA AF E6 39 FA B4 5D 1D
  85 44 3C 27 2F 41 EE EE 61 F8 8D 2A EC 65 43 40 89 E9 B9 D7 70 3F CD 44
  AA 75 D8 7B 75 13 B1 DD 23 9E DC C1 C0 C7 8D A9 C6 FE 15 99 1F 82 32 BF
  4A 02 EA C0 7E 3D 90 BE D3 6F B3 35 23 42 B0 DF C6 2B 00 6A

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 10 D3 1C A7 68 18 D3 6A 40 A4 0B 06 B9 82 31 82 2E 1F 0B 4D 37
  D2 FA 09 49 2C E1 E1 C6 C3 15 10 E8 62 11 86 42 35 B6 7F 97 78 1F A0 7F
  6B FB 8D 48 86 17 42 F5 5F 49 AF 13 4D D0 58 F9 CE 45 DF 8C 35 F7 0D A2
  5C B4 B7 79 4B EF C6 49 1F 65 14 1D 3D 41 15 78 D3 12 32 2A A0 3F FD BE
  C9 15 41 F7 95 F9 0C 98 BE 3B D1 52 F4 94 3C 4A 6E 52 A5 9F 96 64 25 E0
  E3 E5 7D 4C C9 20 E4 22 D7 3E 35 8D 86 C0 B9 56 0F 7C 09 0F 12 C7 82 25
  98 43 E0 FA AE 40 7F 01 BC 5D 8B 7B 40 7E 99 B8 F2 3B 03 69 1B B2 BE 83
  3F 85 2D 37 BE 5D 26 75 07 42 87 E4 60 C0 2A 26 C3 04 EC EB

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 10 97 DF 59 02 B7 21 E5 64 53 A0 96 38 28 38 F8 5E D6 D5 F7 9D
  0D D4 ED 78 2F E0 2B D8 4D 19 95 9A 5C 12 29 89 94 5F E3 33 64 09 3A E6
  91 D0 2E 11 32 56 D0 BE D3 D9 46 AE 6E 89 8E 61 05 66 EF 8B 21 89 1C B5
  AD 9F 71 04 02 CC B1 C1 F8 41 2C 1D 9C 83 3E FB 50 B3 7C 7B E6 13 D3 79
  04 1A A9 8B 6C C5 DA 0D CA 07 B2 19 D7 6D 3C B2 05 01 A4 AB 8C 5D 70 CF
  68 BA 22 F9 96 B7 3D BA B7 F6 6C CE 17 DA 16 D9 FC F0 25 E2 BC E2 79 C4
  A0 E2 FB 75 45 5C FC B8 F5 34 F6 CE F4 4A 96 95 17 17 67 C1 E1 74 4A D0
  76 F0 99 18 CC D4 74 DE 1B FC 55 A8 0F 42 23 E6 78 B8 BD 2F

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 10 B7 EF 7E 5F D1 DA 0C C2 92 A3 23 C3 5D 8F 98 28 A7 D2 59 92
  9B F5 42 66 1B BF 6A 99 EF 39 6F 4F 77 FA A9 38 DE D0 E5 A8 5A A0 24 BA
  11 A5 FE FB 55 14 3A 39 07 64 B8 1A F4 1A A2 60 C7 CC F8 5E 08 36 93 D0
  51 26 BC D8 FF F9 27 EB 9C 89 4F CC 17 08 96 4B 76 0B 0B 97 93 31 67 0C
  07 E8 B3 79 06 75 CF 06 1F 91 6E BF 9B 9E 04 30 73 A9 E4 B2 77 90 D9 22
  52 16 7F 85 71 8A 7E 15 04 83 50 36 82 6B EC 97 2C F5 24 60 0B D1 8F 14
  27 01 AD F0 51 D8 12 2F 17 EA 68 52 C9 C4 38 9F DD 39 B5 5E 19 70 01 09
  35 D9 39 7F F5 D3 A4 82 30 83 04 FC C9 1A F1 C7 11 B9 E9 15
~~~

I suggest that you use such test to validate your CW.

marcabramson, 2019-02-01:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
sorry, My fault
when I try to descramble the ORIGINAL ts (the on which I have not removed the ecm and emm using TsDuck and filter plugin with a fix_cw.txt containing the first CW), descramling works 100% fine...
but if I put 2 CW (or more) it does not/
when I try to descramble the ts that I have created by removing the emm and ecm, still by tsduck, then, you're right, descrambling does not work).

I will look why and send you a new proper ts

BR
Marc

Le vendredi 1 février 2019 à 12:37:06, vous écriviez :

Hi,
I cannot descramble anything from the file you uploaded. I guess that the CW are incorrect.
Beyond the global descrambling, it is relatively easy to check the validity of a CW with TSDuck. In all PES PID's, a TS packet with PUSI set contains a PES header. And a PES header always start with 00 00 01. So, simply extract one packet with PUSI from the video PID and descramble it. If the payload starts with 00 00 01, the CW is good. Otherwise, it is not.
Example dumping the first TS packet with PUSI in your stream:
$ tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 2, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 90 30 31 72 B2 10 B3 92 2B 60 48 49 2D 38 88 58 04 54 3C 59 69
  91 79 E3 15 AE C1 37 48 12 A1 14 27 25 50 DD EE 05 4D 4F 8C 3B ED 16 82
  0D 98 AC 88 86 39 A0 63 7B FE 66 D9 5A B8 9F 82 CE D5 DF 48 A0 25 53 C4
  B2 2D AF 82 C8 EF 78 DB 7E 62 AC C7 16 CB 11 E3 D0 F3 BA B9 4E 05 4B 85
  F8 51 9D 47 88 2E EC 02 0B D1 12 E2 5A 75 BF 90 7A 60 E1 2B 76 19 CF 5E
  11 96 72 E0 7B 93 0C D2 E1 6F 83 EB 43 D1 F1 13 03 E4 F6 1F EF 80 43 6B
  17 4C FF F0 CF E6 AA B3 28 B7 17 69 9D C4 CC AE 50 06 D4 70 44 FE B0 11
  08 0D 0D 1D 52 DF B0 8B 76 E4 DF CB C6 C7 34 35 68 C1 7C D3
We see that the packet is scrambled with the transport scrambling control value 2. Its payload starts at the 5th byte.
In the following command, we descramble this packet with each CW from file fix_cw.txt sequentially. In no case, the payload starts with 00 00 01. So, no CW matches.
$ for cw in $(cat fix_cw.txt); do tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 -P descrambler --pid 0x914 --atis-idsa --cw $cw | tsdump; done

* Packet 0

  ---- TS Header ----

  PID: 2324 (0x0914), header size: 4, sync: 0x47

  Error: 0, unit start: 1, priority: 0

  Scrambling: 0, continuity counter: 0

  Adaptation field: no (0 bytes), payload: yes (184 bytes)

  ---- Full TS Packet Content ----

  47 49 14 10 78 54 3D 12 7D F5 74 8A 6C 23 F9 56 96 0E CB 73 BC 6D 93 7B

  AB BB E9 58 9F CB 9C DA E0 2D 87 B2 35 C1 42 53 99 FC 90 F4 83 34 14 30

  19 18 0D D1 D6 B6 3C F0 FA 58 17 5A 1C CF D0 87 73 C7 2B D0 BA FE AD C2

  37 EC 7D 51 A8 8A B1 BA 39 9B 1A 2E D9 29 AA 66 A3 39 5D 82 2F 99 41 BD

  05 5B 3A C4 26 E5 ED 13 D5 A3 6A 3C DB 2C 95 C8 EA AF E6 39 FA B4 5D 1D

  85 44 3C 27 2F 41 EE EE 61 F8 8D 2A EC 65 43 40 89 E9 B9 D7 70 3F CD 44

  AA 75 D8 7B 75 13 B1 DD 23 9E DC C1 C0 C7 8D A9 C6 FE 15 99 1F 82 32 BF

  4A 02 EA C0 7E 3D 90 BE D3 6F B3 35 23 42 B0 DF C6 2B 00 6A

[... 70 lines cut ...]
  Adaptation field: no (0 bytes), payload: yes (184 bytes)

  ---- Full TS Packet Content ----

  47 49 14 10 B7 EF 7E 5F D1 DA 0C C2 92 A3 23 C3 5D 8F 98 28 A7 D2 59 92

  9B F5 42 66 1B BF 6A 99 EF 39 6F 4F 77 FA A9 38 DE D0 E5 A8 5A A0 24 BA

  11 A5 FE FB 55 14 3A 39 07 64 B8 1A F4 1A A2 60 C7 CC F8 5E 08 36 93 D0

  51 26 BC D8 FF F9 27 EB 9C 89 4F CC 17 08 96 4B 76 0B 0B 97 93 31 67 0C

  07 E8 B3 79 06 75 CF 06 1F 91 6E BF 9B 9E 04 30 73 A9 E4 B2 77 90 D9 22

  52 16 7F 85 71 8A 7E 15 04 83 50 36 82 6B EC 97 2C F5 24 60 0B D1 8F 14

  27 01 AD F0 51 D8 12 2F 17 EA 68 52 C9 C4 38 9F DD 39 B5 5E 19 70 01 09

  35 D9 39 7F F5 D3 A4 82 30 83 04 FC C9 1A F1 C7 11 B9 E9 15

I suggest that you use such test to validate your CW.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-02-01:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,

I'm sorry, Thierry, I just retried with the file clean_scrambled.ts I have sent you yesterday (I even took it from the zip file)

if I try to descramble using fix_cw.txt containing
B3B83F7A7DC95262718B6E188CAD3629 
ONLY
then, chennel CNN HD AES StSECMob is correctly descramble from 0 to 15 s as shown in this screen copy

now, if I try to descramble with fix_cw.txt containing ONLY
F16F7EB000E9804366B066D15625E271

then, same channel is descrambled from t=15 s to t=30

but if I try to descramble with a fix_cw.txt containing BOTH lines
so
B3B83F7A7DC95262718B6E188CAD3629 
F16F7EB000E9804366B066D15625E271

and that's the problem.
thanks you for the trick to recognise if a program is descrambled without ising a player. this can be usefull, but does not explain why if I put the 2 CW, which are GOOD (otherwise, I would NOT have succeded to descramble at all), it's not working anymore

BR
Marc

Le vendredi 1 février 2019 à 12:37:06, vous écriviez :

Hi,
I cannot descramble anything from the file you uploaded. I guess that the CW are incorrect.
Beyond the global descrambling, it is relatively easy to check the validity of a CW with TSDuck. In all PES PID's, a TS packet with PUSI set contains a PES header. And a PES header always start with 00 00 01. So, simply extract one packet with PUSI from the video PID and descramble it. If the payload starts with 00 00 01, the CW is good. Otherwise, it is not.
Example dumping the first TS packet with PUSI in your stream:
$ tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 | tsdump

* Packet 0
  ---- TS Header ----
  PID: 2324 (0x0914), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 2, continuity counter: 0
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  47 49 14 90 30 31 72 B2 10 B3 92 2B 60 48 49 2D 38 88 58 04 54 3C 59 69
  91 79 E3 15 AE C1 37 48 12 A1 14 27 25 50 DD EE 05 4D 4F 8C 3B ED 16 82
  0D 98 AC 88 86 39 A0 63 7B FE 66 D9 5A B8 9F 82 CE D5 DF 48 A0 25 53 C4
  B2 2D AF 82 C8 EF 78 DB 7E 62 AC C7 16 CB 11 E3 D0 F3 BA B9 4E 05 4B 85
  F8 51 9D 47 88 2E EC 02 0B D1 12 E2 5A 75 BF 90 7A 60 E1 2B 76 19 CF 5E
  11 96 72 E0 7B 93 0C D2 E1 6F 83 EB 43 D1 F1 13 03 E4 F6 1F EF 80 43 6B
  17 4C FF F0 CF E6 AA B3 28 B7 17 69 9D C4 CC AE 50 06 D4 70 44 FE B0 11
  08 0D 0D 1D 52 DF B0 8B 76 E4 DF CB C6 C7 34 35 68 C1 7C D3
We see that the packet is scrambled with the transport scrambling control value 2. Its payload starts at the 5th byte.
In the following command, we descramble this packet with each CW from file fix_cw.txt sequentially. In no case, the payload starts with 00 00 01. So, no CW matches.
$ for cw in $(cat fix_cw.txt); do tsp -I file clean_scrambled.ts -P filter --pid 0x914 -P filter --unit-start -P until --packets 1 -P descrambler --pid 0x914 --atis-idsa --cw $cw | tsdump; done

* Packet 0

  ---- TS Header ----

  PID: 2324 (0x0914), header size: 4, sync: 0x47

  Error: 0, unit start: 1, priority: 0

  Scrambling: 0, continuity counter: 0

  Adaptation field: no (0 bytes), payload: yes (184 bytes)

  ---- Full TS Packet Content ----

  47 49 14 10 78 54 3D 12 7D F5 74 8A 6C 23 F9 56 96 0E CB 73 BC 6D 93 7B

[... 84 lines cut ...]
  Adaptation field: no (0 bytes), payload: yes (184 bytes)

  ---- Full TS Packet Content ----

  47 49 14 10 B7 EF 7E 5F D1 DA 0C C2 92 A3 23 C3 5D 8F 98 28 A7 D2 59 92

  9B F5 42 66 1B BF 6A 99 EF 39 6F 4F 77 FA A9 38 DE D0 E5 A8 5A A0 24 BA

  11 A5 FE FB 55 14 3A 39 07 64 B8 1A F4 1A A2 60 C7 CC F8 5E 08 36 93 D0

  51 26 BC D8 FF F9 27 EB 9C 89 4F CC 17 08 96 4B 76 0B 0B 97 93 31 67 0C

  07 E8 B3 79 06 75 CF 06 1F 91 6E BF 9B 9E 04 30 73 A9 E4 B2 77 90 D9 22

  52 16 7F 85 71 8A 7E 15 04 83 50 36 82 6B EC 97 2C F5 24 60 0B D1 8F 14

  27 01 AD F0 51 D8 12 2F 17 EA 68 52 C9 C4 38 9F DD 39 B5 5E 19 70 01 09

  35 D9 39 7F F5 D3 A4 82 30 83 04 FC C9 1A F1 C7 11 B9 E9 15

I suggest that you use such test to validate your CW.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

lelegard, 2019-02-01:
OK, I found the reason. Your CW file is unsynchronized.

Try this command:
~~~
tsp -I file clean_scrambled.ts -P zap 0x044E -P descrambler 0x044E --cw B3B83F7A7DC95262718B6E188CAD3629 --atis -P filter --pid 0x0914 -P filter --unit | tsdump|less
~~~

It descrambles the stream using the first CW in your file. We display all video TS packets with PUSI. You will see that only the 7th packet has a payload starting with 00 00 01.

If you retry the same command without the descrambling, you will see that the transport scrambling control value switches from 2 to 3 at this 7th packet with PUSI.

So, your file starts with a short end of a previous crypto period. The first CW of you file applies to the second crypto period only. So, there is a desynchronization between the stream and your CW file.

Some answers in this thread (including this one) give you some guidelines and recipes to analyze your problem before submitting it here.

marcabramson, 2019-02-01:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
ok. Agreed...  I did try to add another CW before the first, thinking that it might not be the real 1st crypto period, but I did it when I was trying to descramble the whole service (not only the video pid), and, in this case, itw as not solving anything because of the not synnchronised video and audio .. I did not retry it after changing descramling from the whole service to one component only.. My fault.

anyway, I still suggest that an improvement in TSduck could be to handle each ES descrambling separately, having one  different CW index for each es, as I have suggested this morning. This will allow to descramble a whole service even when the change of TSC bit are not perfectly synchronised.

Lot of thanks for your support

BR
Marc

Le vendredi 1 février 2019 à 14:29:49, vous écriviez :

And I forgot to mention that when you use the first CW as fixed CW, you think that you see the first crypto-period while in fact you don't see the first second (or less) and then you see the second crypto-period.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

lelegard, 2019-02-01:
>  I still suggest that an improvement in TSduck could be to handle each ES descrambling separately, having one  different CW index for each es, 

In your case, you can achieve your goal using two `descrambler` plugins in sequence, one on the video PID and one on the audio PID, using the same CW file.

The plugins `scrambler` and `descrambler` were mainly designed to implement the logic of a CAS. The `scrambler` plugin is probably the most complicated plugin in TSDuck because it implements the ECM generation, ECMG interface, ECMG latency and asynchronism, degraded mode management in case of ECMG failure, etc. The `descrambler` plugin is a bit simpler but also implements the logic of a CAS. The standard `descrambler` plugin only implements the "fake CAS" from TSDuck to test transmission systems. But the C++ architecture is such that it is quite simple to derive other implementations.

The options `--cw` and  `--cw-file` are designed for local, simple and low-level tests. They are not designed to handle service management. They are rarely used as you have found out. If you add the fact that alternative methods exist (dedicated `descrambler` instances for audio and video), I will not make these complicated plugins more complex. This is not worth the investment.

lelegard, 2019-02-01:
One last point:

> I have often seen in real life tha scrambler have problem, and, for unknow reason, they "often" drop some tp and forget to scramble them.. for example, you might have one non scrambled tp every 100 [...]

As far as I am concerned, I have never seen this. However, in your specific case, this happens to be the case. The following two commands count all and clear packets in the video PID. In fact, 6% of the packets are left in the clear.
~~~
$ tsp -I file clean_scrambled.ts -P filter --pid 0x0914 -P count -O drop
* count: PID 2324 (0x0914):    339,285 packets
$ tsp -I file clean_scrambled.ts -P filter --pid 0x0914 -P filter --clear -P count -O drop
* count: PID 2324 (0x0914):     20,674 packets
$ 
~~~

Note that the recent "cbcs" mode in DASH and HLS reproduces this behaviour, but intentionally this time.

However, in the case of the `descrambler` plugin, this is not the cause of the issue because clear packets are simply ignored by the descrambler and are not considered as triggers to switch to the next CW. This is consistent with the fact that it works when you skip the initial residual end of crypto-period and work directly on the video PID.

lelegard, 2019-02-01:
... but in fact the 6% clear blocks are those with a short payload, shorter than the AES block size (16 bytes). The following command shows that there is no clear packet with a payload size of 16 bytes or more:
~~~
$ tsp -I file clean_scrambled.ts -P filter --pid 0x0914 -P filter --clear -P filter --min-payload-size 16 -P count -O drop
$ 
~~~

lelegard, 2019-02-02:
> I have often seen in real life tha scrambler have problem, and, for unknow reason, they "often" drop some tp and forget to scramble them.. for example, you might have one non scrambled tp every 100 [...]

With DVB-CSA2, it is specified that TS payloads which are smaller than the CSA2 block size (8 bytes) must be left clear. This is probably what you have experienced. If you find such a stream, you can use TSDuck to check. Note that, in that case, the scrambler does not "drop" or "forget" the packets. This is the nominal behaviour.

With ATIS, short blocks (smaller than the AES block size) are scrambled. The method is described in the ATIS standard. So, you should not see such clear packets. In your stream, we have seen that _all_ clear packets have a short payload (less than 16 bytes). This is not standard and your ATIS scrambler does not work as specified in the standard. (Well, I should check TSDuck too...) But, the resulting stream remains valid. Maybe the scrambler does this to avoid problems with descrambling short blocks since the scrambling method for these short blocks is rather exotic.

marcabramson, 2019-02-04:
for your info

Bonjour Thierry,
I checkes tsduck for you
As you know, I have scrambled one cleam stream using TSDuck like that
tsp --verbose -P scrambler --atis-idsa --cw-file cw_fix.txt -d 10  2001 -I file clear_with_pmt_scramble.ts -O file stream_scramble.ts

when I use your line to count the packet into this stream
tsp -I file .\stream_scramble.ts -P filter --pid 0x83e -P count -O drop
* count: PID 2110 (0x083E):    694,593 packets
tsp -I file .\stream_scramble.ts -P filter --pid 0x83e -P filter --clear -P count -O drop
* count: PID 2110 (0x083E):     43,739 packets

so, we can see that also 6.2% of the packet  are in clear.... it seams that Ts Duck is doing the same thing as the scrambler used in case of my other stream.

BR
Marc

Le samedi 2 février 2019 à 15:46:51, vous écriviez :

I have often seen in real life tha scrambler have problem, and, for unknow reason, they "often" drop some tp and forget to scramble them.. for example, you might have one non scrambled tp every 100 tp (so, 99 % scrambled). from an "viewer" point of view, this can not been seen, because one clear tp does not allow to get any picture..

With DVB-CSA2, it is specified that TS payloads which are smaller than the CSA2 block size (8 bytes) must be left clear. This is probably what you have experienced. If you find such a stream, you can use TSDuck to check. Note that, in that case, the scrambler does not "drop" or "forget" the packets. This is the nominal behaviour.
With ATIS, short blocks (smaller than the AES block size) are scrambled. The method is described in the ATIS standard. So, you should not see such clear packets. In your stream, we have seen that all clear packets have a short payload (less than 16 bytes). This is not standard and your ATIS scrambler does not work as specified in the standard. (Well, I should check TSDuck too...) But, the resulting stream remains valid. Maybe the scrambler does this to avoid problems with descrambling short blocks since the scrambling method for these short blocks is rather exotic.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-02-04:
Re: [tsduck/tsduck] problem with AES ATIS descrambling (again) (#208)

Bonjour Thierry,
just added some command to see the adaptation field length 

E:\streams\test_crypt>tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.17-1116

tsp -I file clear_with_pmt_scramble.ts  -P filter --pid 0x83e -P filter --min-adaptation-field 0xB7 -P count -O drop
* count: PID 2110 (0x083E):     43,739 packets

tsp -I file clear_with_pmt_scramble.ts -P scrambler 0x7D1 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x83e -P count -O drop
* count: PID 2110 (0x083E):    694,593 packets

tsp -I file clear_with_pmt_scramble.ts -P scrambler 0x7D1 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x83e -P filter --max-payload 15 -P count -O drop
* count: PID 2110 (0x083E):     45,005 packets

tsp -I file clear_with_pmt_scramble.ts -P scrambler 0x7D1 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x83e -P filter --clear -P count -O drop
* count: PID 2110 (0x083E):     43,739 packets

I think the reason why there are so many packet in clear is that, when the adaptation field length is 0xB7, which mean that there is no payload at all ,  TsDuck decide to keep this packet in clear (witch is normal since there is not payload)

BR
Marc

Le lundi 4 février 2019 à 11:25:19, vous écriviez :

Hi Marc,
Sorry, but I do not see the same thing:
$ tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.17-1116
$
$ tsp -I file cnews2.ts -P scrambler 0x0204 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x01A4 -P count -O drop
* count: PID  420 (0x01A4):    449,033 packets
$
$ tsp -I file cnews2.ts -P scrambler 0x0204 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x01A4 -P filter --max-payload 15 -P count -O drop
* count: PID  420 (0x01A4):        278 packets
$
$ tsp -I file cnews2.ts -P scrambler 0x0204 --atis --cw-file cw_file.txt -d 4 -P filter --pid 0x01A4 -P filter --clear -P count -O drop
$
You can see that the video PID of the service has 449,033 packets, including 278 with a "short payload". After scrambing with -P scrambler --atis, no clear packet is left.
This is consistent with the code where no specific check is done for short blocks. In the DVS042 template class, a short block is processed as other blocks (ATIS IDSA is a simple transposition of DVS-042 with AES). You must have done something wrong with your files. The file you are analysing must have been scrambled outside TSDuck.
Note that, with -P scrambler --dvb-csa2, you will find that all packets are marked scrambled as well, including packets with short payloads (1 to 7 bytes). But in that case, the payload is left clear. This is an interesting side effect of the DVB-CSA2 spec which does not say that short block shall be left clear but says that encrypting these blocks is a direct copy. So, when encryping a short payload with DVB-CSA2, you don't modify the payload and you are free to set the packet as scrambled or clear, both are valid.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

closed 2019-02-11 by lelegard