tsduck/tsduck #209: Excellent library. Will it work with an ATSC tuner?

oviano, 2019-02-02:
I am in the UK so I can only test with DVB-T/T2. Limited tests so far suggest It works with a Hauppauge WinSolo HD, so would it be a reasonable assumption that something similar in the US (http://www.hauppauge.com/pages/products/data_dualhd.html) would work with the ATSC system?

lelegard, 2019-02-02:
Yes, ATSC is supported on Windows and Linux. I have, however, no experience on direct ATSC reception in the US. If you have the opportunity to test, please report your feedback here.

oviano, 2019-02-02:
Ok, I will feedback when I get the chance. Probably won't be for a few months though.

lelegard, 2019-03-16:
Some recent feedbacks in other issues report that ATSC tuners are now fully functional (mostly thanks to you @oviano by the way). Closing this issue.

closed 2019-03-16 by lelegard