tsduck/tsduck #210: Thinking if integrating TSDuck library into my project...

oviano, 2019-02-02:
...my project is emustream.tv.

I'm looking at adding DTV tuner support. Currently I'm using the FFmpeg API but there isn't proper support for DTV tuners. I discovered this library and it seems to tick all the boxes I require. I ran a simple capture using tsp and it worked fine.

At the moment, my code roughly works as follows:

- I have a thread which uses the ffmpeg library to start a raw directshow capture (1).
- I have another thread which uses the ffmpeg library to encode the captured data and write it to HLS segments which are sent to my player (2).

So data captured in (1) is moved directly to thread (2).

With tsduck I would be implementing an alternative for (1) for digital tv tuners, so passing the TS data to be encoded and segmented.

I think the simplest way would be for my app to launch tsp.exe in (1) above and have it grab the TS data and then send the output back to my stdin, and I'll probably take that approach to begin with, but ideally I'd like to have a tighter integration and call the API directly. Looking at the source code for tsp it seems quite straightforward how I could get it work, I'd basically just be setting up the input->process->output chain manually except I think it would probably need another output plugin which calls a callback instead inside which I would be sending the data to thread (2) above for it to be encoded and segmented.

Obviously that would be a plugin that would only be relevant to internal use of the API such as this, so wouldn't make sense to expose.

Does that sound a reasonable approach or is there a better way?

oviano, 2019-02-02:
It needs re-encoding for transmission long-distance over the Internet. My project is explicitly aimed towards that. So the integration would be on my server; the server grabs the stream (directshow currently, but also now hopefully digital tv using TSDuck), compresses it down using the latest Nvidia HEVC encoding tech (very good by the way) and sends to the player.

The reason I thought I would have to set things up with plugins was because I will at minimum need the 'zap' plugin functionality to filter out the TSs that I don't need. I found in my test that when I captured the TS it contained lots of channels. But maybe I can do that more directly in the code too...

lelegard, 2019-02-02:
Then why not:
~~~
tsp -I dvb ... -P zap ... | ffmpeg ...
~~~
or even
~~~
tsp -I dvb ... -P zap ... | ffmpeg ... | tsp -O hls ...
~~~

oviano, 2019-02-02:
Well the first one is essentially what I was suggesting doing except I link into the ffmpeg libraries directly, so I would read from tsp's stdout and write this directly into the ffmpeg library rather than actually launch ffmpeg.

But as an final solution I'd prefer to do things more directly; it seems clunky to be launching a process, and then reading its output back into myself, if there was a way of doing it more directly.

lelegard, 2019-02-02:
So, I recommend to use a `ts::Tuner` object, read the TS, use a section demux to grab the PAT and PMT (and SDT if the service is known by name). Then select audio and video PID's from the PMT and pass them to the ffmpeg library. Drop all other packets. You can use the source code of the `zap` plugin as an example. It may seems difficult at first but, with some experience in TSDuck, it is quite straightforward.

oviano, 2019-02-02:
Great, that sounds like a way forward then. Thank you for your assistance.

oviano, 2019-02-06:
Just to clarify as I'm working on a test class; if I already know the service id, audio pid, and video pid, is it sufficient to get a functioning stream if I pass on the PAT (a modified version only referring to the service I require), the PMT (unmodified) and the PES for the required audio and video pids?

i.e. I do not require any other tables in order to be able to play the stream in a video player?

lelegard, 2019-02-06:
> if I already know the service id, audio pid, and video pid, is it sufficient to get a functioning stream if I pass on the PAT (a modified version only referring to the service I require), the PMT (unmodified) and the PES for the required audio and video pids?

This is fine. You do not need anything else.

To be perfectly standard, you should also cleanup the PMT from any additional unused PID (multiple audio, subtitles, data, ECM's, etc)

oviano, 2019-02-06:
Thank you very much!

oviano, 2019-02-07:
I have a small (probably dumb) question, relating to this code snippet in the zap plugin:

    // If the PMT PID was previously unknown wait for the PMT.
    // If the PMT PID was known but was different, we need to rescan the PMT.

    if (!_service.hasPMTPID(it->second)) {

        if (_service.hasPMTPID()) {

            // The PMT PID was previously known but has changed.

            // We need to rescan the PMT. All PIDs related to the service are erased.

What are the circumstances where a service's PMT PID changes?

lelegard, 2019-02-08:
> What are the circumstances where a service's PMT PID changes?

Typically when the broadcast operator does a global service remapping. This happens typically a few times a year. Among the reasons for service remapping:

- New channels appear.
- Channels disappear (end of broadcasting contract, bankruptcy, etc.)
- Existing channels requesting more bandwidth (new contract for more quality or increased resolution SD -> HD -> 4K), forcing other channels to move to other transponders.
- Channel reorganization (moving premium channels to a transponders with EMM's, optimizing zapping time by moving frequently used channels on the same transponder, etc)
- Acquisition of new frequencies, speading existing channels to other transponders.
- Etc.

The service plan (location of channels on transport streams and technical structure of each channel) is relatively stable but not completely static.

oviano, 2019-02-08:
Thank you, that makes total sense.

closed 2019-02-15 by lelegard