tsduck/tsduck #216: HLS input with missing playlist

joolzg, 2019-02-05:
When trying to pull in a master m3u8 in which one of the sub m3u8 is offline tsp quits.

so in master you have
 #EXTM3U
 #EXT-X-STREAM-INF:PROGRAM-ID=1792382335,BANDWIDTH=4096000
 a-720.m3u8
 #EXT-X-STREAM-INF:PROGRAM-ID=1792382335,BANDWIDTH=2048000
 a-576.m3u8
 #EXT-X-STREAM-INF:PROGRAM-ID=1792382335,BANDWIDTH=1024000
 a-360.m3u8
 
If a-576.m3u8 is offline at the time of starting tsp and gets a 404 returned, you get this

2019-02-05_13:48:11.70646 * tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
 2019-02-05_13:48:11.70662 * regulate: minimum packet burst duration is 10,000,000 nano-seconds
 2019-02-05_13:48:11.75637 * hls: downloaded a.m3u8, master playlist, static, 3 media playlists
 2019-02-05_13:48:11.75650 * hls: selected playlist: a-720.m3u8, 4,096,000 b/s
 2019-02-05_13:48:10.60879 * Error: hls: invalid HLS playlist, does not start with #EXTM3U

And then a quit, what would be nice would for tsp to drop this playlist and use only available.

Maybe a "--ignore-missing"
•Error: hls: invalid HLS playlist, does not start with #EXTM3U

joolzg, 2019-02-05:
Ignore the last line, should not be there..

lelegard, 2019-02-11:
The fix has been pushed last night. When the specified playlist is a master, automatically skip media playlists which fail to load and try the next one in the master playlist (with respect to the selection criteria).

No need for an extra option, this seems a reasonable behaviour for an HLS client.

closed 2019-02-15 by lelegard