tsduck/tsduck #217: Unable to build against development includes/libraries with Windows SDK 17763.0

oviano, 2019-02-06:
I'm not sure about earlier SDKs as I don't have any installed, I will try that next!

------ Rebuild All started: Project: sample-app, Configuration: Release Win32 ------
myexec.cpp
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4499): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4541): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4559): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4630): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4733): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4770): error C2065: 'TRUE': undeclared identifier
Done building project "sample-app.vcxproj" -- FAILED.
------ Rebuild All started: Project: sample-app, Configuration: Release x64 ------
myexec.cpp
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4499): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4541): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4559): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4630): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4733): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4770): error C2065: 'TRUE': undeclared identifier
Done building project "sample-app.vcxproj" -- FAILED.
------ Rebuild All started: Project: sample-app, Configuration: Debug Win32 ------
myexec.cpp
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4499): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4541): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4559): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4630): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4733): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4770): error C2065: 'TRUE': undeclared identifier
Done building project "sample-app.vcxproj" -- FAILED.
------ Rebuild All started: Project: sample-app, Configuration: Debug x64 ------
myexec.cpp
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4499): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4541): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4559): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4630): error C2065: 'TRUE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4733): error C2065: 'FALSE': undeclared identifier
c:\program files (x86)\windows kits\10\include\10.0.17763.0\um\winioctl.h(4770): error C2065: 'TRUE': undeclared identifier
Done building project "sample-app.vcxproj" -- FAILED.
========== Rebuild All: 0 succeeded, 4 failed, 0 skipped ==========

oviano, 2019-02-06:
Just confirming that it builds fine against SDK 15063.0.

I'm using Windows 10 Pro 1809, Visual Studio Community 2017 15.9.6.

oviano, 2019-02-07:
I think it relates to these lines in tsPlatform.h:

#if defined(TRUE)
#undef TRUE
#endif

#if defined(FALSE)
#undef FALSE
#endif

With SDK 15063 nothing further in the tsduck library references TRUE or FALSE even via the included windows header files, whereas this is not the case under 17763.

I also can get a similar compile error under 15063 if I include certain Windows header files after tsduck.h. If I move the include for tsduck.h to after any Windows header files, then you get a re-definition error about WIN32_LEAN_AND_MEAN. You can probably fix that with an ifdef defined wrap.

My feeling though is that undefining these is likely to cause ongoing problems - might be better to rename your own types to avoid clashing?

oviano, 2019-02-07:
Ok, so I found that if you just remove the undefs above, and change the TriState enum to use YES and NO instead, then fix up a small number of places in the code, it will all build under 17763.

So there is one possible fix for you :o)

lelegard, 2019-02-07:
Hi @oviano,

As so many people before, you discover that the Windows system API is insane. Defining (`#define`) common words such as `min`, `max`, `TRUE` or `FALSE` is stupid. Do you realize that Windows defines this:
~~~
#define max(a,b)     (((a) > (b)) ? (a) : (b))
~~~

First, even in C, this breaks all programming rules because the macro parameters are used more than once (what about `max(x++, y--)` ?). Then in C++, `min()` is a standard function. So, the perfectly valid C++ expression:
~~~
x = std::max(x,y);
~~~
becomes the following invalid code:
~~~
x = std::(((x) > (y)) ? (x) : (y));
~~~

So, cleaning up insane Windows definitions is not only required, this is a duty 😉 . This is especially true with portable code such as TSDuck. With TSDuck, application code is not supposed to directly call Windows API.

I have made a few modifications which should have been done in the first place such as guarding definitions for `WIN32_LEAN_AND_MEAN` and others using appropriate `#if !defined()`. About the rest, you have to live with Windows...

oviano, 2019-02-07:
Yeah it's crazy, I 100% agree that what the WIN API is doing is bad.

The trouble is though, you have a situation now where your own project does not build under the latest Windows SDK so.... :o)

"With TSDuck, application code is not supposed to directly call Windows API."

This seems problematic - surely you would want to allow people to compile and link against your wonderful library while also doing other unrelated stuff with the Windows API?

It's only TRUE or FALSE that is the problem right now, in any case. If you were able to rename just these two you're back building again with no issues.

lelegard, 2019-02-07:
I fixed a few things to compile TSDuck with the SDK 10.0.17763.0. This will be in the next commit.

The difficulty is that Visual Studio 2017 always selects 8.1 as default SDK, even when more recent SDK's are present. You need to specify a precise SDK version to use another one. This is apparently a known issue which annoys many users. I have read that it _may_ be fixed with VS 2019.

I never wanted to hard-code a specific SDK version in TSDuck project files. So, it uses the default one, namely 8.1 with VS 2017. After googleing a bit, I implemented a hacky method in `msvc-common-base.props` which selects the latest available SDK, at least within the 10.0.* range. The way it is done, it should do nothing if the hack no longer works. So, it should be safe for the future.

lelegard, 2019-02-11:
Could you confirm that you can now build with the latest Windows SDK ?

oviano, 2019-02-11:
I am hoping to try this tomorrow.

oviano, 2019-02-11:
So this does build now, or at least the above errors have gone away.

I am experiencing something odd though - I uninstalled the released version of TSDuck, and then built from tsduck from master. I copied all headers + Windows headers flat into an include folder somewhere, copied tsduck.dll and tsduck.lib into a lib folder somewhere, adjusted my project to point towards these folders and to link against tsduck.lib, and it builds except I get link errors referencing the function value() in the enumValue class:

int inversion = ts::SpectralInversionEnum.value(some UTString...)

Also, I get a link error for:

ts::Enumeration::UNKNOWN

and finally for this one, even though I don't directly reference it:

ts::NoPid

This wasn't happening when linking against the release version of tsduck, and I also don't get these link errors when building against tsducklib.lib, it's only when using the tsduck.lib + tsduck.dll combination.

oviano, 2019-02-12:
Ok, I had overlooked the need for _TSDUCKDLL_USE - everything is working now so you may close this issue!

lelegard, 2019-02-12:
> Ok, I had overlooked the need for _TSDUCKDLL_USE

Yes, this is one of the many insanities of Windows development. You need to differently compile the same library code three times: 1) implementing the DLL, 2) using the DLL, 3) implementing and using the static library. This is why I definitely prefer Linux or macOS as a development platform.

In your case, my recommendation is to build the TSDuck installers. This is as simple as executing the PowerShell script `build\Build-Installer.ps1` (the build itself is done here too). Then, install TSDuck and select option _development environment_. Use the instructions [here](https://tsduck.io/doxy/usinglibrary.html) to reference the TSDuck library. All Windows insanities such as the "use DLL" definitions are automatically handled.

closed 2019-02-12 by lelegard