tsduck/tsduck #218: tuner->receive: how to differentiate between error and timeout?

oviano, 2019-02-07:
//!
        //! Receive packets.
        //! @param [out] buffer Address of TS packet receive buffer.
        //! Read only complete 188-byte TS packets.
        //! @param [in] max_packets Maximum number of packets to read in @a buffer.
        //! @param [in] abort If non-zero, invoked when I/O is interrupted
        //! (in case of user-interrupt, return, otherwise retry).
        //! @param [in,out] report Where to report errors.
        //! @return The number of actually received packets (in the range 1 to @a max_packets).
        //! Returning zero means error or end of input.
        //!
        size_t receive(TSPacket* buffer, size_t max_packets, const AbortInterface* abort, Report& report);

I've set a timeout, but if the above function returns zero, I don't see how to determine whether there was an error or the operation timed out?

Maybe a negative number could indicate the timeout was reached? Or a negative number for error?

lelegard, 2019-02-07:
I will check. But please note that making the `Tuner` class portable was a challenge since Windows DirectShow and LinuxTV have very different interfaces and ways of working. LinuxTV is rather straightforward but DirectShow is doing everything in a uselessly complex way (for instance, simply reading packets from a tuner requires to write a complete DirectShow filter!).

DirectShow is also much less reliable than LinuxTV. To use reliable DVB reception applications, I highly recommend to use Linux instead of Windows. Using TSDuck, the source code will work on both platforms but the reception is usually much more stable on Linux.

oviano, 2019-02-07:
Oh that is sad to hear.

Is that true of all Windows apps using DS for tuners then, or something about TSDuck I’m particular.

My test captures seem to have worked ok so far but I’ve not excessively tested things yet.

lelegard, 2019-02-07:
I mean it works on Windows. But if you'd like a 24x7 application, I would recommend Linux. On Windows, I experimented more packet loss than on Linux, for instance. But this is quite understandable if you compare the complexity of the respective frameworks.

On Linux, you have a common `ioctl` interface for all tuner operations. All tuner drivers are compliant. When you want to receive packets, just use the system call `read()` and directly grab your packets from the device through the driver.

On Windows, you need to build a graph of OLE objects which endlessly communicate and push huge buffers of data between each other. The art of killing a fly with a sledgehammer... You cannot expect this to be as reliable as a simple and neat solution.

DirectShow is also unable to read back tuning characteritics. For instance, consider a hardware DVB-T tuner which can reports back the current tuning characteristics. With DVB-T (unlike DVB-S), only the frequency is important. Whatever value you set for constellation, guard interval, etc. it will work. For instance, you may specify 1/32 as guard interval and you can lock the signal even if the actual value is 1/8.

With the same hardware device, once you are locked to a signal, on Linux you can query the current modulation parameters and the correct values will be returned. In the example, you set 1/32 and you get 1/8. Once you know all exact modulation parameters, you can compute the precise bitrate for instance. On Windows, you set 1/32 and you get ... 1/32. In fact, you always get what you set as input, never the actual parameter value. 

You mention _"Windows apps using DS for tuners"_. But what are those applications? The only tuner applications I saw on Windows were proprietary TV watching applications which come with the hardware devices. These applications use their proprietary characteristics to interface their own tuner. If you look at the awful code for the Windows implementation of the class `Tuner`, you will see that the initialization of a tuner tries to build various topologies of DirectShow graphs until one is correctly built. I did that approximately 10 years ago. Using several different tuners, I realized that they would not work with the same type or graph. Using the DirectShow Spy utility, I noticed that the various proprietary TV watching applications were using different types of graph. So, to be generic and to maximize the chance to work with any tuner, TSDuck tries all known forms of graphs until one works. Despite this, 2 years ago, I found another tuner which did not use any form of graph. The proprietary application worked differently. A generic application such as TSDuck is not able to use it.

DirectShow may be fine to build complex media applications. But if your application simply needs to read the raw transport stream out of a tuner, DirectShow is a wonderful piece of crap...

oviano, 2019-02-07:
Thanks for the very interesting information. All very good to know.

"You mention  "Windows apps using DS for tuners". But what are those applications?"

Well I meant future applications, that might be tempted to use your library. Your library was the only such library I have found soo far. I was tentatively starting to look at what would be involved in using DShow directly, and found almost no examples of working tuner code and what was there seemed very complicated. Your description above of how you had to use DirectShow Spy to figure stuff out, shows why - it's clearly a mess!

Your library manages to brilliantly conceal all this away, much how FFmpeg/libav does similarly for DS capture amongst many other things, so it really appeals. Already I'm compiling and linking against your library and capturing an HD TS file from my Freeview tuner - the next stage is (as mentioned elsewhere) to figure out the demux and processing of tables, stripping out of unwanted services. Your library has given me a real headstart in adding tuner support to my hobby project - kudos and thank you!

lelegard, 2019-02-11:
Going back the initial question: the timeout. Depending on the operating system, I think that there are several sources of timeouts for TS reception. There is the one you set with `setReceiveTimeout` and other driver-level timeouts. It is difficult to distinguish between the two. The one from `setReceiveTimeout` is a guarantee that the application will not get stuck forever if there is no signal (not all tuners or drivers can correctly report the absence of signal). 

My point is that there are so many different sources of errors or timeouts depending on the system, driver and hardware tuner that it is a bit pointless to try to detect each specific type of errors.

Said otherwise, if you want to discriminate between specific errors, it must be because you want to take different actions depending on errors. But, because of this diversity of cases, there would be no guarantee that your application would behave identically with different systems and tuners.

On a pragmatic standpoint, stay simple and robust. Either you receive packets or you don't. If you don't, just adopt always the same behaviour: exit, retry later, check another frequency, whatever.

oviano, 2019-02-11:
Yes that makes sense, thank you.

closed 2019-02-12 by lelegard