tsduck/tsduck #219: Compilation error on Ubuntu on WSL

davemevans, 2019-02-08:
When attempting to compile on Ubuntu running on Windows Subsystem for Linux, the build fails with some kind of template error

```
$ make NOTEST=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTELETEXT=1
  [CXX] tsAACDescriptor.cpp
In file included from /usr/include/c++/7/bits/move.h:54:0,
                 from /usr/include/c++/7/bits/stl_pair.h:59,
                 from /usr/include/c++/7/bits/stl_algobase.h:64,
                 from /usr/include/c++/7/bits/char_traits.h:39,
                 from /usr/include/c++/7/string:40,
                 from tsPlatform.h:655,
                 from tsByteBlock.h:36,
                 from tsAbstractSignalization.h:36,
                 from tsAbstractDescriptor.h:36,
                 from tsAACDescriptor.h:36,
                 from tsAACDescriptor.cpp:34:
/usr/include/c++/7/type_traits: In instantiation of 'struct std::underlying_type<unsigned char>':
tsxmlElement.h:426:107:   required by substitution of 'template<class ENUM, typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous>, class INT> bool ts::xml::Element::getIntAttribute(ENUM&, const ts::UString&, bool, ENUM, INT, INT) const [with ENUM = unsigned char; typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous> = <missing>; INT = <missing>]'
tsAACDescriptor.cpp:186:88:   required from here
/usr/include/c++/7/type_traits:2242:38: error: 'unsigned char' is not an enumeration type
       typedef __underlying_type(_Tp) type;
                                      ^~~~
../../Makefile.common:898: recipe for target 'release-x86_64/tsAACDescriptor.o' failed
make[2]: *** [release-x86_64/tsAACDescriptor.o] Error 1
```

```

lelegard, 2019-02-08:
Hi,

I do not have any WSL environment. I do not see this error on a true Ubuntu system. Which TSDuck commit are you using?
~~~
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.10
Release:	18.10
Codename:	cosmic
$ gcc --version
gcc (Ubuntu 8.2.0-7ubuntu1) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ grep COMMIT src/libtsduck/tsVersion.h 
#define TS_COMMIT 1120
$
$ make NOTEST=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTELETEXT=1
  [CXX] tsAACDescriptor.cpp
  [CXX] tsAC3Attributes.cpp
  [CXX] tsAC3Descriptor.cpp
  [CXX] tsAC4Descriptor.cpp
  [CXX] tsAES.cpp

~~~

davemevans, 2019-02-08:
I was attempting to build master
```

lelegard, 2019-02-08:
The specific declaration which triggers the error was added recently. This may explain a new behavior.

However, it compiles correctly with GCC 4.8.5 (CentOS 7.6), GCC 8.2.0 (Ubuntu 18.10). Note that it also compiles correctly on LLVM/clang 10.0.0 (macOS Mojave) and Visual Studio 2017 (Windows 10).

Your compilation fails on GCC 7.3.0 on an "hybrid" Windows/Linux environment. But it works on both older and newer versions of GCC, on "true" Linux platforms.

Additionally, when looking at the error message you report, it seems a compiler bug. This specific error should never happen because it would be a violation of the SFINAE principle (C++ templates).

My conclusion is that your specific version of the compiler is buggy. I can only advise to get another one.

davemevans, 2019-02-11:
Build succeeds when using gcc 8.2.0, so your conclusion seems correct.

Thanks for your help.

closed 2019-02-11 by davemevans

lars18th, 2019-04-19:
Hi,

I confirm that the version 3.16 of the TSDuck compiles in Debian WSL with gcc 6.3.0.
Commented here: https://github.com/tsduck/tsduck/issues/231#issuecomment-466795665

So after the `git clone`you can execute `git checkout 00d13f9c4fb7db52e9cf3599afcc03a8facb078b` and compile inside WSL.

I really need this because I prefer to compile/develop inside WSL, as the VC is a pain!
So, if the code of 3.17 can support some Debian distribution (without GCC 8) and circunvent the GCC 6 & 7 bug, then that will be welcome.

Regards.