tsduck/tsduck #220: pcrextract: Offset from PCR

rezna11, 2019-02-08:
Hello. I have a question/request about the existing plugin `pcrextract`.

The `pcrextract` plugin reports "Offset from PCR". This works mostly for video. But I'd like to know if there's a way to report the offset from PCR for streams that don't contain PCR within the same PID.

lelegard, 2019-02-11:
I do not see any way to do so with `pcrextract`. This plugin only considers PID's one by one, independently of the TS structure. When there is no PCR in a PID, you must know the associated PID with PCR. This means that the structure of the services shall be analyzed and PID associations should be built. This is not the case.

Also note that not all video PID's have PCR's. I have seen several times services with a dedicated PCR PID. This PCR PID does not contain any audio or video. There are just a few empty TS packets only containing a PCR. In a media player, all audio and video PID's must be matched against this dedicated PCR PID.

rezna11, 2019-02-11:
Yes, that's what I meant, in some cases it might not work for video either. So the idea was to extend `pcrextract` to include an option to define a separate PCR PID in order to be able to compare PCR with PTS of a PES. But that's just an idea. Otherwise I think this can be done by calculating the corresponding PCR manually, which should be doable with the help of the `pes` and `pcrextract` plugins and some Excel tables...

lars18th, 2019-02-12:
Hi,

Only to understand about it... You know how to generate such type of streams? That's an A/V TS with a PCR in a "empty" PID. I don't see any tool that has support for it.

Thank you!

rezna11, 2019-02-12:
For video, I've seen this happen with NASA UHD for example. So at least some Harmonic encoders can do this. I don't know about a "tool", though, never looked for it.

lelegard, 2019-02-12:
Added the following option:
~~~
-e
--evaluate-pcr-offset
    Evaluate the offset from the PCR to PTS/DTS for packets with PTS/DTS but
    without PCR. This evaluation may be incorrect if the bitrate is not
    constant or incorrectly estimated. By default, the offset is reported
    only for packets containing a PTS/DTS and a PCR.
~~~
But note that this option can be effective only on CBR streams. On packets without PCR, the offset between PTS/DTS and PCR is computed from the theoretical PCR of the packet containing the PTS/DTS. And this theoretical PCR can be precisely computed on CBR streams only. Since the offset between PCR and PTS/DTS is usually small, any imprecision in the computation of the theoretical PCR makes the result not really useable.

lelegard, 2019-02-12:
> Only to understand about it... You know how to generate such type of streams? That's an A/V TS with a PCR in a "empty" PID. I don't see any tool that has support for it.

There is an example in the TSDuck stream repository. This is an [ISDB-Tb stream from Brasil](https://tsduck.io/streams/?name=brazil-isdb-tb).

See this service:
~~~
|=============================================================================|
|  Service: 0xE898 (59544), TS: 0x0744 (1860), Original Netw: 0x0744 (1860)   |
|  Service name: EPTV MOBILE, provider: (unknown)                             |
|  Service type: 0xC0 (unknown)                                               |
|  TS packets: 63,180, PID's: 5 (clear: 5, scrambled: 0)                      |
|  PMT PID: 0x1FC8 (8136), PCR PID: 0x0200 (512)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  unknown (0xC0) ............................... C      366,851 b/s  |
|  0x0200  PCR (not otherwise referenced) ............... C        6,497 b/s  |
|  0x0211  AVC video (320x180, baseline profile, level 1  C      299,548 b/s  |
|  0x0212  HE-AAC Audio (Audio layer I, 256 kb/s, @44,10  C       50,132 b/s  |
|  0x0216  MPEG-2 PES private data ...................... C        3,164 b/s  |
|  0x1FC8  PMT .......................................... C        7,507 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
~~~ 

The PCR PID is a dedicated low-bandwidth PID. 

See the analysis of the audio and video PID's. They contain zero PCR.
~~~
|=============================================================================|
|  PID: 0x0211 (529)                                               AVC video  |
|  PES stream id: 0xE0 (Video 0)                                              |
|  320x180, baseline profile, level 1.3, 4:2:0                                |
|  Service: 0xE898 (59544) EPTV MOBILE                                        |
|-----------------------------------------------------------------------------|
|  Single Service PID        Transport:                Discontinuities:       |
|  Bitrate: ... 299,548 b/s  Packets: ........ 51,589  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ........ 1,569  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  PES:                   |
|                            PCR: ................. 0  Packets: ...... 1,574  |
|                                                      Inv.Start: ........ 0  |
|=============================================================================|
|  PID: 0x0212 (530)                                            HE-AAC Audio  |
|  PES stream id: 0xC0 (Audio 0)                                              |
|  Audio layer I, 256 kb/s, @44,100 Hz, dual channel                          |
|  Service: 0xE898 (59544) EPTV MOBILE                                        |
|-----------------------------------------------------------------------------|
|  Single Service PID        Transport:                Discontinuities:       |
|  Bitrate: .... 50,132 b/s  Packets: ......... 8,634  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ............ 0  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  PES:                   |
|                            PCR: ................. 0  Packets: ...... 2,158  |
|                                                      Inv.Start: ........ 0  |
|=============================================================================|
~~~
Now, see the dedicated PCR PID. All packets contain a PCR and only adaptation field.
~~~
|=============================================================================|
|  PID: 0x0200 (512)                          PCR (not otherwise referenced)  |
|  Service: 0xE898 (59544) EPTV MOBILE                                        |
|-----------------------------------------------------------------------------|
|  Single Service PID        Transport:                Discontinuities:       |
|  Bitrate: ..... 6,497 b/s  Packets: ......... 1,119  Expected: ......... 0  |
|  Access: .......... Clear  Adapt.F.: ........ 1,119  Unexpect: ......... 0  |
|                            Duplicated: .......... 0  Sections:              |
|                            PCR: ............. 1,119  Unit start: ....... 0  |
|=============================================================================|
~~~

lars18th, 2019-02-12:
Thank you!
I'll do some tests with the ISDB-Tb test stream.

rezna11, 2019-02-12:
I tested the new parameter and it does exactly what I had in mind. Thank you!
The only issue I have is that I can't build on Windows, but worked on Linux. Here's a screenshot of the error I get at the end:
[image]

lelegard, 2019-02-12:
This is an error in the Dektec API. Your probably have an obsolete version of the Dektec Windows SDK. Install the latest one from https://www.dektec.com/downloads/SDK/

rezna11, 2019-02-13:
thanks, that helped

closed 2019-02-15 by lelegard