tsduck/tsduck #222: "* Error: configuration file 'tsduck.names' not found" when ever my app linked with tsduck launches

oviano, 2019-02-12:
Not a big deal at all, but maybe this can be suppressed? It appears twice before any call has been made to any tsduck library function.

I know I could probably just add this file to the directory, but for my usage I don't think I need it so it would be better if this failed silently?

lelegard, 2019-02-12:
Thanks for reporting this. This is fixed now. 

It was my original intend to load the names configuration files only when needed. But for some reason, the wrong structure was used and they were always loaded on application startup. Now, they are loaded at their first use. Incidently, since these files are quite large and sometimes rarely used (especially the OUI file), the fix will significantly speed up the application startup time.

Additionally, the file name was wrong. There is no `tsduck.names` file. There is `tsduck.dvb.names` and `tsduck.oui.names`. In fact, in the initial implementation, there was one single `tsduck.names` which was later split in two files but the file name remained hardcoded in the message.

oviano, 2019-02-13:
Thank you, I will test again later.

oviano, 2019-02-13:
Confirmed fixed. Thanks.

closed 2019-02-13 by lelegard