tsduck/tsduck #223: Weird instability when running my app in debug, linked to tsduck libraries

oviano, 2019-02-13:
This is quite possibly not an issue with tsduck, but I'm bit lost.

With a release build all is working ok, I'm using the new channel XML API and it seems to work.

However, when I run my app in debug I get some really basic issues; see attached I'm just passing a valid std::string p into ts::UString::fromUTF8 but when I step into this function the string it has received is garbage. Many levels later it crashes down inside the visual C runtime.

I initially thought this might be because I was linking my debug build against release tsduck libraries, but the same thing happens when linking against the tsduck debug libraries.

The rest of my code functions fine in debug; it's only when calling into the tsduck functions that this seems to be happening. 

[image]
[image]

oviano, 2019-02-13:
Just to add, I only get this issue linking against tsduck.lib for use with the DLL. If I link against tsducklib.lib then it works fine.

lelegard, 2019-02-13:
Well, this is a Visual Studio debugger issue. We cannot really help here. With experience, we all learn that we should not blindly trust debuggers. What they display and what is in C++ objects are not always the same. Try to add traces in the code to check if the string is what it should be. The Microsoft C++ compiler also has code generation bugs whan handling temporary objects (see [here](https://developercommunity.visualstudio.com/content/problem/334025/problem-with-lifetime-of-some-specific-temporary-c.html)). Again, Windows is not the ideal development platform. You will learn 😉

oviano, 2019-02-13:
Thanks - well it's not just when running in the debugger. Simply running the debug exe outside VS produces the same eventual call to abort() due to whatever corruption has occurred (in the end it finds the string it's handling is crazy long).

Maybe it's something in my debug settings, although I cannot see what.

It's also odd that my program functions fine, it's only when I call into the tsduck library that it goes wrong. I've linking against many other third party libraries and I've never seen this (libcurl, libconfig, libav, libVLC, SDL, SDL_ttf, freetype, etc etc).

That's not to say it's caused by tsduck, just that I've not come across it before :o)

lelegard, 2019-02-13:
The libraries you mention are written in C and manipulate simpler objects. The Microsoft C++ compiler is crap. Many people are hit by the code generation bug I mentioned and Microsoft simply responds after 6 months that it _may_ be fixed one year after the report.

Just to check if you are hit by the Microsoft C++ compiler bug, try to first create a ts::UString object which is initialized by you std::string, then pass the ts::UString object to the function.

oviano, 2019-02-13:
I tried that, but it gives the same result i.e. crashes inside FromUTF8.

It works only with a hardcoded u"my path goes here" string. In fact, I replaced all the calls in that function with hardcoded strings to test, but it then crashes inside searchSeviceInternal inside the call to net.clear().

So I think my issue is more fundamental than the Microsoft bug you mention. As I say, if I link to the debug static lib it all works fine, it's only when using the DLL.

lelegard, 2019-02-13:
The same code is compiled twice by the Microsoft C++ compiler: once for the static library and once for the DLL. You say that one works and the other one crashes. Welcome in Windows world... We cannot do anything remotely without diving into the generated code. At best, try to isolate this into one small source code, one small application which reproduces the problem and post the code here.

oviano, 2019-02-13:
Ok, looks like it was my own incompetence; a typo in one of my build scripts meant it ended up trying to run in debug using the release tsduck DLL.

Thank you for taking the time to try and help me and I'm very sorry to waste your time.

lelegard, 2019-02-13:
>  I've seen some packet loss but it also appears using the WinTV software so I'm not sure if that is DS related,

WinTV and other TV watching applications are based on DirectShow.

> a typo in one of my build scripts meant it ended up trying to run in debug using the release tsduck DLL.

> One thing I have discovered is your love of Windows :)

Exactly, the same typo would have been harmless on Linux :))

closed 2019-02-15 by lelegard