tsduck/tsduck #224: Recorded udp stream starts after a few seconds

arka54, 2019-02-13:
I am recording a number of UDP mpegts streams to my system. All recorded videos freezes for a few seconds and motion starts after 2-3 seconds while the recorded audio is correct. My tests was in two different situations:
1. I start 'tsp' using linux 'crontab' at specified time to record the stream and use 'until' plugin to stop stream recording at specified seconds.
2. I start 'tsp' a while before recording time and use 'time' plugin to start/end recording at specified time.
In the both situations the results is same and videos are frozen (motionless) for a few seconds at the start of the file.
Also I used 'ffmpeg' to save same stream on the same server and there is no problem.
I am using CentOS 7 System. SELINUX/Firewall/rp_filter disabled.

tsp -v --debug=4 -I ip 236.1.1.3:1003 -P until -s 600 -O file ch1_ts.ts
debug level set to 4
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 0 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 4
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 1
  Input plugin 1:
    Name: ip
    Type: input
    Arg[0]: "236.1.1.3:1003"
  Packet processor plugin 1:
    Name: until
    Type: packet processor
    Arg[0]: "-s"
    Arg[1]: "600"
  Output plugin 1:
    Name: file
    Type: output
    Arg[0]: "ch1_ts.ts"
* Debug[4]: debug level set to 4
trying to load /usr/bin/tsplugin_ip.so
symbol tspNewProcessor not found in /usr/bin/tsplugin_ip.so
trying to load /usr/bin/tsplugin_file.so
trying to load /usr/bin/tsplugin_until.so
symbol tspNewInput not found in /usr/bin/tsplugin_until.so
symbol tspNewOutput not found in /usr/bin/tsplugin_until.so
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug[4]: ip: debug level set to 4
* Debug[4]: until: debug level set to 4
* Debug[4]: file: debug level set to 4
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 236.1.1.3:1003
* ip: joining multicast group 236.1.1.3:1003 from local address 192.168.126.27
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown input bitrate
* Debug: file: creating file ch1_ts.ts, fd=5, error_code=0
* Debug: ip: input thread started
* Debug: until: packet processing thread started
* Debug: file: output thread started
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003
* Debug[2]: ip: received UDP packet, source: 192.168.100.2:1003, destination: 236.1.1.3:1003

lelegard, 2019-02-13:
What is the GOP size of the stream? Some IP streams have multi-seconds GOP size (against 500 ms for a typical broadcast stream). `tsp` receives and stores raw TS packets, maybe starting in the middle of a GOP while `ffmpeg` processes consistent video and audio streams. So, it is possible that both `tsp` and `ffmpeg` start the reception in the middle of a long GOP but that `ffmpeg` drops the initial partial GOP and starts the actual recording only at the beginning of the next one. This is just an hypothesis of course.

arka54, 2019-02-13:
Thank you for quick response, I don't know how to measure the GOP size of the stream exactly, if there is a tool to do this or I can do it by 'tsp' please guide me.

lelegard, 2019-02-13:
You may use MediaInfo. Or use AviDemux and move from I-frame to I-frame. Or post here the first 10 seconds of the captured TS.

closed 2019-06-04 by lelegard