tsduck/tsduck #225: bitrate_monitor doesn't report zero bitrate

xjoska, 2019-02-13:
Hi,
I am monitoring bitrate of multicast stream by TSP plugin: 
`tsp -I ip 239.7.1.1:1234 -P bitrate_monitor 0x0000 -p 3 -t 3 --min 0 -O drop`
Unfortunately in case that multicast stream disappear, plugin stops report the bitrate.  I would expect the plug-in will report zero bitrate.

Can you please check it?

lelegard, 2019-02-13:
This cannot work like that. `tsp` runs a chain of plugins. Each plugin is activated each time a packet (or set of packets) are passing through it. If no packet is flowing through the chain, no plugin is activated. So, no plugin can do or report anything.

If you want to detect that the stream disappears, you can use the option `--receive-timeout` in the `ip` input plugin.

xjoska, 2019-02-14:
Understood. I tried it with `-receive-timeout 100`, So, once the stream disappeared, the `tsp` logged expected message `* Error: ip: error receiving from UDP socket: Resource temporarily unavailable`, but the process completely exited.
Is there any option how to continuously monitor bitrate of incoming multicast stream and detect that that the stream disappears?
As option I tested reporting directly from IP input plugin (--display-interval and --evaluation-interval), but the behaviour is same.
Thanks.

lelegard, 2019-02-14:
> Is there any option how to continuously monitor bitrate of incoming multicast stream and detect that that the stream disappears?

Not directly this way.

The question is: what kind of action to you want to take when the stream disappears? Is this just log a message and then continue waiting? Repeatedly report an error until the stream reappears?

You may have a look at the command `tsswitch`. It has been designed to handle multiple inputs and handling failover to do things like this:

[image]

lelegard, 2019-03-18:
I have enhanced the plugin API to allow plugins to be notified of packet reception timeout. As an application, the plugin `bitrate_monitor` can now detect a complete absence of input TS (TS bitrate = zero). The appropriate alarms are reported.

xjoska, 2019-03-21:
Great! Thank you.

closed 2019-03-21 by lelegard