tsduck/tsduck #226: wrong PES size reporting in pes plugin

rezna11, 2019-02-18:
Hello. It seems to me that the plugin `pes` with option `--packet-index` reports incorrect "last" packet for some PES packets. The problematic reporting happens sometimes on PES where the length of the payload is within 1 TS packet. 
Here's an example: [ch_test2.zip](https://github.com/tsduck/tsduck/files/2875666/ch_test2.zip)
...the first payload in the PID 4072 should be 1 TS packet long, instead, `pes` plugin reports it to be 357 TS packets. The other 2 PES packets in the sample recording are OK.

[image]

rezna11, 2019-02-18:
By the way, I think that the stream is probably invalid because the payload following the first PES packet shouldn't be there in first place... and maybe that might be an excuse for `pes` plugin not working and getting confused... but still, I just wanted to ask what's the expected behavior of `pes`.

lelegard, 2019-02-18:
You are right, the stream is incorrect. In the PID, the first PES packet fits into one TS packet but it is followed by many garbage TS packets in the same PID.

"Normally", a PES packet is delimited by two PUSI marks in TS headers. The size field in the PES header is typically set to zero in video and (sometimes) audio packets. In that case, the size of the PES packet is simply "everything up to the next PUSI". This is what the PES demux is doing in TSDuck.

In the case where the size field in the PES header is set to an explicit (non-zero) value, as it is the case here since this is a subtitles PID, the PES demux fails to check the consistency of the packet and fails to detect the extra garbage.

I just pushed a fix where this garbage is detected and printed as illustrated below:
~~~
$ tsp -I file ch_test2.ts -O drop -P pes --pid 4072 --packet-index --header
* PID 0x0FE8, stream_id 0xBD (Private stream 1), size: 42 bytes (header: 14, payload: 28), 7845 spurious trailing bytes
  First TS packet: 3,305, last: 3,661
  PES header:
    0000:  00 00 01 BD 00 24 85 80 05 25 F2 51 E4 B1
* PID 0x0FE8, stream_id 0xBD (Private stream 1), size: 42 bytes (header: 14, payload: 28)
  First TS packet: 5,328, last: 5,328
  PES header:
    0000:  00 00 01 BD 00 24 85 80 05 25 F2 5D 46 D1
* PID 0x0FE8, stream_id 0xBD (Private stream 1), size: 5331 bytes (header: 14, payload: 5317)
  First TS packet: 8,155, last: 8,679
  PES header:
    0000:  00 00 01 BD 14 CD 85 80 05 25 F2 6B 4E D1
$
~~~

Note that the packet indexes in _"First TS packet: 3,305, last: 3,661"_ are indexes in the transport stream. This does not mean that the PES packet uses all 357 TS packets in between. There are other PID's.

rezna11, 2019-02-19:
great, thanks!

closed 2019-02-19 by lelegard