tsduck/tsduck #227: WinTV HVR-950 BDA Tuner not detected

oviano, 2019-02-18:
I'm trying to get this one working for a friend in the US.

tslsdvb doesn't find it. It works with WinTV and I've just updated the drivers but it made no difference.

Attaching the output from the various tslsdvb options. Is there anything that can be done?

[bda-tuners.txt](https://github.com/tsduck/tsduck/files/2877114/bda-tuners.txt)
[tuning-spaces.txt](https://github.com/tsduck/tsduck/files/2877115/tuning-spaces.txt)
[enumerate-devices.txt](https://github.com/tsduck/tsduck/files/2877116/enumerate-devices.txt)

oviano, 2019-02-19:
Just adding that we get the same result with a different PC running Windows 8.1.

lelegard, 2019-02-19:
This an endemic problem with tuners on Windows. There is no standard way of interfacing them with DirectShow.  I already discussed that in another thread. 

In your case, this is an hybrid analog + ATSC tuner. I have never used that kind of device. There is probably some kind of specific graph to build. 

The first suggestion is to use DirectShow Spy and get a snapshot of the graph which is used by the TV watching application which is bundled with the tuner.

The second suggestion is to use Linux.

oviano, 2019-02-19:
Right so I've figured out how to get DirectShowSpy working on my own USB tuner, so I will now try and do the same for my friend's.

What information would be useful - an image of the graphs joined together, or the contents of the DirectShow Filter Graph Properties, or both?

oviano, 2019-02-19:
Hmm, on second thoughts I haven't yet figured out how to get the correct filter graph for my own working USB tuner. I am seeing the graph for the WinTV exe but that is just for the decoding and rendering.

The actual DVB-T tuning and capture appears to be done via the background service HauppaugeTVServer. However, I cannot get a filter graph for this at all. It doesn't show up in the DirectShow Spy list. It's installed as a service, so runs under admin, and it's 32-bit, so I'm using the 32-bit DirectShow Spy DLL from an admin command prompt. But nothing shows up. If I try from a non-admin prompt I just get the WinTV exe and IR.exe.

Do you have any experience of grabbing the filter graph from the Hauppauge TV Server service?

Once I've figured out how to get a proper graph of my own working tuner, I hope to apply the same methodology to get the graph for my friend's that doesn't work.

I'm keen to help support this and other tuners - many of the Hauppauge current devices are hybrid analogue/digital models, so it'd be great if tsduck could support these too and I'm ready to help!

oviano, 2019-02-19:
Ok, so I tried the TSReaderLite application with my own tuner, and was able to successfully get the filter graph etc, and this application was able to capture the streams just fine.

I then tried it with my friend's, choosing the "Hauppauge Generic ATSC" option. This seemed to work insofar as it was able to go through the lengthy scanning process, but of course being too far from any transmitter it didn't find any channels. During the scanning however, I grabbed the filter graph information which I am attaching.

[tsreader-dsspy.txt](https://github.com/tsduck/tsduck/files/2882160/tsreader-dsspy.txt)
[image]

oviano, 2019-02-19:
I also tested the same tuner out with EasyHDTV, which specifically claims to support the tuner in question (Hauppauge HVR 950Q).

Here is the filter graph info from that.

[easyhdtv-dsspy.txt](https://github.com/tsduck/tsduck/files/2882357/easyhdtv-dsspy.txt)

[image]

[image]

[image]

oviano, 2019-02-20:
And another set for good luck, this time produced when running the tuner inside jriver media center.

[jriver-dsspy.txt](https://github.com/tsduck/tsduck/files/2884307/jriver-dsspy.txt)

[image]

[image]

lelegard, 2019-02-20:
Thanks for these useful graphs. They would clearly be useful to a developer which has such a tuner to test. This is unfortunately not my case (I am TSDuck's developer but I have no ATSC tuner - and no local ATSC network). See my response in issue #229

oviano, 2019-02-20:
Ok, I understand :o)

In that case, do you have any knowledge of ay ATSC tuners that work reasonably well with tsduck under Windows?

lelegard, 2019-02-20:
I do not know any on Windows. Again, I recommend to try on Linux. If there is an appropriate driver for the tuner on Linux, then it is more likely to work right away. And if there is some issue, it will be way much easier to fix remotely, without physical tuner to test.

I have said that thousands of times in the last ten years: willing to work with TV tuners on Windows is looking for trouble. This is my experience and it has been verified so many times. Now, it is your choice...

oviano, 2019-02-20:
Ok, thanks.

Linux is an option, I have a build of my project it's just a bit out of date. However, in the absence of any concrete evidence of Tsduck working under Linux with ATSC it's probably not a route I'll go down, as I may just end up with a different bunch of graphs and debug trace and no easier way forward :o)

By the way, is tsscan supposed to scan ATSC channels? Only I noticed that in the code it seems to make an explicit check for DVB-T and bails out if it's anything else.

lelegard, 2019-02-20:
> However, in the absence of any concrete evidence of Tsduck working under Linux with ATSC it's probably not a route I'll go down,

_"Concrete evidences"_ appear when someone in the community of TSDuck users make the relevant tests. I cannot do these tests, living in Europe without ATSC device and without ATSC network. TSDuck is not different from other open source projects : if you appreciate help from the community, then please provide some help in return. Here, you could help by testing Linux ATSC devices on an ATSC network. If neither you nor any other US-based user make this effort, then ATSC will never be tested for sure.

> By the way, is tsscan supposed to scan ATSC channels? Only I noticed that in the code it seems to make an explicit check for DVB-T and bails out if it's anything else.

There are many different ways to scan a network. `tsscan` implements what I understand and what I can test. Currently, `tsscan` implements two scanning methods:

- DVB NIT scan : Starting from a known frequency, `tsscan` gets the NIT, collects all delivery system descriptors and then scans all corresponding frequencies. This should work an any properly structured DVB network (not all of them are however).
- UHF-band scan for DVB-T. Here, we use the known conventional structure of UHF channels (21 to 69) to scan all possible frequencies.

How do we scan an ATSC network? Well, I have no idea.... If anyone can explain this or direct to a good documentation on the ATSC scanning process, this would be great. Of course, testing would be required by US-based users.

oviano, 2019-02-20:
I've been trying to help. That's why I spent the last couple of days trying all manner of software via an excruciatingly slow Teamviewer login to my friend in the far north of the US to produce the filter graphs that you suggested to try and help pinpoint why tsduck wasn't working for what is actually a very very common type of tuner in the US and I'm sure would benefit your project in supporting.

But the fact is, I'm not based in the US either. So I cannot test a working ATSC tuner under Linux any more than you can, thus why I would need to be sure there was one that worked before asking my friend to buy another device.

As far as scanning is concerned, from what I saw when testing the software above, it would be a very similar UHF based scan for ATSC, although I haven't seen the concept of channel offsets.

lelegard, 2019-02-20:
> But the fact is, I'm not based in the US either. So I cannot test a working ATSC tuner under Linux any more than you can, thus why I would need to be sure there was one that worked before asking my friend to buy another device.

At least, he could test the one he already has on Linux, can't he ?

> As far as scanning is concerned, from what I saw when testing the software above, it would be a very similar UHF based scan for ATSC, although I haven't seen the concept of channel offsets.

Do they use the same frequency layout, from channel 21 (474 MHz) to 69 (858 MHz) with 8 MHz between channels?

oviano, 2019-02-20:
[image]

lelegard, 2019-02-20:
Thanks. So, the frequency layout is completely different from what I see here in Europe. This will not be very difficult to implement in TSDuck. However, I cannot leave code like this untested in the product. If you can make an ATSC tuner work on Linux and if you provide testing, then I can propose an implementation of its scanning.

lenosj, 2019-02-20:
On Wed, Feb 20, 2019 at 1:39 PM Thierry Lelegard <notifications@github.com>
wrote:

> But the fact is, I'm not based in the US either. So I cannot test a
> working ATSC tuner under Linux any more than you can, thus why I would need
> to be sure there was one that worked before asking my friend to buy another
> device.
>
> At least, he could test the one he already has on Linux, can't he ?
>
I have two tuners that I have used in the past (on linux) and would be
willing to bring them out of mothballs to test.

> As far as scanning is concerned, from what I saw when testing the software
> above, it would be a very similar UHF based scan for ATSC, although I
> haven't seen the concept of channel offsets.
>
> Do they [...]
No, ATSC is 6Mhz channels.

lelegard, 2019-02-20:
> I have two tuners that I have used in the past (on linux) and would be willing to bring them out of mothballs to test.

I assume they are ATSC tuners, right? As you can see from this thread, testing would be more than welcome.

lenosj, 2019-02-20:
On Wed, Feb 20, 2019 at 2:19 PM Thierry Lelegard <notifications@github.com>
wrote:

> I have two tuners that I have used in the past (on linux) and would be
> willing to bring them out of mothballs to test.
>
> I assume they are ATSC tuners, right? As you can see from this thread,
> testing would be more than welcome.
>

Yes, ATSC (both 8VSB and QAM256 not that it's OTA) and I'm just outside
Boston, MA so I've got many OTA channels to scan.

Punch zip code 02056 into https://antennaweb.org/ as a reference.

Scott

oviano, 2019-02-20:
Thierry; I was just having a look at Windows tuner.cpp and specifically buildGraph.

Now I'm only just familiarising myself with how this all works, but looking at the code path, I'm struggling too see how it would *ever* accept an ATSC tuner?

Because whenever it finds a matching ATSC tuner, won't the check for the IID_IDVBTuningSpace always fail as this is for DVB-T/S-C only? So it'll always skip ATSC tuners?

My trace is like this:

found tuner filter "WinTV HVR-950 BDA Tuner"
found tuning space "Local Analog Antenna" (Antenna)
tslsdvb: fail to set default tuning space "Local Analog Antenna": The parameter is incorrect.
found tuning space "Local ATSC Digital Antenna" (ATSC)
tslsdvb: IUnknown::QueryInterface: No such interface supported
tuning space "Local ATSC Digital Antenna" does not support IID_IDVBTuningSpace interface
found tuning space "Local ATSC Digital Cable" (ATSCCable)
tslsdvb: IUnknown::QueryInterface: No such interface supported
tuning space "Local ATSC Digital Cable" does not support IID_IDVBTuningSpace interface
found tuning space "ATSC_TV_TUNING_SPACE_ENTRY" (ATSC_TV_TUNING_SPACE_ENTRY)
tslsdvb: IUnknown::QueryInterface: No such interface supported
tuning space "ATSC_TV_TUNING_SPACE_ENTRY" does not support IID_IDVBTuningSpace interface
found tuning space "Analog Auxiliary Input #1" (AuxIn1)
tslsdvb: fail to set default tuning space "Analog Auxiliary Input #1": The parameter is incorrect.
found tuning space "Local Analog Cable" (Cable)
tslsdvb: fail to set default tuning space "Local Analog Cable": The parameter is incorrect.

oviano, 2019-02-20:
Ok, so I think I'm getting somewhere now. Apart from the above, the code is also missing ts::CreateLocatorATSC.

I will aim to submit a patch in the next couple of days with this all fixed up.

oviano, 2019-02-21:
I will test further at the weekend when in a reception area but now at least we are detecting ATSC and attempting to tune:

debug level set to 1
* Debug: debug level set to 1
realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: dvb: debug level set to 1
* Debug: drop: debug level set to 1
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: dvb: found tuner filter "WinTV HVR-950 BDA Tuner"
* Debug: dvb: found tuning space "Local Analog Antenna" (Antenna)
* Error: dvb: fail to set default tuning space "Local Analog Antenna": The parameter is incorrect.
* Debug: dvb: found tuning space "Local ATSC Digital Antenna" (ATSC)
* Debug: dvb: using tuning space "ATSC" ("Local ATSC Digital Antenna")
* Debug: dvb: trying direct connection from tuner (no receiver)
* Error: dvb: failed to connect pins: No combination of intermediate filters could be found to make the connection.
* Debug: dvb: trying receiver filter "WinTV HVR-950 BDA Capture"
* Debug: dvb: trying TIF "BDA MPEG2 Transport Information Filter"
* Debug: dvb: using TIF "BDA MPEG2 Transport Information Filter"
* Debug: dvb: using receiver filter "WinTV HVR-950 BDA Capture"
* Debug: dvb: IBDA_DigitalDemodulator in tuner: 0
* Debug: dvb: IBDA_DigitalDemodulator2 in tuner: 0
* Debug: dvb: IBDA_SignalStatistics in tuner: 0
* Debug: dvb: IKsPropertySet in tuner: 2
* dvb: using WinTV HVR-950 BDA Tuner (ATSC)
* Debug: dvb: tuning to transponder --frequency 509000000 --modulation 8-VSB --spectral-inversion auto
* Error: dvb: IUnknown::QueryInterface: No such interface supported
* dvb: tuned to transponder --frequency 509000000 --modulation 8-VSB --spectral-inversion auto
* Debug: dvb: starting tuner reception
* Error: dvb: no input DVB signal after 5,000 milliseconds
* Debug: Report logging thread terminated

I will set things up on github as per your contributor workflow., and once I've successfully captured a stream at the weekend, I will issue a pull request.

oviano, 2019-02-21:
FYI

https://github.com/oviano/tsduck/commit/6dc24327e0649c437caf339c491ace3095b694e3

lelegard, 2019-02-22:
Hi @oviano 

I have integrated your pull request. Could you please check that your ATSC tuner is correctly detected with the latest version from the repo? And could you ask your contact in the US to perform reception test ?

oviano, 2019-02-22:
I already debugged and verified the ATSC detection from my branch code but yes I will check from master. 

At the weekend my friend will be in a reception area and he has provided my a login to test ATSC reception. I plan to capture a few streams to make sure all is ok and will post back my feedback.

oviano, 2019-02-24:
Ok, so I have had the chance to test latest code with this tuner against a live ATSC signal.

Firstly, the tuner is recognised, as expected:

[tslsdvb.txt](https://github.com/tsduck/tsduck/files/2897804/tslsdvb.txt)

Unfortunately, it wouldn't lock to the signal. The tune request etc was accepted, but it could never find a signal. I tried everything but eventually figured out the problem. I tried the app EPGCollector and found this was able to lock the signal and capture streams. Fortunately, the source code is available so I was able to see what parameters etc are set in this app vs TSDuck.

Firstly, tsduck currently has this code for DVB-T:

    // If this is a DVB tuning space, get DVB interface of tune request and set ids to wildcards.
    ComPtr<::IDVBTuneRequest> dvb_request;
    dvb_request.queryInterface(tune_request.pointer(), ::IID_IDVBTuneRequest, report);
    if (!dvb_request.isNull() &&
        (!PUT(dvb_request, ONID, -1) ||
         !PUT(dvb_request, TSID, -1) ||
         !PUT(dvb_request, SID, -1))) {
        return false;
    }

It seems that we should have the equivalent for ATSC:

    // If this is an ATSC tuning space, get ATSC interface of tune request and set channels to wildcards.
    ComPtr<::IATSCChannelTuneRequest> atsc_request;
    atsc_request.queryInterface(tune_request.pointer(), ::IID_IATSCChannelTuneRequest, report);
    if (!atsc_request.isNull() &&
        (!PUT(atsc_request, Channel, -1) ||
         !PUT(atsc_request, MinorChannel, -1))) {
        report.error(u"Coouldn't Set channel or minor channel");
        return false;
    }

It still didn't work though. What I found was that at least this tuner does not work when you set the carrierFrequency in the ATSC Locator. I know that sounds pretty stupid, but it seems you have to set this to -1 and instead set the corresponding physicalChannel.

This is current TSDuck code for the ATSC Locator, that I added the other day:

    if (loc.isNull() ||
        !CheckModEnum(params.inversion, u"spectral inversion", SpectralInversionEnum, report) ||
        !CheckModEnum(params.modulation, u"modulation", ModulationEnum, report) ||
        !PUT(loc, CarrierFrequency, long(params.frequency / 1000)) ||  // frequency in kHz
        !PUT(loc, Modulation, ::ModulationType(params.modulation)))
    {
        return false;
    }

Here is what I hacked it to to make it work:

    if (loc.isNull() ||
        !PUT(loc, CarrierFrequency, -1) ||  // frequency in kHz
        !PUT(loc, InnerFEC, ::BDA_FEC_METHOD_NOT_SET) ||
        !PUT(loc, InnerFECRate, ::BDA_BCC_RATE_NOT_SET) ||
        !PUT(loc, OuterFEC, ::BDA_FEC_METHOD_NOT_SET) ||
        !PUT(loc, OuterFECRate, ::BDA_BCC_RATE_NOT_SET) ||
        !PUT(loc, Modulation, ::BDA_MOD_NOT_SET) ||
        !PUT(loc, SymbolRate, -1) ||
        !PUT(loc, PhysicalChannel, long(params.frequency)) ||
        !PUT(loc, TSID, -1))
    {
        report.error(u"Couoldn't set ATSC locator info");
        return false;
    }

i.e. to test I am passing in the channel in the frequency parameter. Obviously this is not the right solution, but it illustrated the problem and allows me to test. The other parameters are all defaulted in the same way EPGCollector does it.

Here is the EPGCollector source code for reference:

https://www.dropbox.com/s/v04f6p1o861r7h6/BDAGraph1.cs?dl=0

So with the above changes, we can lock a signal and capture the streams. Note that again, the signal strength is reported always as 100%. Perhaps it is a feature of Hauppauge cards that the raw signal strength reported is already a percentage (TSDuck's calculation then divides it by 1000 and subtracts it from 100).

For your information, here are the filter graphs and DirectShow Spy output for both TSDuck and EPGCollector when capturing using this card.

[ATSC TSDuck DSSPY.txt](https://github.com/tsduck/tsduck/files/2897811/ATSC.TSDuck.DSSPY.txt)

[image]

[ATSC EPGC DSSPY.txt](https://github.com/tsduck/tsduck/files/2897812/ATSC.EPGC.DSSPY.txt)

[image]

[image]

The next issue I found was that the transport streams did not have all the tables I was anticipating. For example there are no SDT tables:

[capture_ch10_analysis.txt](https://github.com/tsduck/tsduck/files/2897814/capture_ch10_analysis.txt)

There is nothing in this output indicating the channel names. The information is there somewhere though, as TSReader and VLC both report it (the example below is channel 12, but it's the same with channel 10).

[image]

I did a small bit of research and it seems there are other ATSC-specific tables that contain this information:

http://www.tvwithoutborders.com/tutorials/dtv_intro/atsc_psip/vct/

So I think to be able to do things like -P zap "channel name" then tsduck would need to be parsing these tables too?

lelegard, 2019-02-24:
Hi @oviano 

Thanks a lot for this excellent feedback. The case of PhysicalChannel is a typical example of DirectShow insanity. The [Microsoft doc here](https://docs.microsoft.com/en-us/previous-versions/ms784486(v=vs.85)) does not define what is a "PhysicalChannel". It is just a `long` without semantics nor unit. The same doc specifies that `IATSCLocator` derives from `IDigitalLocator` and `ILocator`. The latter has a `CarrierFrequency` field which is defined in kHz (why not Hz?). But ATSC needs to put the frequency in `PhysicalChannel` (in Hz as I see in your code).

Honestly, I wonder why anyone is still surprised that I __hate__ DirectShow...

This is also why I mentioned that producing Windows DVB (or ATSC) code without access to real devices and networks is impossible.

Given the MS doc about `IATSCLocator`, I would not be surprised if all ATSC tuners behave that way. So, could you please integrate your modifications with the latest code from the repo, check that it works correctly and then submit a pull request.

Concerning the signalisation, TSDuck is clearly influenced by DVB for practical reasons (working in Europe). I am ready to enhance it with ATSC signalisation. But it needs some real data to test. It would be a great help if you could capture 20 or 30 seconds of a few complete ATSC transport streams from the US. Please post them in some file sharing site and I will post them in the [TSDuck streams repository](https://tsduck.io/streams/) and use them as test base.

oviano, 2019-02-24:
Just to clarify - the value that goes into physicalChannel is the RF channel in the range 2-69. It's not a frequency at all, although it maps to a frequency which I presume is fixed for ATSC. I was just using the params.frequency for convenience in my testing. I believe the correct solution would be to continue to pass in frequency, but in the create ATSC locator function it should convert this to a channel in the range 2-69. It would then pass -1 to the CarrierFrequency.

I can easily put a mapping from frequency to channel directly into that function but this seems to crossover with some of your in-progress work for channel frequency configuration - maybe you already have a function that would take a frequency, look it up in the ATSC table and give back a channel number? What we can then do is do the look up, but if it fails (an invalid frequency) then reject creating the locator?

Yesterday I captured five full ATSC transport streams, you can find them all here:

https://www.dropbox.com/sh/f5xz8y86dbzmcay/AACEC9bVjpwckDvMmsLnrD2Ka?dl=0

The .txt files are the corresponding verbose debug output produced when capturing. The one with the epgc suffix was captured using EPGCollector, for comparison. It was captured at a different time which I think is why it has 3 transport streams but the tsduck equivalent for the same channel only has 2. At least I hope that is the reason!

My friend is no longer in an ATSC reception zone so these will have to do for now. If there is anything else specific you need, post it here and next time he's in a reception zone I will capture some more.

lelegard, 2019-02-24:
@oviano,
> Just to clarify - the value that goes into physicalChannel is the RF channel in the range 2-69. It's not a frequency at all, although it maps to a frequency which I presume is fixed for ATSC.

This is incredibly stupid from DirectShow. As I previously mentioned, the frequency allocation in the UHF or VHF band is not a matter of ATSC or DVB, it is not even a matter of analog vs. digital TV and not even TV at all. It is just a matter of local regulation. I do not know if ATSC tuners are used outside US, in a country with a different frequency layout. If this is the case, then DirectShow will not work in these territories.

@lenosj, I would be very interested in getting feedback on Linux and further tests on real ATSC streams. One reliable technical source is required on the US soil to be able to support ATSC correctly.

oviano, 2019-02-24:
Ok, I've only just started looking at the new channel region stuff, but couldn't I just do this for the ATSC Locator:

	// It seems that with DirectShow, the CarrierFreequency must be set to -1
	// for ATSC tuning to work and the physicalChannel used instead. This means
	// we need to take the frequency and map it to the corresponding HF channel.
	const HFBandPtr uhf(HFBand::Factory(u"", HFBand::UHF));
	const HFBandPtr vhf(HFBand::Factory(u"", HFBand::VHF));
	long physical_channel;
	if (uhf->inBand(params.frequency)) {
		physical_channel = uhf->channelNumber(params.frequency);
	}
	else {
		physical_channel = vhf->channelNumber(params.frequency);
	}

...then pass -1 to CarrierFrequency, and pass in the physical channel?

lelegard, 2019-02-24:
This is the right way in principle. However, there is a difficulty here. By using `u""` in `HFBand::Factory`, you use the default region. This ignores the command line option `--hf-band-region`.

Until now, my perception was that UHF and VHF channel numbers were just legal conventions and they were converted to a frequency during command line parsing. The frequency looked like the only valid technical information. Now, I have to think about a mechanism to make the UHF/VHF bands more widely accessible. Because of, guess what, ... DirectShow. I really really hate it. Really.

Stay tuned...

oviano, 2019-02-24:
Yes, I realised it would use the default region after posting and looking into it more.

In the interests of getting some code working for my project I will hardcode to "usa" for now. If you want me to issue a pull request with this then I can do that, it's up to you.

My only caveat with all of this, is that this is one test done against one tuner card. Are we even sure that the CarrierFrequency won't work with other cards?

oviano, 2019-02-24:
Or another approach might be simply to make ATSC use a new parameter physical_channel, on both Windows and Linux and not use frequency at all?

lelegard, 2019-02-24:
I juste pushed a radical solution: the value of `--hf-band-region` is globally set in the application. So, you code should work.

I would recommend to also check that the frequency is in the VHF band before getting the channel and report an error if the frequency is neither in UHF nor VHF band.

I do not know if always setting the region to "usa" when the tuner is an ATSC one is a good idea or not.

I just found this [stackoverflow topic](https://stackoverflow.com/questions/21606254/what-is-the-correct-way-to-tune-and-scan-for-an-atsc-channel/24308200) which addresses the question. They always set CarrierFrequency to -1. However, they mention "minor" and "major" channels. I have no idea what it is.

lelegard, 2019-02-24:
This [Microsoft doc](https://docs.microsoft.com/en-us/previous-versions/ms784466(v=vs.85)) mentions major and minor channels. I mean _mentions_, not _explains_. Possibly, the "major channel" and "physical channel" are the same. Yet another insanity? Please, have a look at the Linux code, not perfect, but so much simpler...

oviano, 2019-02-24:
Ok so what I found was that the major and minor channels are whatever has been assigned to the tv station (major) and it’s channels (minor). They are not the same as the physical channel which is essentially mapped one-to-one to the frequency.

Please see the antennaweb link I posted - you will see the concept of channels like 13.1 (major.minor) and then separately RF channels.

I believe by setting channel and minor channel to -1 it will get a stream with all broadcasts at that RF channel.

I’m certainly not an authority on the subject though this was a weekend googling!

oviano, 2019-02-24:
On the subject of channel/minorChannel, I believe (but I have not tested it) that if you specify these it would act much like -P zap in that the focusing on one channel only could be done before the stream even reaches tsduck.

This might be nice to expose, assuming it's also possible on Linux.

oviano, 2019-02-25:
Well I couldn't see any reference to this in the Linux TV DVB API, so I assume this higher-level filtering is a Windows/DirectShow convenience.

I assume that it must tune to the frequency/physical channel and then do it's own zapping based on the major/minor channel, but within the DirectShow framework.

Which makes sense actually, as I can't see how it could somehow only receive certain AV streams from a given frequency!

lelegard, 2019-02-25:
TSDuck is supposed to expose a portable and rational interface and hide the dirt of DirectShow. So, please keep Windows-specific concepts out of the public interface or command line.

A transport stream comes from a frequency (and associated modulation parameters). Any other concept such as VHF/UFH channel or the channel definition file is just a command-line-level convenience. When it comes to actual tuning, we handle frequencies only. If some crappy specific implementation requests to convert back the frequency into UHF channel (and naming it "physical channel" to make sure nobody understands what it means), this is what I call "dirt". Let's keep it hidden.

The same applies to this major/minor channel stuff. This is some non-portable Windows equivalent of the channels definition file. Let's keep it hidden as well.

oviano, 2019-02-25:
That's fine, I won't issue a pull request for anything I do in that area.

The only thing I would say is that the major/minor channel doesn't seem to be a Windows concept - it's an ATSC concept. Or more precisely, a PSIP concept:

https://en.wikipedia.org/wiki/Program_and_System_Information_Protocol

In fact, if you look up any of the US channels on wikipedia, such as this one...

https://en.wikipedia.org/wiki/KUEN

...the terminology they use is not frequency, but

Virtual channel (this is the major channel in DirectShow terms, indicating the station)
UHF Channel (this is the physical channel)

Then the minor channel, .1, .2, etc:

[image]

So yes, only DirectShow exposes a way of selecting the major/minor channel, but rather than see that as a drawback to Directshow you could argue that Linux could also do the same.

lelegard, 2019-02-25:
What I mean is that major.minor is a higher-level concept which is equivalent to the channel definition file. There must be some database of association between the major/minor and the physical frequencies. To make it portable, we need to write application code for something that is ATSC-specific.

I prefer to enhance the recognition of the ATSC signalization and put the right channel names in the existing portable network-agnostic channel definition file.

oviano, 2019-02-25:
Sounds good. I see you’ve begun already - look forward to trying it out!

lelegard, 2019-03-16:
Hi @oviano 

Can you confirm that `tsscan` works on an ATSC network and the right channel names and tuning parameters are recorded in the XML channel file? 

If this works, we may close this issue.

oviano, 2019-03-16:
Yes I can confirm it works just perfectly.

The only issue I’ve found was that the Zap plugin won’t work if you give it a service name because I think it relies on finding the name in the SDT, which doesn’t exist for ATSC.

Technically you have covered this in your manual as it states that you must use the SID in the absence of an SDT but logically it might be nice to add the option for ATSC too.

lelegard, 2019-03-16:
> Technically you have covered this in your manual as it states that you must use the SID in the absence of an SDT but logically it might be nice to add the option for ATSC too.

Yes, I planned this. Selecting the service by name on ATSC is easy. I just need to check how to properly rebuild the PSIP stream with a stripped xVCT in PID 0x1FFB. This is not strictly required for the plugin to work. But this is required to rebuild a clean and consistent SPTS. The problem with ATSC is that PID 0x1FFB contains many different things, cycled tables (MGT, xVCT) and one-shot tables (STT). This is a very poor design choice which complicates things quite a lot. In DVB, time tables (TOT and TDT) are sent in a separate PID. Generally speaking, with DVB, cycled and one-shot tables are dissociated. With ATSC, they are mixed.

oviano, 2019-03-16:
By the way, let me know if you require any more ATSC streams. I hope the last ones were useful - my friend has a machine permanently in the reception zone now so I can probably get some more captures at some point if they would be useful.

lelegard, 2019-03-16:
Thanks for the offer. Ideally, a 60 seconds capture of each available TS would be great. As well as the XML channels file for the complete network, to keep track of frequencies.

oviano, 2019-03-16:
Ok will do. You can close this issue. I’ll reopen a new issue when I’ve got some streams.

closed 2019-03-16 by lelegard