tsduck/tsduck #229: So with a multimode tuner device, how to select DVB-C?

oviano, 2019-02-20:
I'm struggling a bit with this; I have a WinTV-SoloHD which is a single tuner supporting DVB-T, DVB-T2,DVB-C.

It is connected to a FreeView aerial, and tsscan successfully finds a bunch of DVB-T and DVB-T2 channels. All works fine.

Let's say I wanted to connect it to tune it to a DVB-C though. How would I do that?

tslsdvb only lists a single device, of type DVB-T.

I tried adding a fake DVB-C channel to the top of my channels.xml as follows:

  <network id="0x1111" type="DVB-C">
    <ts id="0x5040" onid="0x233A">
      <dvbc frequency="123,456,789"/>
      <service id="0x5640" name="Test" PMTPID="0x02BD" type="0x01"/>
    </ts>
  </network>

I then try:

tsp -v -I dvb -c "Test" -P zap "Test" -P until -s 1 -O drop
* dvb: using Hauppauge WinTV-soloHD DVBT Tuner (DVB-T)
* Error: dvb: inconsistent tuner parameter type

I would expect it would be trying to use the tuner's DVB-C capability, even though it would of course fail to tune, but as you can see it thinks it is only DVB-T?

lelegard, 2019-02-20:
Multi-mode tuners are handled differently by their respective drivers.

Some drivers for multi-mode tuners expose two tuner devices, typically for distinct modulations. On an application standpoint, you simply select the device with the right modulation.

Some other tuners expose only one tuner with one specific modulation. This is your case where the tuner is supposed to support DVB-T/T2 (same tuner) and DVB-C but the driver exposes only the DVB-T/T2 part. There is no standard API, neither in Windows DirectShow nor in Linux TV, to handle multi-mode tuners. I guess that each driver has a specific API to configure the tuner in either mode. This can be used by applications which are aware of the various proprietary extensions, which is not the case with TSDuck.

On a pragmatic standpoint, many tests can be done to "guess" the right way to handle various tuners, including which graph to build. This is what I have done in the past with DirectShow. But this is a developer task. So, if no developer (including myself) has a given problematic tuner, nothing will be done and the tuner will not be supported by TSDuck. Doing lengthy exchanges, test code, back and forth between a developer without tuner and a user with a tuner takes time and I have clearly no time to do this. So, if you want support for a specific exotic tuner which is not properly handled in standard ways and which I have not, you need to find a developer to adapt the Tuner layer in TSDuck. Not very satisfying I am afraid, but I have no better offer, sorry.

oviano, 2019-02-21:
So where a tuner has more than one mode, is it not just a case that during buildGraph we need an optional filter so that it only selects according to tuning spaces of a certain type?

i.e. currently it loops through all the tuning spaces it knows, to see which work with the tuner. As soon as it finds one it selects it and ignores the rest. In the case of the two tuners I've tested, luckily it chooses DVB-T and ATSC respectively, but in theory my UK one at least might match DVB-C. So maybe you would pass in an argument for the preferred tuner type which would be passed down (somehow) to buildGraph?

I'm talking Windows obviously, I do not know the equivalent Linux mechanism.

oviano, 2019-02-21:
So it turns out that the issue with this tuner only showing up as DVB-T was actually because there is no DVB-C tuning space by default in a Windows installation in the registry.

However, as soon as you tell WinTV to attempt to tune DVB-C channels then it WinTV creates one. After that Tsduck can see a separate DVB-C tuner which appears to function just fine.

Interesting that it wasn't the installation of WinTV, or the driver for the tuner, that caused the DVB-C space to be created, simply the first time WinTV attempted to use DVB-C.

Which makes me wonder if tsduck shouldn't be doing something similar, and making sure a tuning space exists for each type that it supports, rather than bail out with an error?

lelegard, 2019-02-21:
Could you post the output of `tslsdvb -v` after the creation of the DVB-C device?

oviano, 2019-02-22:
Sure, here you go.

[tslsdvb-wintv-solohd.txt](https://github.com/tsduck/tsduck/files/2892703/tslsdvb-wintv-solohd.txt)

lelegard, 2019-02-22:
So, if I understand correctly, you have initially only one tuner device:
~~~
0: "Hauppauge WinTV-soloHD DVBT Tuner" (DVB-T)
~~~
Then, you use Hauppauge WinTV software to tune to DVB-C and then you get two tuner devices:
~~~
0: "Hauppauge WinTV-soloHD DVBC Tuner" (DVB-C)
1: "Hauppauge WinTV-soloHD DVBT Tuner" (DVB-T)
~~~
Is that correct?

What do you see after rebooting? Does the DVB-C tuner persist or does it disappear?

WinTV is a Hauppauge proprietary application, working with a Hauppauge device. From a generic application, we can only use what is installed by Windows and the various drivers (or software).

oviano, 2019-02-22:
Yes, that's correct, and yes it persists between reboots because WinTV has created the DVB-C registry entry.

I also noticed a few other registry entries appearing during my testing lately, presumably created by the various pieces of software I was trying.

lelegard, 2019-02-22:
OK, the good news is that the DVB-C setup is persistent across reboots.

So I would say that the Hauppauge installation process is incorrect.

When you get a new device, you install the corresponding drivers (and associated support software such as DirectShow filters). The installed software should give you access to the new device.

In your case, the software installation from Hauppauge only installs what is necessary for DVB-T but not for DVB-C. The missing software for DVB-C gets installed the first time you run WinTV on DVB-C. My conclusion is that this DVB-C setup should have been performed during installation. Clearly, the Hauppauge people could not imagine that their device could be used without their own application.

I will update the TSDuck documentation to mention this additional setup phase. Many tuners have specificities and most of them have additional setup notes. This is just one more...

I would be interested to see how this works on Linux.

oviano, 2019-02-22:
Yes, it does seem to be something that should be done at install time.

lelegard, 2019-02-24:
I have updated the TSDuck user's guide so that users know what to do to get access to the DVB-C tuner. So, this issue seems complete now.

@oviano, I know you don't like Linux as much as I don't like Windows. Nonetheless, would you have the opportunity to test your tuners on Linux? It would be interesting to know if and how they work and how the DVB-T / DVB-C duality is handled.

oviano, 2019-02-24:
Hi Thierry; I'm in the process of setting up a Ubuntu machine so I will be able to test both my WinHD-Solo and I also have a DVB-S tuner on order. I don't think I will be able to test anything US-based with Linux any time soon though.

I will also shortly be feeding back on my ATSC Windows testing (WinTV HVR-950Q) from yesterday.

oviano, 2019-02-24:
Just to add; I posted in the WinTV HVR 950Q thread about how an app called EPGCollector works.

It’s another example of an app that creates its own tuning spaces on the fly (not permanently in the registry). It creates them for all variants DVB-C, DVB-T presumably to avoid having to deal with this kind of issue.

Just thought I’d mention it.

closed 2019-03-11 by lelegard