tsduck/tsduck #230: Suggestion: make tuning uhf_channel and vhf_channel also applicable to ATSC

oviano, 2019-02-21:
Currently the documentation states:

--uhf-channel value
Used for DVB-T tuners only.
Specify the UHF channel number of the carrier. Can be used in replacement to --frequency. Can be combined with an --offset-count option. Valid UHF channels are usually 21 to 69. The resulting frequency is
306 MHz + (uhf-channel * 8 MHz) + (offset-count * 166.6 kHz).
--vhf-channel value
Used for DVB-T tuners only.
Specify the VHF channel number of the carrier. Can be used in replacement to --frequency. Can be combined with an --offset-count option. Valid VHF channels are usually 5 to 12. The resulting frequency is
142.5 MHz + (vhf-channel * 7 MHz) + (offset-count * 166.6 kHz).

I propose this becomes:

--uhf-channel value
Used for DVB-T and ATSC tuners only.
Specify the UHF channel number of the carrier. Can be used in replacement to --frequency. For DVB-T can be combined with an --offset-count option. Valid UHF channels are usually 21 to 69 for DVB-T and 14 to 51 for ATSC. The resulting frequency is
DVB-T: 306 MHz + (uhf-channel * 8 MHz) + (offset-count * 166.6 kHz).
ATSC: 389 Mhz + ((uhf-channel * 6 MHz)

--vhf-channel value
Used for DVB-T and ATSC tuners only.
Specify the VHF channel number of the carrier. Can be used in replacement to --frequency. For DVB-T can be combined with an --offset-count option. Valid VHF channels are usually 5 to 12 for DVB-T and 5 to 13 for ATSC. The resulting frequency is
DVB-T: 142.5 MHz + (vhf-channel * 7 MHz) + (offset-count * 166.6 kHz).
ATSC: 49 MHz + (vhf-channel * 6 MHz) for channels 5-6
ATSC: 135 MHz + (vhf-channel * 6 MHz) for channels 7-13

lelegard, 2019-02-21:
Yes, I am working on a configurable definition of UVH and UHF layout by country or region, based on an XML file.

oviano, 2019-02-21:
Great!

lelegard, 2019-02-21:
More precisely, the VHF & UHF band layouts depend on the region, not the modulation. So, the next version will include an XML file containing the following (for now):
```xml
<tsduck>

  <hfband type="UHF">
    <region name="europe"/>
    <region name="uk"/>
    <region name="france"/>
    <channels first_channel="21" last_channel="69" base_frequency="474,000,000" channel_width="8,000,000"
              first_offset="-1" last_offset="3" offset_width="166,666"/>
  </hfband>
  
  <hfband type="VHF">
    <region name="europe"/>
    <region name="uk"/>
    <region name="france"/>
    <channels first_channel="5" last_channel="12" base_frequency="177,500,000" channel_width="7,000,000"
              first_offset="-1" last_offset="3" offset_width="166,666"/>
  </hfband>
  
  <hfband type="UHF">
    <region name="usa"/>
    <region name="us"/>
    <channels first_channel="14" last_channel="69" base_frequency="473,000,000" channel_width="6,000,000"/>
  </hfband>
  
  <hfband type="VHF">
    <region name="americas"/>
    <region name="usa"/>
    <region name="us"/>
    <region name="korea"/>
    <region name="taiwan"/>
    <region name="philippines"/>
    <channels first_channel="1" last_channel="1" base_frequency="47,000,000" channel_width="6,000,000"/>
    <channels first_channel="2" last_channel="4" base_frequency="57,000,000" channel_width="6,000,000"/>
    <channels first_channel="5" last_channel="6" base_frequency="79,000,000" channel_width="6,000,000"/>
    <channels first_channel="7" last_channel="13" base_frequency="177,000,000" channel_width="6,000,000"/>
  </hfband>
  
  <hfband type="VHF">
    <region name="japan"/>
    <channels first_channel="1" last_channel="3" base_frequency="93,000,000" channel_width="6,000,000"/>
    <channels first_channel="4" last_channel="12" base_frequency="173,000,000" channel_width="6,000,000"/>
  </hfband>
  
</tsduck>
```

The default region is `"europe"`(sorry, I decide 😄). But a user can define a different default using command line options or in his TSDuck configuration file (see user's guide appendix A) using option `"default.region"`.

The same layout can be defined for several regions. There are small differences such as the use of offsets -1/+3 in France but -1/+1 in UK, but I choose to ignore them, only keeping the largest range.

If you have additional information about VHF and UHF channel layouts in other parts of the world, please post them here.

Support code is ready but it must be integrated with the rest of the applications. Will be done in the next few days.

oviano, 2019-02-22:
Looks nice and flexible and covers the structure I posted elsewhere for the US. Not that I'm based in the US (I'm in the UK), but that's where my focus has been lately.

lelegard, 2019-02-24:
The handling of region-specific UHF/VHF band layouts has been integrated. You can try it from the latest repo.

closed 2019-03-11 by lelegard