tsduck/tsduck #231: compile error debian  g++ 6

samerajarmeh, 2019-02-21:
my g++ version
no problem with older revision tsduck-3.16-1110
root@debian:~/tsduck# g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)

 [CXX] tsAACDescriptor.cpp
In file included from /usr/include/c++/6/bits/move.h:57:0,
                 from /usr/include/c++/6/bits/stl_pair.h:59,
                 from /usr/include/c++/6/bits/stl_algobase.h:64,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/string:40,
                 from tsPlatform.h:655,
                 from tsByteBlock.h:36,
                 from tsAbstractSignalization.h:36,
                 from tsAbstractDescriptor.h:36,
                 from tsAACDescriptor.h:36,
                 from tsAACDescriptor.cpp:34:
/usr/include/c++/6/type_traits: In instantiation of 'struct std::underlying_type<unsigned char>':
tsxmlElement.h:426:107:   required by substitution of 'template<class ENUM, typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous>, class INT> bool ts::xml::Element::getIntAttribute(ENUM&, const ts::UString&, bool, ENUM, INT, INT) const [with ENUM = unsigned char; typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous> = <missing>; INT = <missing>]'
tsAACDescriptor.cpp:186:88:   required from here
/usr/include/c++/6/type_traits:2256:38: error: 'unsigned char' is not an enumeration type
       typedef __underlying_type(_Tp) type;
                                      ^~~~
../../Makefile.common:898: recipe for target 'release-x86_64/tsAACDescriptor.o' failed
make[2]: *** [release-x86_64/tsAACDescriptor.o] Error 1

lelegard, 2019-02-21:
This looks like a duplicate of #219. 

There is a range of broken versions of GCC which fail to compile this code (which is correct code anyway but new code since 3.16).

It compiles correctly with
- GCC 4.8.5 (CentOS 7.6)
- GCC 8.2.0 (Ubuntu 18.10)
- GCC 8.2.1 (Fedora 29)
- GCC 9.0.0 (Fedora 30 beta)
- LLVM/clang 10.0.0 (macOS Mojave)
- Visual Studio 2017 (Windows 10).

Your compilation fails with GCC 6.3.0 and in #219 it failed with GCC 7.3.0. So, GCC 6 & 7 seem broken. You need to use either an older or more recent one. Since you use Ubuntu, upgrading to the latest Ubuntu is fine.

athoik, 2019-02-24:
Hi, 

I am having this problem as well. 

```
  [CXX] tsAACDescriptor.cpp
In file included from /opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/bits/move.h:57:0,
                 from /opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/bits/stl_pair.h:59,
                 from /opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/bits/stl_algobase.h:64,
                 from /opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/bits/char_traits.h:39,
                 from /opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/string:40,
                 from tsPlatform.h:655,
                 from tsByteBlock.h:36,
                 from tsAbstractSignalization.h:36,
                 from tsAbstractDescriptor.h:36,
                 from tsAACDescriptor.h:36,
                 from tsAACDescriptor.cpp:34:
/opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/type_traits: In instantiation of 'struct std::underlying_type<unsigned char>':
tsxmlElement.h:426:107:   required by substitution of 'template<class ENUM, typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous>, class INT> bool ts::xml::Element::getIntAttribute(ENUM&, const ts::UString&, bool, ENUM, INT, INT) const [with ENUM = unsigned char; typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous> = <missing>; INT = <missing>]'
tsAACDescriptor.cpp:186:88:   required from here
/opt/oe/build/tmp/sysroots/osmio4k/usr/include/c++/6.2.0/type_traits:2245:38: error: 'unsigned char' is not an enumeration type
       typedef __underlying_type(_Tp) type;
                                      ^~~~
../../Makefile.common:898: recipe for target 'release-aarch64/tsAACDescriptor.o' failed
make[2]: *** [release-aarch64/tsAACDescriptor.o] Error 1
```

So I guess I need to use a specific revision before 3.16.

Version of problematic gcc is 6.2.0.

```
$CXX --version
aarch64-oe-linux-g++ (GCC) 6.2.0
```

lelegard, 2019-02-24:
TSDuck 3.16 compiles OK on GCC 6. Some new code in 3.17 triggers the bug in GCC 6 & 7. Fortunately, most latest Intel-based distros have either older or new versions. On the other hand, embedded distros like yours (and my Raspbian) seem to like GCC 6.

On my 32-bit Rasbian (Raspberry Pi), I have successfully used [this article](https://solarianprogrammer.com/2017/12/08/raspberry-pi-raspbian-install-gcc-compile-cpp-17-programs/) to install GCC 8.1.0.

closed 2019-03-14 by lelegard