tsduck/tsduck #233: Thread-safety of libtsduck?

oviano, 2019-02-24:
More of a theoretical Q really than a practical one - obviously tsp, the tools etc run with one input, tuner whatever, the processors and then the output so not an issue, but an app that links to the library on a system with multiple tuner hardware, would the library cope with concurrent tuning/reception/processing going on in separate threads?

lelegard, 2019-02-25:
Hi @oviano,

Yes, the TSDuck library is thread-safe. Utilities such as `tsp` or `tsswitch` are multi-threaded and the TSDuck library contains all the necessary tools to manage threads and synchronize between each other.

There is no specific restriction about tuner reception. For instance, you may run `tsswitch`  with multiple `dvb` input plugins. Each plugin will run in its own thread, managing its own tuner.

Of course, like most thread-safe libraries, this does not mean that you can do anything without wondering about responsibilities. You can concurrently work on multiple tuners but using one tuner object per thread. If you want to share data, then you must synchronize access using `Mutex` and `Guard` on this data object.

As an example, you may have a look at the `SafePtr` template class. This is an implementation of the safe pointer design pattern. It keeps track of reference counters on allocated objects and deallocate them when they are no longer referenced. Using this pattern, you can write C++ code almost as easily as you would write Java code. `SafePtr` is potentially thread-safe in the sense that each instantiation of the template class is associated with a mutex class. Using `NullMutex`, you have no synchronization at all. This provides the best performance by removing the synchronization overhead but you shall not use the same safe pointer object from distinct threads. Using `Mutex`, you have explicit synchronization on the safe pointer object. You may assign or dereference the _safe pointer object_ from multiple threads. But this does not protect the _pointed object_. Let's take the example of a class you recently inspected: `HFBand`. An `HFBand` object is read-only. All its public methods are `const`. So, once it is created, it can be used without restriction from distinct threads, despite its lack of internal synchronization. The pointer type `HFBandPtr`, on the other hand, is defined as `SafePtr<HFBand,Mutex>` with explicit synchronization. And finally, the internal private class `HFBandRepository` is read/write and indirectly accessible from concurrent threads. As a consequence, it contains an internal `Mutex`.

oviano, 2019-02-25:
Brilliant, thank you.

It's an edge case in my project that a server would be tuning against two devices but it's good to know that I don't need to restrict it. All tuning code for a particular tuner is done in its own thread, so that's fine.

closed 2019-02-28 by lelegard