tsduck/tsduck #236: REQ: Enhancement for bitrate_monitor pid selection

ryci-us, 2019-03-03:
Hello,

it would be very handy to have possibility to specify multiple pid's for bitrate_monitor, and to have possibility to not specify any in case you want to monitor bitrate of whole input stream.

Regards.

ryci-us, 2019-03-03:
I figured out that it's possible to monitor multiple pid's by simple configuring multiple bitrate_monitor processors each with different pid.

But it's still not clear how to monitor whole input stream bitrate.

lelegard, 2019-03-17:
> But it's still not clear how to monitor whole input stream bitrate

Just added this capability. If you do not specify a PID, then the bitrate of the complete TS is monitored.

Moreover, I added the option `--tag` to specify some tag string for the alarm message (eg. "Foo video", "xxx EMM", etc.) So, when you use several instances of the plugin for various purposes, each message can be more easily identified. The `--tag` trick was already used in other plugins.

closed 2019-03-21 by lelegard