tsduck/tsduck #238: Was trying TSP on ATSC and it crashes

joolzg, 2019-03-04:
Valgrind output

==393== Using Valgrind-3.11.0 and LibVEX; rerun with -h for copyright info
==393== Command: tsp -v -I dvb --adapter 7 --frequency 12486000000 --symbol-rate 20000000 --polarity vertical --lnb 9750,11250,11700 -P until --seconds 10 -P analyze -o out_12486.txt -P eit -o eit_12486.txt -O drop
==393== 
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* eit: creating eit_12486.txt
==393== Warning: noted but unhandled ioctl 0x6f42 with no size/direction hints.
==393==    This could cause spurious value errors to appear.
==393==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
* dvb: using /dev/dvb/adapter7 (DVB-S)
==393== Warning: noted but unhandled ioctl 0x6f43 with no size/direction hints.
==393==    This could cause spurious value errors to appear.
==393==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==393== Warning: noted but unhandled ioctl 0x6f41 with no size/direction hints.
==393==    This could cause spurious value errors to appear.
==393==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==393== Warning: noted but unhandled ioctl 0x6f2d with no size/direction hints.
==393==    This could cause spurious value errors to appear.
==393==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
* dvb: tuned to transponder --frequency 12486000000 --symbol-rate 20000000 --fec-inner auto --spectral-inversion auto --polarity vertical --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 --lnb 9750,11250,11700 --satellite-number 0
* dvb: signal locked: yes, strength: 54%
==393== Invalid read of size 8
==393==    at 0xCC84792: ts::DVBInput::getBitrate() (in /usr/bin/tsplugin_dvb.so)
==393==    by 0x407AF2: ts::tsp::InputExecutor::getBitrate() (in /usr/bin/tsp)
==393==    by 0x408435: ts::tsp::InputExecutor::initAllBuffers(ts::ResidentBuffer<ts::TSPacket>*) (in /usr/bin/tsp)
==393==    by 0x4068D6: MainCode(int, char**) (in /usr/bin/tsp)
==393==    by 0x4048CA: main (in /usr/bin/tsp)
==393==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==393== 
==393== 
==393== Process terminating with default action of signal 11 (SIGSEGV)
==393==  Access not within mapped region at address 0x0
==393==    at 0xCC84792: ts::DVBInput::getBitrate() (in /usr/bin/tsplugin_dvb.so)
==393==    by 0x407AF2: ts::tsp::InputExecutor::getBitrate() (in /usr/bin/tsp)
==393==    by 0x408435: ts::tsp::InputExecutor::initAllBuffers(ts::ResidentBuffer<ts::TSPacket>*) (in /usr/bin/tsp)
==393==    by 0x4068D6: MainCode(int, char**) (in /usr/bin/tsp)
==393==    by 0x4048CA: main (in /usr/bin/tsp)
==393==  If you believe this happened as a result of a stack
==393==  overflow in your program's main thread (unlikely but
==393==  possible), you can try to increase the size of the
==393==  main thread stack using the --main-stacksize= flag.
==393==  The main thread stack size used in this run was 8388608.
==393== 
==393== HEAP SUMMARY:
==393==     in use at exit: 17,634,678 bytes in 7,467 blocks
==393==   total heap usage: 10,782 allocs, 3,315 frees, 17,923,606 bytes allocated
==393== 
==393== LEAK SUMMARY:
==393==    definitely lost: 328 bytes in 2 blocks
==393==    indirectly lost: 118 bytes in 2 blocks
==393==      possibly lost: 0 bytes in 0 blocks
==393==    still reachable: 17,634,232 bytes in 7,463 blocks
==393==                       of which reachable via heuristic:
==393==                         stdstring          : 208 bytes in 6 blocks
==393==                         multipleinheritance: 1,152 bytes in 3 blocks
==393==         suppressed: 0 bytes in 0 blocks
==393== Rerun with --leak-check=full to see details of leaked memory
==393== 
==393== For counts of detected and suppressed errors, rerun with: -v
==393== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)

lelegard, 2019-03-04:
Hi,

Why do you mention ATSC while your tuner is DVB-S? Do you mean that the same command worked before the recent updates to support ATSC and now fails with these updates? But these modifications were merely applied on Windows-specific code while you are using Linux.

I am away this week and I will not be able to make any test until last week. Could you rebuild in debug mode (`make debug`), enable core dumps, runs the `tsp` command, open the core dump using `gdb` and get a stack backtrace (`bt`)?

joolzg, 2019-03-04:
This feed is from a mexican satellite and uses the ATSC tables.

I did a standard analyze and it crashed.

Last time i tested was on a tsp from Feb 3rd 2019

lelegard, 2019-03-04:
Does this mean that the crash occurs only on this (or similar) TS but the same command with the same tuner works with another satellite, typically DVB? 

Does it work with an official 3.16 binary?

Finally, can you make the debug test I suggested?

joolzg, 2019-03-04:
This was my first test of the updated tsp on a server which is not using it for any real work.

The command is the same, the tuner is the same, the script is the same, only tsp changed.

It used to work, not it does not.

I will rebuild and test with debug on.

joolzg, 2019-03-04:
3.16-1110 works

joolzg, 2019-03-09:
encoder@joemex6:~/src/tsduck$ gdb --args tsp  -v -I dvb --adapter 7 --frequency 12486000000 --symbol-rate 20000000 --polarity vertical   --lnb 9750,11250,11700 -P until --seconds 10 -P analyze -o out_12486.txt -P eit -o eit_12486.txt -O drop
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from tsp...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/tsp -v -I dvb --adapter 7 --frequency 12486000000 --symbol-rate 20000000 --polarity vertical --lnb 9750,11250,11700 -P until --seconds 10 -P analyze -o out_12486.txt -P eit -o eit_12486.txt -O drop
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffef966700 (LWP 21108)]
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* eit: creating eit_12486.txt
* dvb: using /dev/dvb/adapter7 (DVB-S)
* dvb: tuned to transponder --frequency 12486000000 --symbol-rate 20000000 --fec-inner auto --spectral-inversion auto --polarity vertical --delivery-system DVB-S --modulation QPSK --pilots off --roll-off 0.35 --lnb 9750,11250,11700 --satellite-number 0
* dvb: signal locked: yes, strength: 54%

Thread 1 "tsp" received signal SIGSEGV, Segmentation fault.
0x00007ffff018b792 in ts::DVBInput::getBitrate() () from /usr/bin/tsplugin_dvb.so
(gdb) bt
#0  0x00007ffff018b792 in ts::DVBInput::getBitrate() () from /usr/bin/tsplugin_dvb.so
#1  0x0000000000407af3 in ts::tsp::InputExecutor::getBitrate() ()
#2  0x0000000000408436 in ts::tsp::InputExecutor::initAllBuffers(ts::ResidentBuffer<ts::TSPacket>*) ()
#3  0x00000000004068d7 in MainCode(int, char**) ()
#4  0x00000000004048cb in main ()

joolzg, 2019-03-09:
encoder@joemex6:~/src/tsduck$ tslsdvb 
/dev/dvb/adapter0 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter1 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter2 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter3 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter4 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter5 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter6 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)
/dev/dvb/adapter7 (TurboSight TBS 6904 DVBS/S2 frontend, DVB-S, DVB-S2)

lelegard, 2019-03-10:
This does not look like a debug executable. With a debug exec, the `bt` command should give the source file and line number on each frame.

lelegard, 2019-03-10:
Could you please retry with a debug exec, using the latest state of the repo (I found something strange that should have never existed).

joolzg, 2019-03-12:
3.17-1168 from a pull today works

closed 2019-03-12 by joolzg