tsduck/tsduck #239: QUERY - DVB SUBTITLES SUPPORT

jacksonvm, 2019-03-05:
Hi @lelegard ,

I want to Multiplex DVB Subtitles to an existing transport stream file, containing audio and video using TSDUCK. The subtitles are in PES form with .DVB extension. The Subtitle file has been uploaded here https://drive.google.com/open?id=1zjASmL2U2VakpFK_0vBnO7LjzTTQocHK . Can you please guide me how i can achieve the same using tsduck. 

Regards,

Jackson

lelegard, 2019-03-11:
As reported many times in response to various issues here, TSDuck is not a MUX. It does not have the precision to accurately multiplex services or components of a service. Developing a MUX inside the TSDuck toolset is a nice idea and I would love to have enough time to do it. However, developing an accurate MUX takes time and I have no such time. So, this will not happen anytime soon. Unfortunately.

lars18th, 2019-03-12:
Hi @lelegard (and others),

> As reported many times in response to various issues here, TSDuck is not a MUX. 

Any _suggestion_ about a software (or low cost hardware) that can be used for muxing, even offline?
There is currently no open source solution. But maybe a closed source or demo software could be used.

Ideas?

premultiply, 2019-08-30:
Avalpa OpenCaster (Open Source)
or
Dektec MuxXpert (commercial)

lars18th, 2019-08-30:
Hi @premultiply ,

> Avalpa OpenCaster (Open Source)
> or
> Dektec MuxXpert (commercial)

Thank you for your suggestion!

I know about both. But MuxXpert is very expensive (I never used it), and OpenCaster is old and a dead project (futhermore the muxing functionality is worse than with TSDuck).

So, I found another solution pointed by other user of the TSduck: Scienta MPTS Muxer (demo).
It work in realtime for 1 hour.

Regards.